webcamoid/akvcam #2: Can't get it working on Raspberry Pi

lexusburn, 2018-12-26:
I'm trying to use akvcam on pi.
build is ok (with some warnings), but if i call v4l2-compliance -d /dev/video1 -f i get kernel failures:

```
v4l2-compliance SHA   : not available

Driver Info:
	Driver name   : akvcam
	Card type     : Virtual Camera (output device)
	Bus info      : platform:akvcam-1
	Driver version: 4.9.80
	Capabilities  : 0x85200002
		Video Output
		Read/Write
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps   : 0x05200002
		Video Output
		Read/Write
		Streaming
		Extended Pix Format

Compliance test for device /dev/video1 (not using libv4l2):

Required ioctls:
	test VIDIOC_QUERYCAP: OK

Allow for multiple opens:
	test second video open: OK
	test VIDIOC_QUERYCAP: OK
	test VIDIOC_G/S_PRIORITY: OK
	test for unlimited opens: OK

Debug ioctls:
	test VIDIOC_DBG_G/S_REGISTER: OK (Not Supported)
	test VIDIOC_LOG_STATUS: OK (Not Supported)

Input ioctls:
	test VIDIOC_G/S_TUNER/ENUM_FREQ_BANDS: OK (Not Supported)
	test VIDIOC_G/S_FREQUENCY: OK (Not Supported)
	test VIDIOC_S_HW_FREQ_SEEK: OK (Not Supported)
	test VIDIOC_ENUMAUDIO: OK (Not Supported)
	test VIDIOC_G/S/ENUMINPUT: OK (Not Supported)
	test VIDIOC_G/S_AUDIO: OK (Not Supported)
	Inputs: 0 Audio Inputs: 0 Tuners: 0

Output ioctls:
	test VIDIOC_G/S_MODULATOR: OK (Not Supported)
	test VIDIOC_G/S_FREQUENCY: OK (Not Supported)
	test VIDIOC_ENUMAUDOUT: OK (Not Supported)
	test VIDIOC_G/S/ENUMOUTPUT: OK
	test VIDIOC_G/S_AUDOUT: OK (Not Supported)
	Outputs: 1 Audio Outputs: 0 Modulators: 0

Input/Output configuration ioctls:
	test VIDIOC_ENUM/G/S/QUERY_STD: OK (Not Supported)
	test VIDIOC_ENUM/G/S/QUERY_DV_TIMINGS: OK (Not Supported)
	test VIDIOC_DV_TIMINGS_CAP: OK (Not Supported)
	test VIDIOC_G/S_EDID: OK (Not Supported)

Test output 0:

	Control ioctls:
		test VIDIOC_QUERY_EXT_CTRL/QUERYMENU: OK (Not Supported)
		test VIDIOC_QUERYCTRL: OK (Not Supported)
		test VIDIOC_G/S_CTRL: OK (Not Supported)
		test VIDIOC_G/S/TRY_EXT_CTRLS: OK (Not Supported)
		test VIDIOC_(UN)SUBSCRIBE_EVENT/DQEVENT: OK (Not Supported)
		test VIDIOC_G/S_JPEGCOMP: OK (Not Supported)
		Standard Controls: 0 Private Controls: 0

	Format ioctls:
		test VIDIOC_ENUM_FMT/FRAMESIZES/FRAMEINTERVALS: OK

Message from syslogd@raspberrypi at Dec 26 12:25:55 ...
 kernel:[ 3493.421966] Internal error: Oops: 817 [#1] ARM

Message from syslogd@raspberrypi at Dec 26 12:25:55 ...
 kernel:[ 3493.468172] Process v4l2-compliance (pid: 1736, stack limit = 0xcc708188)

Message from syslogd@raspberrypi at Dec 26 12:25:55 ...
 kernel:[ 3493.470338] Stack: (0xcc709e70 to 0xcc70a000)

Message from syslogd@raspberrypi at Dec 26 12:25:55 ...
 kernel:[ 3493.472746] 9e60:                                     be92fbc8 00000000 bf20d9ec 000000cc

Message from syslogd@raspberrypi at Dec 26 12:25:55 ...
 kernel:[ 3493.476937] 9e80: cc3ace20 be92fbc8 d5239b00 bf20d9ec cc709ec4 cc709ea0 bf20ed54 bf20d9f8

Message from syslogd@raspberrypi at Dec 26 12:25:55 ...
 kernel:[ 3493.482122] 9ea0: d50ffc00 ce237000 be92fbc8 c0cc5616 00000000 cc708000 cc709ed4 cc709ec8

Message from syslogd@raspberrypi at Dec 26 12:25:55 ...
 kernel:[ 3493.486760] 9ec0: bf2100dc bf20ec6c cc709efc cc709ed8 bf09b66c bf2100b4 bf09b5b4 be92fbc8

Message from syslogd@raspberrypi at Dec 26 12:25:55 ...
 kernel:[ 3493.492060] 9ee0: d520c2f8 ce237000 00000003 00000003 cc709f7c cc709f00 c0157a58 bf09b5c0

Message from syslogd@raspberrypi at Dec 26 12:25:55 ...
 kernel:[ 3493.496519] 9f00: c092573c 00000001 ce2376e8 cc709f18 00000002 c0025cbc 00000000 00000000

Message from syslogd@raspberrypi at Dec 26 12:25:55 ...
 kernel:[ 3493.501139] 9f20: 00000001 00000035 ce2376e0 00000002 00000000 d6f063c0 ce2376e8 00000000

Message from syslogd@raspberrypi at Dec 26 12:25:55 ...
 kernel:[ 3493.506295] 9f40: cc709f7c cc709f50 c014599c c0188068 00000000 ce237000 be92fbc8 ce237000

Message from syslogd@raspberrypi at Dec 26 12:25:55 ...
 kernel:[ 3493.511857] 9f60: c0cc5616 00000003 cc708000 00000000 cc709fa4 cc709f80 c01581d4 c01579c8

Message from syslogd@raspberrypi at Dec 26 12:25:55 ...
 kernel:[ 3493.517691] 9f80: be92fbb8 0005ce50 be930354 00000002 00000036 c000ffc4 00000000 cc709fa8

Message from syslogd@raspberrypi at Dec 26 12:25:55 ...
 kernel:[ 3493.524502] 9fa0: c000fe40 c015819c 0005ce50 be930354 00000003 c0cc5616 be92fbc8 be92fbb8

Message from syslogd@raspberrypi at Dec 26 12:25:55 ...
 kernel:[ 3493.529875] 9fc0: 0005ce50 be930354 00000002 00000036 00000001 be93050c 0005d268 be92fd4c

Message from syslogd@raspberrypi at Dec 26 12:25:55 ...
 kernel:[ 3493.535299] 9fe0: 0005d088 be92fb9c 00018300 b6b7080c 20000010 00000003 86868686 86868686

Message from syslogd@raspberrypi at Dec 26 12:25:55 ...
 kernel:[ 3493.582511] Code: ba00001d e3530002 b4c02001 d4c02001 (e4c02001)
Speicherzugriffsfehler
```

I build like it's decribed on wiki and use the same config to test it.
Pi has kernel 4.9.80.

hipersayanX, 2018-12-26:
On clean environment:

1. Open a terminal and do: **sudo dmesg -C && dmesg -w**
3. In other terminal do: **sudo insmod akvcam loglevel=7**
4. and **v4l2-compliance -d /dev/video1 -f**
5. and give me the output of dmesg.

lexusburn, 2018-12-26:
````
[ 2657.272848] [akvcam] frame.c(394): Bitmap file name not valid
[ 2657.283515] [akvcam] frame.c(394): Bitmap file name not valid
[ 2657.302006] [akvcam] driver.c(595): Virtual Devices:
[ 2657.302020] [akvcam] driver.c(596):
[ 2657.302025] [akvcam] driver.c(604): Device: /dev/video1
[ 2657.302031] [akvcam] driver.c(605): 	Description: Virtual Camera (output device)
[ 2657.302035] [akvcam] driver.c(608): 	Type: Output
[ 2657.302039] [akvcam] driver.c(613): 	Modes:
[ 2657.302043] [akvcam] driver.c(617): 		ReadWrite
[ 2657.302047] [akvcam] driver.c(620): 		MMap
[ 2657.302051] [akvcam] driver.c(623): 		UserPtr
[ 2657.302055] [akvcam] driver.c(626): 	User Controls: No
[ 2657.302060] [akvcam] driver.c(652): 	Formats:
[ 2657.302070] [akvcam] driver.c(666): 		RGB24 640x480 20/1 Hz
[ 2657.302076] [akvcam] driver.c(666): 		RGB24 640x480 15/2 Hz
[ 2657.302084] [akvcam] driver.c(666): 		YUY2 640x480 20/1 Hz
[ 2657.302090] [akvcam] driver.c(666): 		YUY2 640x480 15/2 Hz
[ 2657.302095] [akvcam] driver.c(684): 	Connections:
[ 2657.302100] [akvcam] driver.c(692): 		/dev/video2
[ 2657.302103] [akvcam] driver.c(633):
[ 2657.302107] [akvcam] driver.c(604): Device: /dev/video2
[ 2657.302112] [akvcam] driver.c(605): 	Description: Virtual Camera
[ 2657.302115] [akvcam] driver.c(610): 	Type: Capture
[ 2657.302119] [akvcam] driver.c(613): 	Modes:
[ 2657.302122] [akvcam] driver.c(617): 		ReadWrite
[ 2657.302125] [akvcam] driver.c(620): 		MMap
[ 2657.302129] [akvcam] driver.c(626): 	User Controls: No
[ 2657.302133] [akvcam] driver.c(652): 	Formats:
[ 2657.302140] [akvcam] driver.c(666): 		YUY2 1280x720 30/1 Hz
[ 2657.302146] [akvcam] driver.c(666): 		RGB24 640x480 20/1 Hz
[ 2657.302153] [akvcam] driver.c(666): 		RGB24 640x480 15/2 Hz
[ 2657.302159] [akvcam] driver.c(666): 		YUY2 640x480 20/1 Hz
[ 2657.302166] [akvcam] driver.c(666): 		YUY2 640x480 15/2 Hz
[ 2657.302170] [akvcam] driver.c(684): 	Connections:
[ 2657.302175] [akvcam] driver.c(692): 		/dev/video1
[ 2657.302178] [akvcam] driver.c(633):
[ 2657.350449] [akvcam] node.c(134): akvcam_node_open()
[ 2657.350532] [akvcam] ioctl.c(274): akvcam_ioctls_querycap()
[ 2657.351200] [akvcam] node.c(281): akvcam_node_release()
[ 2657.389660] [akvcam] node.c(134): akvcam_node_open()
[ 2657.389741] [akvcam] ioctl.c(274): akvcam_ioctls_querycap()
[ 2657.390409] [akvcam] node.c(281): akvcam_node_release()
[ 2669.115562] [akvcam] node.c(134): akvcam_node_open()
[ 2669.115714] [akvcam] ioctl.c(274): akvcam_ioctls_querycap()
[ 2669.115746] [akvcam] ioctl.c(325): akvcam_ioctls_query_ext_ctrl()
[ 2669.115784] [akvcam] ioctl.c(377): akvcam_ioctls_try_ext_ctrls()
[ 2669.115801] [akvcam] ioctl.c(393): akvcam_ioctls_queryctrl()
[ 2669.115857] [akvcam] ioctl.c(274): akvcam_ioctls_querycap()
[ 2669.133684] [akvcam] ioctl.c(274): akvcam_ioctls_querycap()
[ 2669.133941] [akvcam] node.c(134): akvcam_node_open()
[ 2669.134038] [akvcam] ioctl.c(274): akvcam_ioctls_querycap()
[ 2669.134065] [akvcam] ioctl.c(325): akvcam_ioctls_query_ext_ctrl()
[ 2669.136938] [akvcam] ioctl.c(377): akvcam_ioctls_try_ext_ctrls()
[ 2669.136986] [akvcam] ioctl.c(393): akvcam_ioctls_queryctrl()
[ 2669.138493] [akvcam] ioctl.c(274): akvcam_ioctls_querycap()
[ 2669.138708] [akvcam] ioctl.c(883): akvcam_ioctls_g_priority()
[ 2669.138729] [akvcam] ioctl.c(883): akvcam_ioctls_g_priority()
[ 2669.138745] [akvcam] ioctl.c(893): akvcam_ioctls_s_priority()
[ 2669.138758] [akvcam] ioctl.c(883): akvcam_ioctls_g_priority()
[ 2669.138769] [akvcam] ioctl.c(883): akvcam_ioctls_g_priority()
[ 2669.138781] [akvcam] ioctl.c(893): akvcam_ioctls_s_priority()
[ 2669.138795] [akvcam] ioctl.c(893): akvcam_ioctls_s_priority()
[ 2669.138807] [akvcam] ioctl.c(883): akvcam_ioctls_g_priority()
[ 2669.138818] [akvcam] ioctl.c(883): akvcam_ioctls_g_priority()
[ 2669.139730] [akvcam] node.c(134): akvcam_node_open()
[ 2669.139848] [akvcam] node.c(134): akvcam_node_open()
[ 2669.143881] [akvcam] node.c(134): akvcam_node_open()
[ 2669.144013] [akvcam] node.c(134): akvcam_node_open()
[ 2669.149009] [akvcam] node.c(134): akvcam_node_open()
[... 280 lines cut ...]
[ 2669.262965] 1e60:                                     be86cb98 00000000 bf1f09ec 000000cc
[ 2669.267136] 1e80: c98c6540 be86cb98 d581ec00 bf1f09ec c8b81ec4 c8b81ea0 bf1f1d54 bf1f09f8
[ 2669.271221] 1ea0: d58e5c00 c8b6eb40 be86cb98 c0cc5616 00000000 c8b80000 c8b81ed4 c8b81ec8
[ 2669.275317] 1ec0: bf1f30dc bf1f1c6c c8b81efc c8b81ed8 bf09b66c bf1f30b4 bf09b5b4 be86cb98
[ 2669.279541] 1ee0: d3e0ba00 c8b6eb40 00000003 00000003 c8b81f7c c8b81f00 c0157a58 bf09b5c0
[ 2669.283980] 1f00: c8b81f24 00000001 d3176508 d585a900 00000002 00000001 00000000 00000000
[ 2669.288622] 1f20: c8b81f64 00000035 d3176500 00000002 00000000 d5f15640 d3176508 00000000
[ 2669.293457] 1f40: c8b81f7c c8b81f50 c014599c c0188068 00000000 c8b6eb40 be86cb98 c8b6eb40
[ 2669.298530] 1f60: c0cc5616 00000003 c8b80000 00000000 c8b81fa4 c8b81f80 c01581d4 c01579c8
[ 2669.303713] 1f80: be86cb88 0005ce50 be86d324 00000002 00000036 c000ffc4 00000000 c8b81fa8
[ 2669.308971] 1fa0: c000fe40 c015819c 0005ce50 be86d324 00000003 c0cc5616 be86cb98 be86cb88
[ 2669.314375] 1fc0: 0005ce50 be86d324 00000002 00000036 00000001 be86d4dc 0005d268 be86cd1c
[ 2669.319824] 1fe0: 0005d088 be86cb6c 00018300 b6bd580c 20000010 00000003 00000000 00000000
[ 2669.325425] [<c031afb8>] (memset) from [<bf1f0ad8>] (akvcam_ioctls_s_parm+0xec/0x1c8 [akvcam])
[... cut ...]
[ 2669.348653] [<bf09b66c>] (v4l2_ioctl [videodev]) from [<c0157a58>] (do_vfs_ioctl+0x9c/0x7d4)
[ 2669.354180] [<c0157a58>] (do_vfs_ioctl) from [<c01581d4>] (SyS_ioctl+0x44/0x6c)
[ 2669.356994] [<c01581d4>] (SyS_ioctl) from [<c000fe40>] (ret_fast_syscall+0x0/0x1c)
[ 2669.362403] Code: ba00001d e3530002 b4c02001 d4c02001 (e4c02001)
[ 2669.396631] ---[ end trace 6c02c777338698d7 ]---
[ 2669.415020] [akvcam] node.c(281): akvcam_node_release()
[ 2669.418960] [akvcam] node.c(281): akvcam_node_release()
````

hipersayanX, 2018-12-26:
Probably [this line](https://github.com/webcamoid/akvcam/blob/8ac8e53d1e12baad146e8a40989fb7ac5338e5fa/src/ioctl.c#L792). Adding some _akpr_debug("CHECK\n")_ messages may confirm. In theory, if **param** were NULL it may not have passed the first check, and **param->parm** is an union so a reference to it can't be NULL, maybe memset is out of bounds?

hipersayanX, 2019-09-04:
No luck, my Raspberry Pi isn't working, and tried emulating with qemu but after resizing the image and the partition it won't boot.
Some Raspberry Pi user will need to take charge of this issue if they want to make the driver work there. If nobody is interested I will close this issue.

hipersayanX, 2020-11-19:
Any chance of testing latest release?

hipersayanX, 2021-01-10:
Closing as no one seems to have interest on this anymore, anyway latest release should be working fine.

closed 2021-01-10 by hipersayanX