webcamoid/akvcam #4: Can't install on Linux Mint 19

nofloda, 2019-04-04:
$ make
make -C /lib/modules/4.15.0-46-generic/build M=/home/akvcam-master/src  modules
make[1]: se entra en el directorio '/usr/src/linux-headers-4.15.0-46-generic'
Makefile:975: "Cannot use CONFIG_STACK_VALIDATION=y, please install libelf-dev, libelf-devel or elfutils-libelf-devel"
  CC [M]  /home/akvcam-master/src/module.o
  CC [M]  /home/akvcam-master/src/attributes.o
  CC [M]  /home/akvcam-master/src/buffer.o
  CC [M]  /home/akvcam-master/src/buffers.o
symbolmap: la: invalid section
  CC [M]  /home/akvcam-master/src/controls.o
  CC [M]  /home/akvcam-master/src/device.o
symbolmap: la: invalid section
  CC [M]  /home/akvcam-master/src/driver.o
  CC [M]  /home/akvcam-master/src/events.o
  CC [M]  /home/akvcam-master/src/file_read.o
  CC [M]  /home/akvcam-master/src/format.o
  CC [M]  /home/akvcam-master/src/frame.o
  CC [M]  /home/akvcam-master/src/global_deleter.o
  CC [M]  /home/akvcam-master/src/ioctl.o
  CC [M]  /home/akvcam-master/src/list.o
  CC [M]  /home/akvcam-master/src/log.o
  CC [M]  /home/akvcam-master/src/map.o
  CC [M]  /home/akvcam-master/src/mmap.o
  CC [M]  /home/akvcam-master/src/node.o
  CC [M]  /home/akvcam-master/src/object.o
  CC [M]  /home/akvcam-master/src/rbuffer.o
  CC [M]  /home/akvcam-master/src/settings.o
  CC [M]  /home/akvcam-master/src/utils.o
  LD [M]  /home/akvcam-master/src/akvcam.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /home/akvcam-master/src/akvcam.mod.o
  LD [M]  /home/akvcam-master/src/akvcam.ko
make[1]: se sale del directorio '/usr/src/linux-headers-4.15.0-46-generic'

$ sudo make install
[sudo] contraseña para ad:            
make -C /lib/modules/4.15.0-46-generic/build M=/home/akvcam-master/src modules_install
make[1]: se entra en el directorio '/usr/src/linux-headers-4.15.0-46-generic'
Makefile:975: "Cannot use CONFIG_STACK_VALIDATION=y, please install libelf-dev, libelf-devel or elfutils-libelf-devel"
  INSTALL /home/akvcam-master/src/akvcam.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:74
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:81
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  4.15.0-46-generic
make[1]: se sale del directorio '/usr/src/linux-headers-4.15.0-46-generic'

hipersayanX, 2019-04-05:
Install ```libelf-dev``` package.

nofloda, 2019-04-05:
Thanks for your response, but I've installed that package and the error is the same

> $ dpkg -s libelf-dev
> Package: libelf-dev
> Status: install ok installed
> Priority: optional
> Section: libdevel
> Installed-Size: 364
> Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
> Architecture: amd64
> Multi-Arch: same
> Source: elfutils
> Version: 0.170-0.4
> Depends: libelf1 (= 0.170-0.4), zlib1g-dev
> Conflicts: libelfg0-dev
> Description: libelf1 development libraries and header files
>  libelf1 provides a shared library which allows reading and writing of ELF
>  files on a high level.
>  .
>  This package contains development libraries and header files for libelf1.
> Original-Maintainer: Kurt Roeckx <kurt@roeckx.be>
> Homepage: https://sourceware.org/elfutils/
> 

> $ sudo make install
> make -C /lib/modules/4.15.0-47-generic/build M=/home/akvcam-master/src modules_install
> make[1]: se entra en el directorio '/usr/src/linux-headers-4.15.0-47-generic'
>   INSTALL /home/akvcam-master/src/akvcam.ko
> At main.c:160:
> - SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:74
> - SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:81
> sign-file: certs/signing_key.pem: No such file or directory
>   DEPMOD  4.15.0-47-generic
> make[1]: se sale del directorio '/usr/src/linux-headers-4.15.0-47-generic'
>

hipersayanX, 2019-04-06:
Doing a quick search it seams that it's more likely a warning in your system rater tan the driver, it may still be working fine.
Can you try loading the module? (also, just in case, save all your work before anything)

```
# mkdir -p /etc/akvcam
# cp -vf akvcam/share/config_example.ini /etc/akvcam/config.ini
# modprobe videodev
# modprobe akvcam loglevel=7
# dmesg | tail -n 100
```

nofloda, 2019-04-17:
to execute 

> # modprobe akvcam loglevel=7

modprobe: FATAL: Module akvcam not found in directory /lib/modules/4.15.0-47-generic

nofloda, 2019-04-17:
to execute
# modprobe akvcam loglevel=7
modprobe: FATAL: Module akvcam not found in directory /lib/modules/4.15.0-47-generic

upost, 2019-04-18:
same here, also on Linuxmint 19.
I think a signing process fails because:
sign-file: certs/signing_key.pem: No such file or directory

That said, I have no idea what tries to sign what and where the missing key file could be.

hipersayanX, 2019-04-19:
Will check, anyways I think this is something related to the system side rater than the driver.
Probably this links could be useful:

https://access.redhat.com/documentation/en-US/Red_Hat_Enterprise_Linux/7/html/Kernel_Administration_Guide/sect-signing-kernel-modules-for-secure-boot.html
https://blog.ubuntu.com/2017/08/11/how-to-sign-things-for-secure-boot
https://wiki.ubuntu.com/UEFI/SecureBoot
https://superdanby.github.io/Blog/signing-kernel-modules-for-secure-boot.html

Maybe someone can test some of those solutions meanwhile.

equake, 2019-05-13:
I'm having the same issue, ubuntu 18.10. Maybe it's something related to secure boot? My computer have it enabled.

Thanks.

hipersayanX, 2019-05-14:
@equake it could be, but I'm pretty sure that there must be some way to make it work there.

hdbhdb, 2019-05-23:
Hey, I got the same signing errors trying to install on Ubuntu 18.04. After looking into v4l2loopback as an alternative, I ran `sudo depmod -a`, as per their [installation instructions](https://github.com/umlaeute/v4l2loopback#install). I can't really speak to what this did, but it seems to have fix the issue for me. No errors with `modprobe akvcam` and running smoothly via webcamoid now

nofloda, 2019-05-31:
Well, the solution proposed by hdbhdb has worked perfectly. After running `sudo depmod -a` everything perfect.

hipersayanX, 2019-09-03:
Tested in Linux Mint 19.2, can't reproduce the error:

> $ sudo make install
> make -C /lib/modules/4.15.0-47-generic/build M=/home/akvcam-master/src modules_install
> make[1]: se entra en el directorio '/usr/src/linux-headers-4.15.0-47-generic'
>   INSTALL [...]

from [here](https://github.com/webcamoid/akvcam/issues/4#issuecomment-480328803). Maybe this is something related to UEFI?
Anyway I've added the solution of @hdbhdb to the wiki.

closed 2019-09-03 by hipersayanX