webcamoid/akvcam #6: Issues Displaying Streamed Content in Browsers

bigtobster, 2019-06-14:
Using Fedora 30 on 5.1.8-300.fc30.x86_64.

I have tested considerably with VLC, ffplay, Webcamoid, etc. All non-browser apps pick up /dev/video1 absolutely perfectly and display the content correctly and without corruption.

The issue for me is using with either the Firefox or Chrome (+ Chromium) web browsers.

The Chrome-based browsers detect the /dev/video1 camera and display the video at the approximately correct frame rate. However, the content is horribly distorted into various shades of pink and the video is rendered multiple times.

The Firefox browser also detecs the /dev/video1 camera but in all test cases (different websites demanding webcam) it seems it attempts to "load" it but it just doesn't ever complete. The image on screen will remain as default and no content as ever rendered. 

The method of transmission to video1 is as follows:

`ffmpeg -i myVideo.mp4 -s 640x480 -r 30 -f v4l2 -vcodec rawvideo -pix_fmt rgb24 /dev/video0`

Please do let me know which logs would be useful and I'll attempt to provide.

hipersayanX, 2019-06-14:
What is the content of _/etc/akvcam/config.ini_?

bigtobster, 2019-06-14:
```
$ cat /etc/akvcam/config.ini
[Cameras]
cameras/size = 2

cameras/1/type = output
cameras/1/mode = mmap, userptr, rw
cameras/1/description = Virtual Camera (output device)
cameras/1/formats = 2

cameras/2/type = capture
cameras/2/mode = mmap, rw
cameras/2/description = Canon XJ4123R Webcam
; cameras/2/description = Virtual Camera (capture device)
cameras/2/formats = 1, 2

[Formats]
formats/size = 2

formats/1/format = RGB24
formats/1/width = 640
formats/1/height = 480
formats/1/fps = 30

formats/2/format = RGB24, YUY2
formats/2/width = 640
formats/2/height = 480
formats/2/fps = 20/1, 15/2

[Connections]
connections/size = 1
connections/1/connection = 1:2
```

hipersayanX, 2019-06-15:
The configuration seems ok, and the width is multiple of 32, the only thing that is wrong is that FFmpeg only support **rw** method, so it must be ```cameras/2/mode = rw```.

bigtobster, 2019-06-16:
Thanks.

New settings file:
```
$ sudo cat /etc/akvcam/config.ini 
[Cameras]
cameras/size = 2

cameras/1/type = output
cameras/1/mode = mmap, userptr, rw
cameras/1/description = Virtual Camera (output device)
cameras/1/formats = 2

cameras/2/type = capture
cameras/2/mode = rw
cameras/2/description = Canon XJ4123R Webcam
; cameras/2/description = Virtual Camera (capture device)
cameras/2/formats = 1, 2

[Formats]
formats/size = 2

formats/1/format = RGB24
formats/1/width = 640
formats/1/height = 480
formats/1/fps = 30

formats/2/format = RGB24, YUY2
formats/2/width = 640
formats/2/height = 480
formats/2/fps = 20/1, 15/2

[Connections]
connections/size = 1
connections/1/connection = 1:2
```

Issue now is Permission Denied (or similar when used through ANY browser).

To reload the config, I ran the below:

```
sudo rmmod /var/lib/dkms/akvcam/5.1.8-300/5.1.8-300.fc30.x86_64/x86_64/module/akvcam.ko.xz
sudo modprobe videodev
sudo insmod /var/lib/dkms/akvcam/5.1.8-300/5.1.8-300.fc30.x86_64/x86_64/module/akvcam.ko.xz
sudo insmod /var/lib/dkms/akvcam/5.1.8-300/5.1.8-300.fc30.x86_64/x86_64/module/akvcam.ko.xz
```

Devices look like this:

```
$ ll /dev/video*                                                                              
crw-rw-rw-+ 1 root video 81, 0 Jun 16 15:14 /dev/video0
crw-rw-rw-+ 1 root video 81, 1 Jun 16 15:14 /dev/video1
```

Nothing useful in /var/log/messages

```
Jun 16 15:09:31 sephiroth kernel: akvcam: loading out-of-tree module taints kernel.
Jun 16 15:09:31 sephiroth kernel: akvcam: module verification failed: signature and/or required key missing - tainting kernel
Jun 16 15:09:31 sephiroth systemd-modules-load[992]: Inserted module 'akvcam'
```

hipersayanX, 2019-06-16:
> ```
> Jun 16 15:09:31 sephiroth kernel: akvcam: loading out-of-tree module taints kernel.
> Jun 16 15:09:31 sephiroth kernel: akvcam: module verification failed: signature and/or required key missing - tainting kernel
> Jun 16 15:09:31 sephiroth systemd-modules-load[992]: Inserted module 'akvcam'
> ```

Could that error be related to [this](https://github.com/webcamoid/akvcam/issues/4#issuecomment-495031977)?

bigtobster, 2019-06-18:
So I executed

```
sudo depmod -a
```
Rebooted. The result was this in /var/log/messages

```
Jun 18 11:59:33 sephiroth systemd-modules-load[324]: Failed to find module 'akvcam'
Jun 18 11:59:42 sephiroth systemd-modules-load[951]: Failed to find module 'akvcam'
Jun 18 11:59:42 sephiroth systemd-modules-load[1102]: Failed to find module 'akvcam'
Jun 18 11:59:42 sephiroth systemd-modules-load[1112]: Failed to find module 'akvcam'
Jun 18 11:59:44 sephiroth sh[1279]: make -C /usr/lib/modules/5.1.9-300.fc30.x86_64/build M=/var/lib/dkms/akvcam/5.1.8-300/build clean
Jun 18 11:59:46 sephiroth sh[1279]: { make -j8 KERNELRELEASE=5.1.9-300.fc30.x86_64 KERNEL_DIR=/lib/modules/5.1.9-300.fc30.x86_64/build all; } >> /var/lib/dkms/akvcam/5.1.8-300/build/make.log 2>&1
Jun 18 11:59:51 sephiroth sh[1279]: make -C /usr/lib/modules/5.1.9-300.fc30.x86_64/build M=/var/lib/dkms/akvcam/5.1.8-300/build clean
Jun 18 11:59:51 sephiroth sh[1279]:  CLEAN   /var/lib/dkms/akvcam/5.1.8-300/build/.tmp_versions
Jun 18 11:59:51 sephiroth sh[1279]:  CLEAN   /var/lib/dkms/akvcam/5.1.8-300/build/Module.symvers
Jun 18 11:59:51 sephiroth sh[1279]: akvcam.ko.xz:
Jun 18 12:00:03 sephiroth kernel: akvcam: loading out-of-tree module taints kernel.
Jun 18 12:00:03 sephiroth kernel: akvcam: module verification failed: signature and/or required key missing - tainting kernel
Jun 18 12:00:03 sephiroth systemd-modules-load[4494]: Inserted module 'akvcam'
```

So **something** definitely happened with depmod but whatever it is didn't solve the problem.

Also, the user now has the video group.

```
$ groups
toby wheel video
```

The devices are now accessed only be group and root.

```
$ ll /dev/video*
crw-rw----+ 1 root video 81, 0 Jun 18 12:00 /dev/video0
crw-rw----+ 1 root video 81, 1 Jun 18 12:00 /dev/video1
```

I've also confirmed that VLC continues to read video1 correctly when casting. This remains a Browser-Only issue both before and after depmod. VLC running as same user. 

IMO those warnings in /var/log/messages are just warnings that the module is untested/certificated on the main Kernel. It otherwise seems to be loading correctly.

bigtobster, 2019-06-28:
Sorry to bump... any thoughts on this? Happy to help test :) Just need something to try :)

hipersayanX, 2019-06-28:
Right now, no. I'm quite busy with the Android port of Webcamoid, I'll take a look of all bugs (including this) once I finish with that. Mean while you or anyone else can start hacking with the code or the system, solutions and PRs are very welcome :smiley:

bigtobster, 2019-06-30:
Understood! Many thanks!! Good luck with the Android port and I'll see what I can do with this in the meantime.

hipersayanX, 2019-09-07:
This issue was fixed in 1.0.2.

closed 2019-09-07 by hipersayanX

bigtobster, 2019-09-21:
I'm experiencing the same issue on Master branch. It's essentially Permission Denied when using akvcam on browsers but it works 100% fine on VLC etc. Any idea why this would happen with all browsers and not local apps? My feeling is that AKVCam is fine but my setup is stupid. Maybe a firewall issue? If any hints come to mind, please let me know. Otherwise I'll look into permissions stuff and post back here if I get it working.

hipersayanX, 2019-09-23:
Still the same problem? I've tested in Firefox, Chrome and Opera in Arch and it's working fine. Did you have checked you _/etc/akvcam/config.ini_ file?

hipersayanX, 2019-09-24:
Try with this _config.ini_:

```
[Cameras]
cameras\1\description=Virtual Camera (out)
cameras\1\formats=1, 2, 3, 4, 5, 6
cameras\1\mode=rw
cameras\1\type=output
cameras\2\description=Virtual Camera
cameras\2\formats=7, 8, 9, 10, 11, 12
cameras\2\mode=mmap, userptr
cameras\2\type=capture
cameras\size=2

[Connections]
connections\1\connection=1:2
connections\size=1

[Formats]
formats\size=12
formats\1\format=RGB24
formats\1\width=640
formats\1\height=480
formats\1\fps=30/1
formats\2\format=RGB24
formats\2\height=120
formats\2\width=160
formats\2\fps=30/1
formats\3\format=RGB24
formats\3\height=240
formats\3\width=320
formats\3\fps=30/1
formats\4\format=RGB24
formats\4\height=600
formats\4\width=800
formats\4\fps=30/1
formats\5\format=RGB24
formats\5\height=720
formats\5\width=1280
formats\5\fps=30/1
formats\6\format=RGB24
formats\6\height=1080
formats\6\width=1920
formats\6\fps=30/1
formats\7\format=YUY2, UYVY
formats\7\width=640
formats\7\height=480
formats\7\fps=30/1
formats\8\format=YUY2, UYVY
formats\8\width=160
formats\8\height=120
formats\8\fps=30/1
formats\9\format=YUY2, UYVY
formats\9\width=320
formats\9\height=240
formats\9\fps=30/1
formats\10\format=YUY2, UYVY
formats\10\width=800
formats\10\height=600
formats\10\fps=30/1
formats\11\format=YUY2, UYVY
formats\11\width=1280
formats\11\height=720
formats\11\fps=30/1
formats\12\format=YUY2, UYVY
formats\12\width=1920
formats\12\height=1080
formats\12\fps=30/1
```

bigtobster, 2019-09-28:
That worked perfectly. Problem solved.
For anyone else with this issue, I ran a Kernel update, restarted, rebuilt akvcam from scratch, installed with DKMS, copied the config.ini exactly as @hipersayanX wrote and 100% fixed.
Note that v4l2-compliance gave me some failures on video1 but it didn't seem to affect the overall outcome. 
Many thanks @hipersayanX - you the damn boss