webcamoid/akvcam #7: Not creating virtual devices on Fedora 30

bigtobster, 2020-01-19:
I've been using AKVcam for a while on Fedora 30 successfully with no issues. Thanks for previous support and an awesome tool.

Upgraded Kernel to 5.4.x and this triggered an issue where dkms would apparently install the module successfully but devices in /dev were not created. 

I inserted module with insmod and increased logging. Here's the dmesg output:

<pre>
[20373.175286] [akvcam] driver.c(246): Error reading formats
[20373.176280] [akvcam] frame.c(394): Bitmap file name not valid
[20373.180903] [akvcam] frame.c(394): Bitmap file name not valid
[20373.181280] ------------[ cut here ]------------
[20373.181365] WARNING: CPU: 1 PID: 17688 at drivers/media/v4l2-core/v4l2-dev.c:863 __video_register_device+0x6f0/0x1030 [videodev]
[20373.181370] Modules linked in: akvcam(OE+) rfcomm md4 nls_utf8 cifs dns_resolver fscache libdes nf_conntrack_netbios_ns nf_conntrack_broadcast xt_CT ip6t_rpfilter ip6t_REJECT nf_reject_ipv6 ipt_REJECT nf_reject_ipv4 xt_conntrack ebtable_nat ebtable_broute ip6table_nat ip6table_mangle ip6table_raw ip6table_security iptable_nat nf_nat iptable_mangle iptable_raw iptable_security nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c ip_set nfnetlink ebtable_filter ebtables ip6table_filter ip6_tables iptable_filter cmac bnep sunrpc vfat fat squashfs zstd_decompress loop iwlmvm intel_rapl_msr intel_rapl_common x86_pkg_temp_thermal intel_powerclamp snd_hda_codec_via mac80211 snd_hda_codec_generic ledtrig_audio snd_hda_codec_hdmi coretemp snd_hda_intel snd_intel_nhlt kvm_intel snd_hda_codec libarc4 btusb snd_hda_core btrtl snd_hwdep btbcm kvm iwlwifi btintel snd_seq snd_seq_device bluetooth snd_pcm irqbypass cfg80211 iTCO_wdt iTCO_vendor_support mei_hdcp intel_cstate snd_timer ecdh_generic
[20373.181438]  intel_uncore snd mei_me intel_rapl_perf ecc pcspkr rtsx_pci_ms joydev rfkill i2c_i801 soundcore lpc_ich mei ie31200_edac memstick videodev mc ip_tables dm_crypt i915 nouveau ttm crct10dif_pclmul crc32_pclmul crc32c_intel i2c_algo_bit drm_kms_helper rtsx_pci_sdmmc mmc_core mxm_wmi drm ghash_clmulni_intel serio_raw rtsx_pci r8169 wmi video fuse [last unloaded: akvcam]
[20373.181480] CPU: 1 PID: 17688 Comm: insmod Tainted: G        W  OE     5.4.10-100.fc30.x86_64 #1
[20373.181482] Hardware name: Notebook                         W35xSTQ_370ST             /W35xSTQ_370ST             , BIOS 4.6.5 11/13/2013
[20373.181549] RIP: 0010:__video_register_device+0x6f0/0x1030 [videodev]
[20373.181553] Code: 49 89 86 80 00 00 00 48 85 c0 0f 84 5b 04 00 00 41 8b 56 2c 85 d2 0f 85 2a 04 00 00 f0 41 80 8e e8 04 00 00 01 45 31 ff eb 08 <0f> 0b 41 bf ea ff ff ff 48 8b 44 24 18 65 48 33 04 25 28 00 00 00
[20373.181555] RSP: 0018:ffffae1b412bfbc0 EFLAGS: 00010246
[20373.181557] RAX: ffff8c8013ebaf58 RBX: ffffffffc13d11c0 RCX: 0000000000000001
[20373.181559] RDX: 00000000ffffffff RSI: 0000000000000000 RDI: 0000000000000000
[20373.181560] RBP: 0000000000000001 R08: ffffffffc13d11c0 R09: 0000000000000001
[20373.181561] R10: 0000000000000000 R11: ffffd46d02a70ca0 R12: 00000000ffffffff
[20373.181562] R13: ffffffffc13c461a R14: ffff8c80e62e4800 R15: ffffffffc13d1210
[20373.181563] FS:  00007f7db01a3740(0000) GS:ffff8c80e7840000(0000) knlGS:0000000000000000
[20373.181564] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[20373.181565] CR2: 00007f7daf976000 CR3: 000000021cd66002 CR4: 00000000001606e0
[20373.181568] Call Trace:
[20373.181631]  akvcam_device_register+0x65/0x80 [akvcam]
[20373.181639]  ? 0xffffffffc02df000
[20373.181673]  akvcam_driver_register+0x29/0x80 [akvcam]
[20373.181700]  ? akvcam_settings_delete+0x72/0x80 [akvcam]
[20373.181709]  akvcam_driver_init+0xea/0x170 [akvcam]
[20373.181732]  do_one_initcall+0x46/0x1f4
[20373.181746]  ? free_unref_page_commit+0x95/0x110
[20373.181763]  ? _cond_resched+0x15/0x30
[20373.181766]  ? kmem_cache_alloc_trace+0x162/0x220
[20373.181777]  ? do_init_module+0x23/0x230
[20373.181779]  do_init_module+0x5c/0x230
[20373.181782]  load_module+0x28c9/0x2b20
[20373.181788]  ? __do_sys_init_module+0x16e/0x1a0
[20373.181789]  __do_sys_init_module+0x16e/0x1a0
[20373.181802]  do_syscall_64+0x5b/0x1a0
[20373.181809]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[20373.181814] RIP: 0033:0x7f7db02cbbbe
[20373.181816] Code: 48 8b 0d cd 32 0c 00 f7 d8 64 89 01 48 83 c8 ff c3 66 2e 0f 1f 84 00 00 00 00 00 90 f3 0f 1e fa 49 89 ca b8 af 00 00 00 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d 9a 32 0c 00 f7 d8 64 89 01 48
[20373.181818] RSP: 002b:00007fff611dabb8 EFLAGS: 00000246 ORIG_RAX: 00000000000000af
[20373.181822] RAX: ffffffffffffffda RBX: 00005569d744c8a0 RCX: 00007f7db02cbbbe
[20373.181824] RDX: 00005569d744c2a0 RSI: 000000000004ce70 RDI: 00007f7daf92a010
[20373.181825] RBP: 00007f7daf92a010 R08: 00005569d744c01a R09: 0000000000000000
[20373.181826] R10: 00005569d744c010 R11: 0000000000000246 R12: 00005569d744c2a0
[20373.181827] R13: 00005569d744c2c0 R14: 00005569d744c840 R15: 00005569d744c2a0
[20373.181830] ---[ end trace afdf6c664e0bb83f ]---
[20373.181870] [akvcam] driver.c(595): Virtual Devices:
[20373.181870] [akvcam] driver.c(596): 
[20373.181871] [akvcam] driver.c(604): Device: /dev/video0
[20373.181872] [akvcam] driver.c(605): 	Description: Virtual Camera (out)
[20373.181872] [akvcam] driver.c(608): 	Type: Output
[20373.181873] [akvcam] driver.c(613): 	Modes:
[20373.181873] [akvcam] driver.c(617): 		ReadWrite
[20373.181874] [akvcam] driver.c(626): 	User Controls: No
[20373.181875] [akvcam] driver.c(652): 	Formats:
[20373.181876] [akvcam] driver.c(661): 		RGB24 640x480 30/1 Hz
[20373.181880] [akvcam] driver.c(661): 		RGB24 160x120 30/1 Hz
[20373.181883] [akvcam] driver.c(661): 		RGB24 320x240 30/1 Hz
[20373.181886] [akvcam] driver.c(661): 		RGB24 800x600 30/1 Hz
[20373.181890] [akvcam] driver.c(661): 		RGB24 1280x720 30/1 Hz
[20373.181892] [akvcam] driver.c(661): 		RGB24 1920x1080 30/1 Hz
[20373.181893] [akvcam] driver.c(679): 	Connections:
[20373.181895] [akvcam] driver.c(687): 		/dev/video0
[20373.181896] [akvcam] driver.c(633): 
[20373.181896] [akvcam] driver.c(604): Device: /dev/video0
[20373.181897] [akvcam] driver.c(605): 	Description: Virtual Camera
[20373.181898] [akvcam] driver.c(610): 	Type: Capture
[20373.181899] [akvcam] driver.c(613): 	Modes:
[20373.181900] [akvcam] driver.c(620): 		MMap
[20373.181901] [akvcam] driver.c(623): 		UserPtr
[20373.181901] [akvcam] driver.c(628): 	User Controls: Yes
[20373.181902] [akvcam] driver.c(652): 	Formats:
[20373.181906] [akvcam] driver.c(661): 		YUY2 640x480 30/1 Hz
[20373.181910] [akvcam] driver.c(661): 		UYVY 640x480 30/1 Hz
[20373.181913] [akvcam] driver.c(661): 		YUY2 160x120 30/1 Hz
[20373.181916] [akvcam] driver.c(661): 		UYVY 160x120 30/1 Hz
[20373.181920] [akvcam] driver.c(661): 		YUY2 320x240 30/1 Hz
[20373.181924] [akvcam] driver.c(661): 		UYVY 320x240 30/1 Hz
[20373.181927] [akvcam] driver.c(661): 		YUY2 800x600 30/1 Hz
[20373.181930] [akvcam] driver.c(661): 		UYVY 800x600 30/1 Hz
[20373.181932] [akvcam] driver.c(661): 		YUY2 1280x720 30/1 Hz
[20373.181934] [akvcam] driver.c(661): 		UYVY 1280x720 30/1 Hz
[20373.181935] [akvcam] driver.c(679): 	Connections:
[20373.181937] [akvcam] driver.c(687): 		/dev/video0
[20373.181938] [akvcam] driver.c(633):
</pre>

I have not modified the config.ini and the config looks good (as per #6).

hipersayanX, 2020-01-19:
Will check, probably the kernel API has changed.

xddxdd, 2020-02-17:
Please take a look at this commit: https://github.com/umlaeute/v4l2loopback/commit/b671c91478b0045984dcc83682a4302a68de2271#diff-08c7b3d1d5e914d9c938a5845a11b423

Might be the same problem.

hipersayanX, 2020-02-17:
@xddxdd Thanks! will check.

stuffinator, 2020-03-19:
Any news on this? I'm running 5.5.7-1 Manjaro and no devices in /dev are created either. 

`19.03.20 16:22	kernel	WARNING: CPU: 9 PID: 2367 at drivers/media/v4l2-core/v4l2-dev.c:885 __video_register_device+0x9b4/0x1190 [videodev]
19.03.20 16:22	kernel	Modules linked in: akvcam(OE+) uvcvideo videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common tun fuse snd_usb_audio snd_usbmidi_lib snd_rawmidi snd_seq_device input_leds xpad hid_roccat_isku mousedev hid_roccat joydev ff_memless hid_roccat_common ebtable_filter ebtables ip6table_filter ip6_tables iptable_filter rfkill edac_mce_amd snd_hda_codec_realtek snd_hda_codec_generic ledtrig_audio snd_hda_codec_hdmi amdgpu snd_hda_intel kvm nls_iso8859_1 snd_intel_dspcfg nls_cp437 snd_hda_codec vfat fat gpu_sched i2c_algo_bit snd_hda_core ttm snd_hwdep crct10dif_pclmul drm_kms_helper snd_pcm crc32_pclmul ghash_clmulni_intel drm ccp wmi_bmof r8169 snd_timer snd sp5100_tco realtek agpgart aesni_intel crypto_simd syscopyarea cryptd sysfillrect glue_helper sysimgblt i2c_piix4 pcspkr k10temp rng_core fb_sys_fops soundcore libphy wmi pinctrl_amd evdev mac_hid acpi_cpufreq uinput vboxnetflt(OE) vboxnetadp(OE) vboxdrv(OE) crypto_user videodev mc ip_tables x_tables ext4
19.03.20 16:22	kernel	 crc32c_generic crc16 mbcache jbd2 sr_mod cdrom hid_generic sd_mod usbhid hid ahci libahci crc32c_intel libata xhci_pci xhci_hcd scsi_mod vfio_pci irqbypass vfio_virqfd vfio_iommu_type1 vfio [last unloaded: akvcam]
19.03.20 16:22	kernel	CPU: 9 PID: 2367 Comm: insmod Tainted: G        W  OE     5.5.7-1-MANJARO #1
19.03.20 16:22	kernel	Hardware name: Gigabyte Technology Co., Ltd. X570 GAMING X/X570 GAMING X, BIOS F11 12/06/2019
19.03.20 16:22	kernel	RIP: 0010:__video_register_device+0x9b4/0x1190 [videodev]
19.03.20 16:22	kernel	Code: ff ff 48 89 83 80 00 00 00 48 85 c0 0f 84 0c 02 00 00 8b 53 2c 85 d2 0f 85 dc 01 00 00 f0 80 8b 08 05 00 00 01 45 31 ff eb 08 <0f> 0b 41 bf ea ff ff ff 48 8b 44 24 20 65 48 33 04 25 28 00 00 00
19.03.20 16:22	kernel	RSP: 0018:ffffbbd90201fb90 EFLAGS: 00010246
19.03.20 16:22	kernel	RAX: ffff9fd228c23358 RBX: ffff9fd228b5d800 RCX: 0000000000000000
19.03.20 16:22	kernel	RDX: 00000000ffffffff RSI: 0000000000000000 RDI: ffff9fd228b5d800
19.03.20 16:22	kernel	RBP: ffffffffc0a749c0 R08: ffffffffc0a749c0 R09: 0000000000000001
19.03.20 16:22	kernel	R10: 000000000000000a R11: f000000000000000 R12: 0000000000000001
19.03.20 16:22	kernel	R13: 00000000ffffffff R14: ffff9fd228abd000 R15: ffffffffc0a74a10
19.03.20 16:22	kernel	FS:  00007f577a31d740(0000) GS:ffff9fd58ec40000(0000) knlGS:0000000000000000
19.03.20 16:22	kernel	CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
19.03.20 16:22	kernel	CR2: 0000560e204cd930 CR3: 000000013a5e4000 CR4: 0000000000340ee0
19.03.20 16:22	kernel	Call Trace:
19.03.20 16:22	kernel	 akvcam_device_register+0x65/0x80 [akvcam]
19.03.20 16:22	kernel	 ? 0xffffffffc0373000
19.03.20 16:22	kernel	 akvcam_driver_register+0x29/0x80 [akvcam]
19.03.20 16:22	kernel	 ? akvcam_settings_delete+0x72/0x80 [akvcam]
19.03.20 16:22	kernel	 akvcam_driver_init+0xfb/0x180 [akvcam]
19.03.20 16:22	kernel	 do_one_initcall+0x59/0x240
19.03.20 16:22	kernel	 do_init_module+0x5c/0x230
19.03.20 16:22	kernel	 load_module+0x285c/0x2b20
19.03.20 16:22	kernel	 ? __do_sys_finit_module+0xbb/0x120
19.03.20 16:22	kernel	 __do_sys_finit_module+0xbb/0x120
19.03.20 16:22	kernel	 do_syscall_64+0x4e/0x150
19.03.20 16:22	kernel	 entry_SYSCALL_64_after_hwframe+0x44/0xa9
19.03.20 16:22	kernel	RIP: 0033:0x7f577a442f8d
19.03.20 16:22	kernel	Code: 00 c3 66 2e 0f 1f 84 00 00 00 00 00 90 f3 0f 1e fa 48 89 f8 48 89 f7 48 89 d6 48 89 ca 4d 89 c2 4d 89 c8 4c 8b 4c 24 08 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d d3 4e 0c 00 f7 d8 64 89 01 48
19.03.20 16:22	kernel	RSP: 002b:00007ffc1187b758 EFLAGS: 00000246 ORIG_RAX: 0000000000000139
19.03.20 16:22	kernel	RAX: ffffffffffffffda RBX: 0000560e217d57b0 RCX: 00007f577a442f8d
19.03.20 16:22	kernel	RDX: 0000000000000000 RSI: 0000560e204d7358 RDI: 0000000000000003
19.03.20 16:22	kernel	RBP: 0000000000000000 R08: 0000000000000000 R09: 00007f577a50c0c0
19.03.20 16:22	kernel	R10: 0000000000000003 R11: 0000000000000246 R12: 0000560e204d7358
19.03.20 16:22	kernel	R13: 0000000000000000 R14: 0000560e217d82e0 R15: 0000000000000000
19.03.20 16:22	kernel	---[ end trace da2588f30156e3bf ]---`

hipersayanX, 2020-03-21:
>Thanks to corona, it's prime time for pranking co-workers. I would love to get this working :D

Seriously, my mail is active 24/7 receiving new reports, because people have a lot of time now :laughing: hopefully some people have enough time send me PRs too

viking, 2020-03-24:
Same issue for me in Fedora 31:

```
[598863.892369] WARNING: CPU: 7 PID: 168613 at drivers/media/v4l2-core/v4l2-dev.c:885 __video_register_device+0x9c6/0x11b0 [videodev]
[598863.892370] Modules linked in: akvcam(OE+) uvcvideo videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common videodev xt_CHECKSUM xt_MASQUERADE nf_nat_tftp nf_conntrack_tftp tun bridge stp llc nf_conntrack_netbios_ns nf_conntrack_broadcast xt_CT ip6t_REJECT nf_reject_ipv6 ip6t_rpfilter ipt_REJECT nf_reject_ipv4 xt_conntrack ebtable_nat ebtable_broute ip6table_nat ip6table_mangle ip6table_raw ip6table_security iptable_nat nf_nat iptable_mangle iptable_raw iptable_security nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c ip_set nfnetlink ebtable_filter ebtables vboxnetadp(OE) vboxnetflt(OE) ip6table_filter ip6_tables iptable_filter vboxdrv(OE) sunrpc vfat fat intel_rapl_msr intel_rapl_common isst_if_common nvidia_drm(POE) nvidia_modeset(POE) nvidia_uvm(OE) snd_hda_codec_hdmi nfit libnvdimm x86_pkg_temp_thermal intel_powerclamp coretemp kvm_intel kvm snd_hda_codec_realtek snd_hda_codec_generic ledtrig_audio snd_hda_intel snd_intel_dspcfg nvidia(POE) irqbypass snd_usb_audio
[598863.892408]  snd_hda_codec crct10dif_pclmul crc32_pclmul snd_usbmidi_lib snd_hda_core iTCO_wdt iTCO_vendor_support ghash_clmulni_intel dell_smm_hwmon snd_hwdep drm_kms_helper intel_cstate snd_seq snd_rawmidi intel_uncore snd_seq_device dell_wmi mc drm intel_rapl_perf snd_pcm dell_smbios sparse_keymap video dcdbas mei_me snd_timer dell_wmi_descriptor ipmi_devintf wmi_bmof intel_wmi_thunderbolt ipmi_msghandler snd ioatdma soundcore i2c_i801 mei dca acpi_tad ip_tables rfkill nvme e1000e nvme_core crc32c_intel serio_raw wmi uas usb_storage fuse [last unloaded: akvcam]
[598863.892437] CPU: 7 PID: 168613 Comm: modprobe Tainted: P        W  OE     5.5.6-201.fc31.x86_64 #1
[598863.892438] Hardware name: Dell Inc. Precision 5820 Tower X-Series/02M8NY, BIOS 2.0.2 12/14/2019
[598863.892446] RIP: 0010:__video_register_device+0x9c6/0x11b0 [videodev]
[598863.892448] Code: 64 fe 48 89 83 80 00 00 00 48 85 c0 0f 84 0c 02 00 00 8b 53 2c 85 d2 0f 85 dc 01 00 00 f0 80 8b 10 05 00 00 01 45 31 ff eb 08 <0f> 0b 41 bf ea ff ff ff 48 8b 44 24 20 65 48 33 04 25 28 00 00 00
[598863.892449] RSP: 0018:ffffad35c8b37bb8 EFLAGS: 00010246
[598863.892451] RAX: ffff9d3675b50158 RBX: ffff9d3768850000 RCX: 0000000000000000
[598863.892452] RDX: 00000000ffffffff RSI: 0000000000000000 RDI: ffff9d3768850000
[598863.892453] RBP: ffffffffc20b01c0 R08: ffffffffc20b01c0 R09: 0000000000000001
[598863.892454] R10: 0000000000000000 R11: ffffd9e81fdb09a0 R12: 0000000000000001
[598863.892455] R13: 00000000ffffffff R14: ffff9d3c49351900 R15: ffffffffc20b0210
[598863.892457] FS:  00007f79e980c740(0000) GS:ffff9d3cbffc0000(0000) knlGS:0000000000000000
[598863.892458] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[598863.892459] CR2: 00007f79e8dbc000 CR3: 000000081e11e001 CR4: 00000000003606e0
[598863.892460] DR0: 0000000000000000 DR1: 0000000000000000 DR2: 0000000000000000
[598863.892461] DR3: 0000000000000000 DR6: 00000000fffe0ff0 DR7: 0000000000000400
[598863.892462] Call Trace:
[598863.892482]  akvcam_device_register+0x65/0x80 [akvcam]
[598863.892485]  ? 0xffffffffc1ff5000
[598863.892495]  akvcam_driver_register+0x29/0x80 [akvcam]
[598863.892504]  ? akvcam_settings_delete+0x72/0x80 [akvcam]
[598863.892513]  akvcam_driver_init+0xea/0x170 [akvcam]
[598863.892521]  do_one_initcall+0x46/0x200
[598863.892527]  ? free_unref_page_commit+0x95/0x110
[598863.892532]  ? _cond_resched+0x15/0x30
[598863.892535]  ? kmem_cache_alloc_trace+0x162/0x220
[598863.892538]  ? do_init_module+0x23/0x230
[598863.892541]  do_init_module+0x5c/0x230
[598863.892543]  load_module+0x28c2/0x2b20
[598863.892549]  ? __do_sys_init_module+0x16e/0x1a0
[598863.892551]  __do_sys_init_module+0x16e/0x1a0
[598863.892555]  do_syscall_64+0x5b/0x1c0
[598863.892558]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[598863.892561] RIP: 0033:0x7f79e993c12e
[598863.892563] Code: 48 8b 0d 5d fd 0b 00 f7 d8 64 89 01 48 83 c8 ff c3 66 2e 0f 1f 84 00 00 00 00 00 90 f3 0f 1e fa 49 89 ca b8 af 00 00 00 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d 2a fd 0b 00 f7 d8 64 89 01 48
[598863.892564] RSP: 002b:00007ffc5a0e5d18 EFLAGS: 00000246 ORIG_RAX: 00000000000000af
[598863.892566] RAX: ffffffffffffffda RBX: 000056242015bf60 RCX: 00007f79e993c12e
[598863.892567] RDX: 000056241f436358 RSI: 000000000004ce70 RDI: 00007f79e8d70010
[598863.892568] RBP: 00007f79e8d70010 R08: 0000000000000000 R09: 00007f79e8d70000
[598863.892569] R10: 0000000000000001 R11: 0000000000000246 R12: 000056241f436358
[598863.892569] R13: 0000000000000000 R14: 000056242015beb0 R15: 000056242015bf60
```

hipersayanX, 2020-03-25:
Can someone check if 931473c fix this problem?

stuffinator, 2020-03-25:
> Can someone check if [931473c](https://github.com/webcamoid/akvcam/commit/931473c6d6eded1d6b54d3e322dca8b7d16841ff) fix this problem?

I can confirm that it's working on 5.5.7-1 Manjaro :grin:

laenco, 2020-03-25:
5.5.10 / 5.5.11 archlinux - works as well.
Thanks!

viking, 2020-03-25:
The module now loads on Fedora 31, however, it's impossible to `rmmod`. Here's my `lsmod` output:

```
Module                  Size  Used by
akvcam                155648  1
xt_CHECKSUM            16384  1
xt_MASQUERADE          20480  3
nf_nat_tftp            16384  0
nf_conntrack_tftp      20480  3 nf_nat_tftp
tun                    57344  1
bridge                208896  0
stp                    16384  1 bridge
llc                    16384  2 bridge,stp
nf_conntrack_netbios_ns    16384  1
nf_conntrack_broadcast    16384  1 nf_conntrack_netbios_ns
xt_CT                  16384  3
ip6t_REJECT            16384  12
nf_reject_ipv6         20480  1 ip6t_REJECT
ip6t_rpfilter          16384  1
ipt_REJECT             16384  5
nf_reject_ipv4         16384  1 ipt_REJECT
xt_conntrack           16384  38
ebtable_nat            16384  1
ebtable_broute         16384  1
ip6table_nat           16384  1
ip6table_mangle        16384  1
ip6table_raw           16384  1
ip6table_security      16384  1
iptable_nat            16384  1
nf_nat                 53248  4 ip6table_nat,nf_nat_tftp,iptable_nat,xt_MASQUERADE
iptable_mangle         16384  1
iptable_raw            16384  1
iptable_security       16384  1
nf_conntrack          163840  8 xt_conntrack,nf_nat,nf_conntrack_tftp,nf_conntrack_netbios_ns,nf_nat_tftp,nf_conntrack_broadcast,xt_CT,xt_MASQUERADE
nf_defrag_ipv6         24576  1 nf_conntrack
nf_defrag_ipv4         16384  1 nf_conntrack
libcrc32c              16384  2 nf_conntrack,nf_nat
ip_set                 53248  0
nfnetlink              16384  1 ip_set
ebtable_filter         16384  1
ebtables               40960  3 ebtable_nat,ebtable_filter,ebtable_broute
ip6table_filter        16384  1
ip6_tables             36864  7 ip6table_filter,ip6table_raw,ip6table_nat,ip6table_mangle,ip6table_security
iptable_filter         16384  1
vboxnetadp             28672  0
vboxnetflt             32768  0
vboxdrv               557056  2 vboxnetadp,vboxnetflt
sunrpc                491520  1
vfat                   20480  1
fat                    86016  1 vfat
intel_rapl_msr         20480  0
intel_rapl_common      32768  1 intel_rapl_msr
isst_if_common         16384  0
nvidia_drm             57344  6
nvidia_modeset       1118208  10 nvidia_drm
nvidia_uvm           1081344  0
nfit                   69632  0
libnvdimm             196608  1 nfit
x86_pkg_temp_thermal    20480  0
intel_powerclamp       20480  0
coretemp               20480  0
snd_hda_codec_hdmi     73728  1
kvm_intel             311296  0
snd_hda_codec_realtek   126976  1
snd_hda_codec_generic    94208  1 snd_hda_codec_realtek
kvm                   802816  1 kvm_intel
ledtrig_audio          16384  2 snd_hda_codec_generic,snd_hda_codec_realtek
nvidia              20467712  449 nvidia_uvm,nvidia_modeset
snd_hda_intel          53248  10
snd_intel_dspcfg       28672  1 snd_hda_intel
iTCO_wdt               16384  0
irqbypass              16384  1 kvm
snd_hda_codec         163840  4 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec_realtek
iTCO_vendor_support    16384  1 iTCO_wdt
crct10dif_pclmul       16384  1
snd_usb_audio         290816  8
drm_kms_helper        233472  1 nvidia_drm
dell_smm_hwmon         24576  0
crc32_pclmul           16384  0
snd_hda_core          102400  5 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec,snd_hda_codec_realtek
snd_usbmidi_lib        40960  1 snd_usb_audio
ghash_clmulni_intel    16384  0
uvcvideo              114688  1
snd_hwdep              16384  2 snd_usb_audio,snd_hda_codec
intel_cstate           20480  0
snd_rawmidi            45056  1 snd_usbmidi_lib
videobuf2_vmalloc      20480  1 uvcvideo
videobuf2_memops       20480  1 videobuf2_vmalloc
intel_uncore          147456  0
snd_seq                86016  0
videobuf2_v4l2         28672  1 uvcvideo
snd_seq_device         16384  2 snd_seq,snd_rawmidi
intel_rapl_perf        20480  0
videobuf2_common       57344  2 videobuf2_v4l2,uvcvideo
drm                   585728  9 drm_kms_helper,nvidia_drm
snd_pcm               126976  5 snd_hda_codec_hdmi,snd_hda_intel,snd_usb_audio,snd_hda_codec,snd_hda_core
videodev              266240  6 videobuf2_v4l2,uvcvideo,videobuf2_common,akvcam
dell_wmi               20480  0
dell_smbios            32768  1 dell_wmi
snd_timer              40960  2 snd_seq,snd_pcm
dcdbas                 20480  1 dell_smbios
sparse_keymap          16384  1 dell_wmi
snd                   102400  49 snd_hda_codec_generic,snd_seq,snd_seq_device,snd_hda_codec_hdmi,snd_hwdep,snd_hda_intel,snd_usb_audio,snd_usbmidi_lib,snd_hda_codec,snd_hda_codec_realtek,snd_timer,snd_pcm,snd_rawmidi
video                  53248  1 dell_wmi
wmi_bmof               16384  0
mc                     61440  5 videodev,snd_usb_audio,videobuf2_v4l2,uvcvideo,videobuf2_common
mei_me                 45056  0
ipmi_devintf           20480  0
intel_wmi_thunderbolt    20480  0
dell_wmi_descriptor    20480  2 dell_wmi,dell_smbios
ipmi_msghandler       118784  2 ipmi_devintf,nvidia
i2c_i801               32768  0
soundcore              16384  1 snd
mei                   122880  1 mei_me
ioatdma                61440  0
dca                    16384  1 ioatdma
acpi_tad               16384  0
ip_tables              32768  5 iptable_filter,iptable_security,iptable_raw,iptable_nat,iptable_mangle
rfkill                 28672  2
nvme                   49152  3
e1000e                290816  0
serio_raw              20480  0
nvme_core             110592  5 nvme
crc32c_intel           24576  5
wmi                    36864  5 intel_wmi_thunderbolt,dell_wmi,wmi_bmof,dell_smbios,dell_wmi_descriptor
uas                    28672  0
usb_storage            77824  1 uas
fuse                  139264  5
```

It says akvcam is in use, but doesn't list what it's being used by. Running `rmmod -f akvcam` outputs:

```
rmmod: ERROR: could not remove 'akvcam': Resource temporarily unavailable
rmmod: ERROR: could not remove module akvcam: Resource temporarily unavailable
```

hipersayanX, 2020-03-25:
@viking maybe some program is using the module, [this](https://superuser.com/questions/396731/which-process-is-using-a-linux-kernel-module) could help.

I will make a new release later.

viking, 2020-03-25:
Thanks, you're right. It looks like a service on my machine monitors the creation of /dev nodes and starts using the newly created /dev/videoX node that akvcam creates.

hipersayanX, 2020-03-27:
Issue fixed in 1.0.3, closing.

closed 2020-03-27 by hipersayanX