webcamoid/akvcam #13: timeval kernel 5.6.2.arch1-2

jpenguin, 2020-04-06:
I'm having the exact same issue as tyis person on the AUR; cI'll add that I was able to build v4l2loopback-dkms-git, but not v4l2loopback-dkms

https://aur.archlinux.org/packages/akvcam-dkms/#comment-737888

> Hi,
> 
> I was able to install akvcam-dkms 1.0.3-1 under kernel 5.5.13-arch2-1. However, both akvcam-dkms and akvcam-dkms-git fail to build under 5.6.2.arch1-2. Error output:
> 
> /home/blah/.cache/yay/akvcam-dkms-git/src/akvcam/src/buffers.c: In function ‘akvcam_buffers_dequeue’: /home/blah/.cache/yay/akvcam-dkms-git/src/akvcam/src/buffers.c:488:30: error: passing argument 1 of ‘akvcam_get_timestamp’ from incompatible pointer type [-Werror=incompatible-pointer-types] 488 | akvcam_get_timestamp(&v4l2_buff->timestamp); | ^~~~~~~~~~~~~~~~~~~~~ | | | struct __kernel_v4l2_timeval * In file included from /home/blah/.cache/yay/akvcam-dkms-git/src/akvcam/src/buffers.h:28, from /home/blah/.cache/yay/akvcam-dkms-git/src/akvcam/src/buffers.c:28: /home/blah/.cache/yay/akvcam-dkms-git/src/akvcam/src/utils.h:101:43: note: expected ‘struct timeval ’ but argument is of type ‘struct __kernel_v4l2_timeval ’ 101 | void akvcam_get_timestamp(struct timeval tv); | ~~~~~~~~~~~~~~~~^~ /home/blah/.cache/yay/akvcam-dkms-git/src/akvcam/src/buffers.c: In function ‘akvcam_buffers_write_frame’: /home/blah/.cache/yay/akvcam-dkms-git/src/akvcam/src/buffers.c:729:34: error: passing argument 1 of ‘akvcam_get_timestamp’ from incompatible pointer type [-Werror=incompatible-pointer-types] 729 | akvcam_get_timestamp(&v4l2_buff->timestamp); | ^~~~~~~~~~~~~~~~~~~~~ | | | struct __kernel_v4l2_timeval * In file included from /home/blah/.cache/yay/akvcam-dkms-git/src/akvcam/src/buffers.h:28, from /home/blah/.cache/yay/akvcam-dkms-git/src/akvcam/src/buffers.c:28: /home/blah/.cache/yay/akvcam-dkms-git/src/akvcam/src/utils.h:101:43: note: expected ‘struct timeval ’ but argument is of type ‘struct __kernel_v4l2_timeval ’ 101 | void akvcam_get_timestamp(struct timeval tv); | ~~~~~~~~~~~~~~~~^~ cc1: some warnings being treated as errors make[2]: *** [scripts/Makefile.build:268: /home/blah/.cache/yay/akvcam-dkms-git/src/akvcam/src/buffers.o] Error 1
> 
> Something changed in the kernel code related to struct timeval? I haven't tried to figure out the details of what's gone wrong.

illuusio, 2020-04-07:
I made a patch for this and it can be found on:
https://build.opensuse.org/package/view_file/home:illuusio/akvcam/akvcam-compile_kernel_5_6_0.diff
Problem is that kernel has changed all the inner things to 64bits to avoid 2038 problem.

poldi1405, 2020-04-07:
> https://build.opensuse.org/package/view_file/home:illuusio/akvcam/akvcam-compile_kernel_5_6_0.diff
Link Broken

illuusio, 2020-04-07:
@poldi1405 Sorry just re-arranged. It's back online again.

fuzunspm, 2020-04-07:
I'm getting the error below during linux kernel update to 5.6.2

`==> dkms install akvcam/1.0.3 -k 5.6.2-arch1-2
Error! Bad return status for module build on kernel: 5.6.2-arch1-2 (x86_64)`

hipersayanX, 2020-04-07:
I've updated the driver, but Arch is still using linux-api-headers 5.4.

illuusio, 2020-04-08:
Compiles with 5.6.0 at least for me

hipersayanX, 2020-04-12:
Fixed this in the last release.

closed 2020-04-12 by hipersayanX