webcamoid/akvcam #14: Driver not working after reboot

ricoderks, 2020-04-09:
Hi all,

First, thanks for the great work you did!!

I followed the wiki for building and installing akvcam and I got it to work. All your examples worked. Also set the module to load on boot. But now it doesn't work anymore. Checking compliance and properties works, both give no errors.  The cat and echo example goes wrong. 

`cat /dev/urandom > /dev/video1` gives: 
`cat: write error: Invalid argument`

I'am on Ubuntu 18.04.4.
My `lsmod | grep videodev` output:

```
videodev              184320  4 videobuf2_core,videobuf2_v4l2,uvcvideo,akvcam
media                  40960  2 videodev,uvcvideo
```
`dmesg | grep akvcam` output:

```
[    7.560406] akvcam: loading out-of-tree module taints kernel.
[    7.560448] akvcam: module verification failed: signature and/or required key missing - tainting kernel
```

Can you help me with this?

Cheers, Rico

ricoderks, 2020-04-10:
I appologize! Apperently the numbers of the devices changed. Just after install:

- my regular webcam /dev/video0
- output device /dev/video1
- capture device /dev/video2

After reboot:

- my regular webcam /dev/video0
- output device /dev/video0
- capture device /dev/video1

It works, but is this not a bit strange?

Cheers, Rico

hipersayanX, 2020-04-12:
>It works, but is this not a bit strange?

The numbers are assigned by the kernel, the driver has not any control of that. It's normal.
Your first question is solved?

ricoderks, 2020-04-14:
Ok, thanks for the answer. My first question is solved! Again, thanks for the great work and stay healthy!

Cheers, Rico

closed 2020-04-14 by ricoderks