webcamoid/akvcam #15: Missing crypto bio on ubuntu 20.04

BryanDollery, 2020-04-11:
I ran into this problem while running `make install`:

$ sudo make install
```sh
make -C /usr/lib/modules/5.4.0-21-generic/build M=/home/bryan/git/webcamoid/akvcam/src modules_install
make[1]: Entering directory '/usr/src/linux-headers-5.4.0-21-generic'
  INSTALL /home/bryan/git/webcamoid/akvcam/src/akvcam.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:69
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:76
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  5.4.0-21-generic
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-21-generic'
```

hipersayanX, 2020-04-12:
Some people reported that running ```sudo depmod -a``` may  fix the problem.

JanekNaymspace, 2020-05-03:
I've got the same error in Ubuntu 18.04 and `sudo depmod -a` does not resolve this.

Sadly I don't know how to provide any helpful information.

drarko, 2020-06-02:
Same issue here

hipersayanX, 2020-06-03:
According to [this](https://stackoverflow.com/questions/42693016/getting-ssl-error-while-installing-intel-driver), try [installing with dkms](https://github.com/webcamoid/akvcam/wiki/Build-and-install#managing-the-module-with-dkms) instead.

JanekNaymspace, 2020-06-07:
Worked for me, thanks

hipersayanX, 2020-06-27:
Well, it seem that [this](https://github.com/webcamoid/akvcam/issues/15#issuecomment-638261405) was the solution, closing.

closed 2020-06-27 by hipersayanX