webcamoid/akvcam #16: confusing documentation

magnets99, 2020-04-11:
Hi,
Thanks for all your hard work. I'm a little confused by the documentation here

https://github.com/webcamoid/akvcam/wiki/Configure-the-cameras

which states

_An Output device can be connected to many Capture devices, but only one Output device can be connected to a Capture device._

i'm not sure if i'm just being thick but i can't see the difference between these two contradictory sentences.

flurpo, 2020-04-12:
I'm a bit of a neubee with this and have a little confusion at the same page.

sudo touch /etc/akvcam/config.ini - note that I have no such directory in /etc after installation. I can see the driver in /lib/modules and the installation notes look OK.....Am I supposed to create this directory and ini file manually?

Best

magnets99, 2020-04-12:
i did create the directory and used nano to make the ini file, i copied all the parts from the instructions and when  i did
ls /lib/modules/$(uname -r)/extra/akvcam.ko*
I got

/lib/modules/5.3.0-40-generic/extra/akvcam.ko

so i think it's working, thought i'm trying to get webcamoid to switch to akvcam from v4l2loopback without success.

hipersayanX, 2020-04-12:
I've  updated the wiki, I hope it's better explained now.
@flurpo yes you must create the path, also added the example file to the wiki.
@magnets99 the idea was that you can't connect the devices in the opposite way, my bad.
Also, in Webcamoid you have two options, either:

1. Just [install the driver](https://github.com/webcamoid/akvcam/wiki/Build-and-install) and let Webcamoid configure it. The configuration is persistent. Or
2. Uninstall the driver, and pass the **--vcam** option with the path to _akvcam/src_ or v4l2loopback sources, to Webcamoid and it will compile, load, and configure the module for you. It won't install the the driver, so if you reboot, all your virtual devices will dissapear.

closed 2020-06-27 by hipersayanX