webcamoid/akvcam #17: Is there an equivalent to v4l2loopback's "video_nr" driver parameter?

fangfufu, 2020-04-13:
In v4l2loopback, there is a driver parameter called ``video_nr``, which specifies the device-node minor number. Is it possible to specify this in akvcam? I basically want to stop akvcam to occupy /dev/video0 and /dev/video1.

hipersayanX, 2020-04-16:
No, I prefer to let the kernel decide which number corresponds to each device, this way it avoids collisions with other drivers.

fangfufu, 2020-04-16:
But that way it messes up the order of my webcams. :(

hipersayanX, 2020-04-16:
Not sure if I want to do that way, I think it will cause more problems than solutions, let me think it.

vStone, 2020-04-23:
You could resolve this by creating a fixed symlink using udev matching the `camera\#\description` with `ENV{ID_V4L_PRODUCT}`

hipersayanX, 2020-05-23:
At the end I'll have to implement this, this could be useful in the future.

fangfufu, 2020-05-23:
> At the end I'll have to implement this, this could be useful in the future.

How comes?

hipersayanX, 2020-05-24:
@fangfufu 

>How comes?

I need the devices to be predictable so I can simplify my [waitForDevices function](https://github.com/webcamoid/webcamoid/blob/90a7d3e01b3bd513aeb12d70aea101c395e5b36a/libAvKys/Plugins/VirtualCamera/src/v4l2sys/VCamIPC/src/ipcbridge.cpp#L2328), instead of waiting for all v4l2 devices created by all drivers and then waiting for the same number of devices to appear, I can wait for the specific devices to appear. I did not figured out at beginning :sweat_smile:

hipersayanX, 2020-06-12:
I've implemented this in the latest commit.

fangfufu, 2020-06-12:
Cool! I will close the issue for now. :)

closed 2020-06-12 by fangfufu

StefanBruens, 2020-07-01:
instead of polling for the devices, you should listen to the relevant udev "ADD" events.