webcamoid/akvcam #19: Crash of the driver - Unknown pixelformat 0x00000000

tbrandysky, 2020-04-17:
This is what I do:

- install driver using DKMS method (everything goes well)
- start ./webcamoid-8.7.1-x86_64.AppImage 
- create output device
- start Teams and select the previously created video device
- I can see the video in a preview from the webcamoid device but the driver crashes
- I am not able to see video stream from the webcamoid virtual device anymore

Apr 17 10:34:56 HP-ProBook-440-G6 kernel: [ 5426.477009] ------------[ cut here ]------------
Apr 17 10:34:56 HP-ProBook-440-G6 kernel: [ 5426.477013] Unknown pixelformat 0x00000000
Apr 17 10:34:56 HP-ProBook-440-G6 kernel: [ 5426.477048] WARNING: CPU: 1 PID: 32211 at /build/linux-UDHJtG/linux-4.15.0/drivers/media/v4l2-core/v4l2-ioctl.c:1293 v4l_fill_fmtdesc+0xf4d/0x1260 [videodev]
Apr 17 10:34:56 HP-ProBook-440-G6 kernel: [ 5426.477050] Modules linked in: akvcam(OE) ccm rfcomm vboxnetadp(OE) vboxnetflt(OE) vboxdrv(OE) bnep hid_multitouch intel_rapl x86_pkg_temp_thermal intel_powerclamp coretemp kvm_intel kvm irqbypass snd_soc_skl snd_soc_skl_ipc snd_hda_codec_hdmi intel_cstate snd_hda_ext_core snd_soc_sst_dsp intel_rapl_perf nls_iso8859_1 snd_soc_sst_ipc arc4 snd_soc_acpi snd_soc_core snd_hda_codec_realtek snd_compress hp_wmi ac97_bus snd_hda_codec_generic sparse_keymap serio_raw wmi_bmof snd_pcm_dmaengine snd_hda_intel snd_hda_codec snd_hda_core snd_hwdep uvcvideo snd_pcm videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 snd_seq_midi snd_seq_midi_event snd_rawmidi videobuf2_core btusb joydev btrtl snd_seq btbcm videodev btintel iwlmvm bluetooth snd_seq_device media snd_timer input_leds snd mac80211 ecdh_generic soundcore
Apr 17 10:34:56 HP-ProBook-440-G6 kernel: [ 5426.477078]  shpchp iwlwifi ucsi_acpi typec_ucsi typec cfg80211 idma64 int3403_thermal virt_dma mei_me processor_thermal_device int340x_thermal_zone hp_accel mei intel_lpss_pci intel_soc_dts_iosf lis3lv02d mac_hid input_polldev acpi_pad intel_lpss intel_pch_thermal hp_wireless int3400_thermal acpi_thermal_rel sch_fq_codel parport_pc ppdev lp parport ip_tables x_tables autofs4 btrfs xor zstd_compress raid6_pq algif_skcipher af_alg hid_logitech_hidpp hid_logitech_dj hid_generic usbhid dm_crypt dm_mirror dm_region_hash dm_log crct10dif_pclmul crc32_pclmul ghash_clmulni_intel pcbc aesni_intel i915 aes_x86_64 crypto_simd glue_helper cryptd i2c_algo_bit drm_kms_helper nvme psmouse nvme_core syscopyarea sysfillrect sysimgblt ahci fb_sys_fops r8169 sdhci_pci drm libahci i2c_hid mii sdhci hid wmi video [last unloaded: akvcam]
Apr 17 10:34:56 HP-ProBook-440-G6 kernel: [ 5426.477114] CPU: 1 PID: 32211 Comm: AudioThread Tainted: G      D W  OE    4.15.0-96-generic #97-Ubuntu
Apr 17 10:34:56 HP-ProBook-440-G6 kernel: [ 5426.477115] Hardware name: HP HP ProBook 440 G6/8537, BIOS R71 Ver. 01.08.03 07/17/2019
Apr 17 10:34:56 HP-ProBook-440-G6 kernel: [ 5426.477124] RIP: 0010:v4l_fill_fmtdesc+0xf4d/0x1260 [videodev]
Apr 17 10:34:56 HP-ProBook-440-G6 kernel: [ 5426.477126] RSP: 0018:ffffafcf82a9fc30 EFLAGS: 00010282
Apr 17 10:34:56 HP-ProBook-440-G6 kernel: [ 5426.477128] RAX: 0000000000000000 RBX: ffffafcf82a9fd68 RCX: 0000000000000006
Apr 17 10:34:56 HP-ProBook-440-G6 kernel: [ 5426.477129] RDX: 0000000000000007 RSI: 0000000000000092 RDI: ffff9850e7456490
Apr 17 10:34:56 HP-ProBook-440-G6 kernel: [ 5426.477131] RBP: ffffafcf82a9fc40 R08: 0000000000001298 R09: 000000000000000f
Apr 17 10:34:56 HP-ProBook-440-G6 kernel: [ 5426.477132] R10: ffff984f9efe7b40 R11: ffffffff9b15c80d R12: ffff985008e8bc00
Apr 17 10:34:56 HP-ProBook-440-G6 kernel: [ 5426.477133] R13: ffffffffc0e38860 R14: ffff984f9efe7b40 R15: ffff985008e8bc00
Apr 17 10:34:56 HP-ProBook-440-G6 kernel: [ 5426.477135] FS:  00007f3e71ffb700(0000) GS:ffff9850e7440000(0000) knlGS:0000000000000000
Apr 17 10:34:56 HP-ProBook-440-G6 kernel: [ 5426.477137] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Apr 17 10:34:56 HP-ProBook-440-G6 kernel: [ 5426.477138] CR2: 00007f3e5800ad98 CR3: 00000003498e2005 CR4: 00000000003606e0
Apr 17 10:34:56 HP-ProBook-440-G6 kernel: [ 5426.477140] Call Trace:
Apr 17 10:34:56 HP-ProBook-440-G6 kernel: [ 5426.477149]  v4l_enum_fmt+0x7d/0x130 [videodev]
Apr 17 10:34:56 HP-ProBook-440-G6 kernel: [ 5426.477155]  __video_do_ioctl+0x370/0x380 [videodev]
Apr 17 10:34:56 HP-ProBook-440-G6 kernel: [ 5426.477159]  ? terminate_walk+0x90/0x100
Apr 17 10:34:56 HP-ProBook-440-G6 kernel: [ 5426.477166]  video_usercopy+0x24d/0x640 [videodev]
Apr 17 10:34:56 HP-ProBook-440-G6 kernel: [ 5426.477170]  ? video_ioctl2+0x20/0x20 [videodev]
Apr 17 10:34:56 HP-ProBook-440-G6 kernel: [ 5426.477175]  video_ioctl2+0x15/0x20 [videodev]
Apr 17 10:34:56 HP-ProBook-440-G6 kernel: [ 5426.477179]  v4l2_ioctl+0xc8/0xf0 [videodev]
Apr 17 10:34:56 HP-ProBook-440-G6 kernel: [ 5426.477181]  do_vfs_ioctl+0xa8/0x630
Apr 17 10:34:56 HP-ProBook-440-G6 kernel: [ 5426.477183]  ? putname+0x4c/0x60
Apr 17 10:34:56 HP-ProBook-440-G6 kernel: [ 5426.477185]  ? do_sys_open+0x13d/0x2c0
Apr 17 10:34:56 HP-ProBook-440-G6 kernel: [ 5426.477187]  SyS_ioctl+0x79/0x90
Apr 17 10:34:56 HP-ProBook-440-G6 kernel: [ 5426.477190]  do_syscall_64+0x73/0x130
Apr 17 10:34:56 HP-ProBook-440-G6 kernel: [ 5426.477193]  entry_SYSCALL_64_after_hwframe+0x3d/0xa2
Apr 17 10:34:56 HP-ProBook-440-G6 kernel: [ 5426.477195] RIP: 0033:0x7f3ecd5845d7
Apr 17 10:34:56 HP-ProBook-440-G6 kernel: [ 5426.477196] RSP: 002b:00007f3e71ffa508 EFLAGS: 00000246 ORIG_RAX: 0000000000000010
Apr 17 10:34:56 HP-ProBook-440-G6 kernel: [ 5426.477198] RAX: ffffffffffffffda RBX: 00007f3e580054f0 RCX: 00007f3ecd5845d7
Apr 17 10:34:56 HP-ProBook-440-G6 kernel: [ 5426.477199] RDX: 00007f3e71ffa530 RSI: ffffffffc0405602 RDI: 000000000000009c
Apr 17 10:34:56 HP-ProBook-440-G6 kernel: [ 5426.477200] RBP: 00007f3e71ffa5a0 R08: 0000000000000004 R09: 0000000000000000
Apr 17 10:34:56 HP-ProBook-440-G6 kernel: [ 5426.477201] R10: 00007f3e580008d0 R11: 0000000000000246 R12: 00007f3e71ffa530
Apr 17 10:34:56 HP-ProBook-440-G6 kernel: [ 5426.477202] R13: 00007f3e71ffa518 R14: 000000000000009c R15: 0000555a252215f0
Apr 17 10:34:56 HP-ProBook-440-G6 kernel: [ 5426.477204] Code: 30 0f 84 b4 01 00 00 3d 4a 4c 32 30 0f 84 97 01 00 00 3d 4d 33 31 30 0f 84 7a 01 00 00 89 c6 48 c7 c7 f7 de dd c0 e8 e3 5d ac d8 <0f> 0b 80 7b 0c 00 0f 85 d6 f3 ff ff 8b 43 2c 48 c7 c1 9e d2 dd 
Apr 17 10:34:56 HP-ProBook-440-G6 kernel: [ 5426.477218] ---[ end trace 6e11abb96fc5a0b1 ]---

hipersayanX, 2020-06-26:
This error is not reproducible, the call trace does not show nothing related to akvcam, it only shows a segfault in v4l2 driver, any more information or I can close this issue?

tbrandysky, 2020-06-27:
You can close the ticket.

thank you

Tomas

closed 2020-06-27 by hipersayanX