webcamoid/akvcam #22: akvcam module fails with kernel 5.7

darkonen, 2020-04-22:
akvcam module fails with kernel 5.7 
v4l2 API have some changes (VFL_TYPE_GRABBER  => VFL_TYPE_VIDEO)
here a patch to solve it
[akvcam.zip](https://github.com/webcamoid/akvcam/files/4516662/akvcam.zip)

darkonen

dx9s, 2020-05-12:
Not sure if this is related.. but recently the akvcam driver has been failing on load and I think there was a kernel update recently.. I recompiled from latest git and still same:

>$ sudo modprobe akvcam 
[sudo] password for dx: 
Killed
$ dmesg
[...]
[   50.385782] media: Linux media interface: v0.10
[   50.390278] videodev: Linux video capture interface: v2.00
[   50.393883] BUG: unable to handle kernel NULL pointer dereference at 0000000000000000
[   50.393886] PGD 0 P4D 0 
[   50.393888] Oops: 0000 [#1] PREEMPT SMP PTI
[   50.393890] CPU: 4 PID: 3451 Comm: modprobe Tainted: P           OE     4.18.0-13-lowlatency #14~18.04.1-Ubuntu
[   50.393891] Hardware name: System76 Wild Dog Pro/Z370 HD3P-CF, BIOS F5 Z5 12/15/2017
[   50.393897] RIP: 0010:akvcam_driver_read_device+0x46/0x240 [akvcam]

dx9s, 2020-05-12:
the above was caused by some issue I could not find in the config.ini .. started from fresh config.ini one "output" to one "capture" and it now works.. still trying to find a cause, predictable it hard to pin down.

hipersayanX, 2020-05-17:
5.7 has not arrived to Arch repositories yet, will check once it arrives to the repository.

WyRe, 2020-06-12:
Is this patch going to be merged to upstream in a future?

hipersayanX, 2020-06-12:
Test latest commit.

WyRe, 2020-06-12:
@hipersayanX I'm trying to compile it again from your [AUR package](https://aur.archlinux.org/packages/akvcam-dkms-git/) but apparently the latest commit in my local is 'Package updated'

hipersayanX, 2020-06-12:
@WyRe try again.

darkonen, 2020-06-13:
I's OK
module with this commit works with kernel 5.7.2

closed 2020-06-13 by darkonen