webcamoid/akvcam #23: Cannot compile with kernel 5.6.3 (Funtoo 1.4)

MoritzMaxeiner, 2020-04-25:
When trying to compile either `master` or `1.0.4` on Funtoo Linux 1.4 with Linux kernel 5.6.3 and gcc 9.2.0 (Gentoo kernel sources) I get this issue:

```
$ make
make -C /lib64/modules/5.6.3-gentoo/build M=/home/mm/tools/akvcam/src  modules
make[1]: Entering directory '/usr/src/linux-5.6.3-gentoo'
  MODPOST 1 modules
ERROR: "video_device_release_empty" [/home/mm/tools/akvcam/src/akvcam.ko] undefined!
ERROR: "video_device_release" [/home/mm/tools/akvcam/src/akvcam.ko] undefined!
ERROR: "v4l2_device_unregister" [/home/mm/tools/akvcam/src/akvcam.ko] undefined!
ERROR: "__video_register_device" [/home/mm/tools/akvcam/src/akvcam.ko] undefined!
ERROR: "v4l2_device_register" [/home/mm/tools/akvcam/src/akvcam.ko] undefined!
ERROR: "video_device_alloc" [/home/mm/tools/akvcam/src/akvcam.ko] undefined!
ERROR: "video_unregister_device" [/home/mm/tools/akvcam/src/akvcam.ko] undefined!
make[2]: *** [scripts/Makefile.modpost:93: __modpost] Error 1
make[1]: *** [Makefile:1596: modules] Error 2
make[1]: Leaving directory '/usr/src/linux-5.6.3-gentoo'
make: *** [Makefile:42: all] Error 2
```

I've attached the kernel config as [config.txt](https://github.com/webcamoid/akvcam/files/4533012/config.txt), since I suspect I may be missing some required setting.

hipersayanX, 2020-04-25:
Use ```sudo modprobe videodev``` before loading the module.

https://github.com/webcamoid/akvcam/wiki/Usage-and-examples

MoritzMaxeiner, 2020-04-26:
Thank you, but that does not change the compilation error:

```
src $ sudo modprobe videodev
src $ sudo lsmod | grep video
uvcvideo              114688  0
videobuf2_vmalloc      20480  1 uvcvideo
videobuf2_memops       20480  1 videobuf2_vmalloc
videobuf2_v4l2         28672  1 uvcvideo
videobuf2_common       57344  2 videobuf2_v4l2,uvcvideo
videodev              249856  3 videobuf2_v4l2,uvcvideo,videobuf2_common
mc                     57344  5 videodev,snd_usb_audio,videobuf2_v4l2,uvcvideo,videobuf2_common
video                  53248  3 dell_wmi,dell_laptop,i915
src $ make
make -C /lib64/modules/5.6.3-gentoo/build M=/home/mm/tools/akvcam/src  modules
make[1]: Entering directory '/usr/src/linux-5.6.3-gentoo'
  MODPOST 1 modules
ERROR: "video_device_release_empty" [/home/mm/tools/akvcam/src/akvcam.ko] undefined!
ERROR: "video_device_release" [/home/mm/tools/akvcam/src/akvcam.ko] undefined!
ERROR: "v4l2_device_unregister" [/home/mm/tools/akvcam/src/akvcam.ko] undefined!
ERROR: "__video_register_device" [/home/mm/tools/akvcam/src/akvcam.ko] undefined!
ERROR: "v4l2_device_register" [/home/mm/tools/akvcam/src/akvcam.ko] undefined!
ERROR: "video_device_alloc" [/home/mm/tools/akvcam/src/akvcam.ko] undefined!
ERROR: "video_unregister_device" [/home/mm/tools/akvcam/src/akvcam.ko] undefined!
make[2]: *** [scripts/Makefile.modpost:93: __modpost] Error 1
make[1]: *** [Makefile:1596: modules] Error 2
make[1]: Leaving directory '/usr/src/linux-5.6.3-gentoo'
make: *** [Makefile:42: all] Error 2
```

Also, if it is a requirement to have that module loaded for compiling the driver, would you consider adding that to https://github.com/webcamoid/akvcam/wiki/Build-and-install ?

hipersayanX, 2020-04-27:
I'm using linux 5.6.7 in Arch and the module is compiling and working fine, maybe you are using a custom kernel or have disabled some feature? all those functions (video_device_release_empty, video_device_release, v4l2_device_unregister, etc) are defined in the kernel.

>Also, if it is a requirement to have that module loaded for compiling this module, would you consider adding that information to https://github.com/webcamoid/akvcam/wiki/Build-and-install ?

Is not a requirement for compiling, but imagined that could be a solution for your problem.

MoritzMaxeiner, 2020-04-27:
> I'm using linux 5.6.7 in Arch and the module is compiling and working fine, maybe you are using a custom kernel or have disabled some feature?

I'm using a custom kernel (originally a debian kernel config without drivers I don't need + a few specializations), which is why I attached my kernel config, as I hoped you might know what kernel requirements your kernel module has.

I'll try to find out if it's a missing kernel option, then, and update here should I find it.

hipersayanX, 2020-04-30:
>I'll try to find out if it's a missing kernel option, then, and update here should I find it.

Yes please, I've not any experience compiling the kernel, so I can't give any tip or help there.

MoritzMaxeiner, 2020-05-03:
Alright, so it turns out the issue was strange, if straight forward: My `Module.symvers` did, for whatever reason, not contain all the symbols it should. The file is generated by `make modules` and *should* also be updated by that step, but wasn't in my case until I deleted it manually and it created a new one with all the correct symbols, after which I could compile the module withour problems.

closed 2020-05-03 by MoritzMaxeiner

hipersayanX, 2020-05-05:
Ok, thanks for the update.