webcamoid/akvcam #24: Questions / not having much luck

dx9s, 2020-04-28:
Not having luck, perhaps more information in Wiki that can answer the following collection of random thoughts:

Does akvcam scale/resize?

I just want to send 720p or 1080p video to akvcam and whatever resolution is taken from "capture" wants to pull, it gets (downscalled), but I've been fighting all day on different INIs.

It's not clear if the order is important (output must before capture in that order), I tried changing the order and perhaps missed something but it didn't work.

output (is what any v4l2sink / ffmpeg / etc SHOULD be sending to)

capture is what other programs will consume (if it sees a valid webcam and stream)

in:
[Connections]
connections/size = 1
connections/1/connection = 1:2

(connections/n/connection = ...) is n important ? is n effect the /dev/videoX (X ~= n )?

So far I've only gotten OBS to write to /dev/video0 (output???) using YUY2 which should not be possible. The output was 640x480 and wrong frame rate creating drift from input to output.

quoted https://github.com/webcamoid/akvcam/wiki/Configure-the-cameras
'simple straight forward INI like configuration'

dx9s, 2020-04-28:
[Cameras]

\# /dev/videoN
cameras/1/type = output
cameras/1/mode = mmap, userptr, rw
cameras/1/description = Virtual Camera (output device)
cameras/1/formats = 1, 2

\# /dev/videoN+1
cameras/2/type = capture
cameras/2/mode = mmap, userptr, rw
cameras/2/description = Virtual Camera
cameras/2/formats = 1, 2, 3, 4

cameras/size = 2

[Formats]
\# formats for "output" (into akvcam)
formats/1/format = RGB24, YUY2
formats/1/width = 640
formats/1/height = 480
formats/1/fps = 30/1

formats/2/format = RGB24, YUY2
formats/2/width = 1280
formats/2/height = 720
formats/2/fps = 30/1

\# formats for "capture" (output from akvcam)
formats/3/format = YUY2
formats/3/width = 640
formats/3/height = 360
formats/3/fps = 30/1

formats/4/format = YUY2
formats/4/width = 1280
formats/4/height = 720
formats/4/fps = 30/1

formats/size = 4

[Connections]
connections/size = 1
connections/1/connection = 1:2

dx9s, 2020-04-28:
If I don't include the same formats used in the input (type=output) that the output (type=capture) uses.. it only creates on dev (the type=output I think)

hipersayanX, 2020-05-01:
>Does akvcam scale/resize?

Yes, if you send a 720p frame to the driver it will downscale or upscale to the frame size selected in the target program.

>(connections/n/connection = ...) is n important ? is n effect the /dev/videoX (X ~= n )?

**connections** is an array, arrays are defined as:

```
array/size = n
array/1/some_key = some_value
array/2/some_key = some_value
array/3/some_key = some_value
...
array/n/some_key = some_value
```

Arrays are 1-indexed. That why I'm also linking to [QSettings file format](https://doc.qt.io/qt-5/qsettings.html), because it follows more or less the same format, [check this](https://doc.qt.io/qt-5/qsettings.html#beginWriteArray).  
In ```connections/1/connection = 1:2```, in the value 1:2, the first number always represents the index of an output device, the device where you send the frames, all other numbers coming after represents the index of the capture devices from where you get the frames, the virtual camera.
For example in the **[Cameras]** group you have the **cameras** array, index 1 is _output_, index 2 is _capture_, and you are connecting them in the *connections* array.

>It's not clear if the order is important (output must before capture in that order)

The order of the elements in an array does not matters, what matters is how you define the devices and how you connect them.
Also in your _config.ini_:

```
[Cameras]

# /dev/videoN
cameras/1/type = output
cameras/1/mode = mmap, userptr, rw
cameras/1/description = Virtual Camera (output device)
cameras/1/formats = 1, 2

[Formats]
# formats for "output" (into akvcam)
formats/1/format = RGB24, YUY2
formats/1/width = 640
formats/1/height = 480
formats/1/fps = 30/1

formats/2/format = RGB24, YUY2
formats/2/width = 1280
formats/2/height = 720
formats/2/fps = 30/1
```

Outputs does not supports YUV formats yet, it only supports RGB24 and BGR24, [check this](https://github.com/webcamoid/akvcam/wiki/Configure-the-cameras#defining-the-formats).

dx9s, 2020-05-03:
Okay.. I guess the use of "output" needs to be more clear.  If I hear you correctly:

* 'camera/X/type = output' can have 'camera/X/formats = .." that container either RGB or YUV
* 'camera/X/type = capture' can have 'camera/X/formats = .." only supports RGB

(or is it the other way around) 

Output to another program -or- output from a program to akvcam  :smirk:

--Doug 'dx9s'

P.S.> thanks for taking the time to clarify.. I am testing now.. I got something to work that appears to contradict the (check this link) what color formats are supported where.

dx9s, 2020-05-03:
P.S.^2> This appears to work, is it correct or is akvcam assuming RGB someplace where I put YUV?

[Cameras]

# /dev/videoN
cameras/1/type = output
cameras/1/mode = mmap, userptr, rw
cameras/1/description = Virtual Camera (output device)
# cameras/1/formats = 1, 2
cameras/1/formats = 2

# /dev/videoN+1
cameras/2/type = capture
cameras/2/mode = mmap, userptr, rw
cameras/2/description = Virtual Camera
# cameras/2/formats = 1, 2, 3, 4
cameras/2/formats =  4

cameras/size = 2

[Formats]
# formats for "output" (into akvcam)
formats/1/format = RGB24, YUY2
formats/1/width = 640
formats/1/height = 480
formats/1/fps = 30/1

formats/2/format = RGB24, YUY2
formats/2/width = 1280
formats/2/height = 720
formats/2/fps = 30/1

# formats for "capture" (output from akvcam)
formats/3/format = YUY2
formats/3/width = 640
formats/3/height = 360
formats/3/fps = 30/1

formats/4/format = YUY2
formats/4/width = 1280
formats/4/height = 720
formats/4/fps = 30/1

formats/size = 4

[Connections]
connections/size = 1
connections/1/connection = 1:2

dx9s, 2020-05-03:
P.S.>^2 This works.. did I get the RGB24 in the correct place or is it assuming something RGB:

[Cameras]
cameras/size = 2

\# /dev/videoN
cameras/1/type = output
cameras/1/mode = mmap, userptr, rw
cameras/1/description = Virtual Camera (output device)
cameras/1/formats = 1

\# /dev/videoN+1
cameras/2/type = capture
cameras/2/mode = mmap, userptr, rw
cameras/2/description = Virtual Camera 1
cameras/2/formats =  3

[Formats]
formats/size = 3
\# formats for "output" (into akvcam)
formats/1/format = RGB24, YUY2
formats/1/width = 1280
formats/1/height = 720
formats/1/fps = 30/1

\# formats for "capture" (output from akvcam)
formats/2/format = RGB24
formats/2/width = 640
formats/2/height = 360
formats/2/fps = 30/1

formats/3/format = RGB24
formats/3/width = 1280
formats/3/height = 720
formats/3/fps = 30/1

[Connections]
connections/size = 1
connections/1/connection = 1:2

hipersayanX, 2020-05-05:
Just in case, take in mind that when I'm talking about **capture** device or **output** device, I'm talking in the same terms as V4L2 documentation:

https://www.kernel.org/doc/html/latest/media/uapi/v4l/dev-capture.html
https://www.kernel.org/doc/html/latest/media/uapi/v4l/dev-output.html

With that said

>Okay.. I guess the use of "output" needs to be more clear. If I hear you correctly:
>
>    'camera/X/type = output' can have 'camera/X/formats = .." that container either RGB or YUV
>    'camera/X/type = capture' can have 'camera/X/formats = .." only supports RGB
>
>(or is it the other way around)

The other way around. 

```
Output: RGB
Capture: RGB and YUV 
```

dx9s, 2020-05-11:
@hipersayanX  thanks for your feedback and support.
Think it would be fair to share the FINAL config below that works as expected but weird observation/comment below that:

> [Cameras]
> \# /dev/videoN
> cameras/1/type = output
> cameras/1/mode = mmap, userptr, rw
> cameras/1/description = VCam1(input)
> cameras/1/formats = 1, 2
> 
> \# /dev/videoN+1
> cameras/2/type = capture
> cameras/2/mode = mmap, userptr, rw
> cameras/2/description = VCam1(720p)
> cameras/2/formats =  3
> 
> \# /dev/videoN+2
> cameras/3/type = capture
> cameras/3/mode = mmap, userptr, rw
> cameras/3/description = VCam1(360p)
> cameras/3/formats =  4
> 
> cameras/size = 3
> 
> [Formats]
> \# formats for "output" (into akvcam)
> formats/1/format = RGB24, YUY2
> formats/1/width = 1280
> formats/1/height = 720
> formats/1/fps = 30/1, 20/1, 15/1, 10/1, 15/2, 5/1, 1/1
> 
> formats/2/format = RGB24, YUY2
> formats/2/width = 640
> formats/2/height = 360
> formats/2/fps = 30/1, 20/1, 15/1, 10/1, 15/2, 5/1, 1/1
> 
> \# formats for "capture" (output from akvcam)
> formats/3/format = RGB24
> formats/3/width = 1280
> formats/3/height = 720
> formats/3/fps = 30/1, 20/1, 15/1, 10/1, 15/2, 5/1, 1/1
> 
> formats/4/format = RGB24
> formats/4/width = 640
> formats/4/height = 360
> formats/4/fps = 30/1, 20/1, 15/1, 10/1, 15/2, 5/1, 1/1
> 
> formats/size = 4
> 
> [Connections]
> connections/size = 1
> connections/1/connection = 1:2:3

At one point I had the 'capture' cameras formats to include the input format:
> cameras/2/formats =  1,3
> [...]
> cameras/3/formats =  2,4

as one program didn't work without it. But I've since removed it and can't figure out which program it was (perhaps is was either in my brain or my config file, either way, above seems to work)

However sometimes have to restart the module for some programs to access more than once. Think it might have to do with the camera mode:
> camera/N/mode = mmap, userptr, rw

Perhaps there is a better mode for certain programs (chromium I think it was in my case)

--Doug

baradhili, 2020-06-10:
so if I had ffmpeg sending dvcam data from /dev/fw0 to /dev/video2 then my config would have 
cameras/2/type = capture
cameras/2/mode = mmap, userptr, rw
cameras/2/description = VCam1(720p)
cameras/2/formats = 3

?

hipersayanX, 2020-06-27:
Question seems to be solved, closing.

closed 2020-06-27 by hipersayanX