webcamoid/akvcam #26: [DOC] Example on how to integrate with OpenCV

lucabelluccini, 2020-05-16:
Hello,
I am interested in writing a video stream from OpenCV to `akvcam`.
Is it possible? Would you share a simple code snippet?

hipersayanX, 2020-05-17:
>Is it possible?

Yes, of course, akvcam devices are just normal v4l2 devices and implements the same [API](https://www.kernel.org/doc/html/latest/media/uapi/v4l/v4l2.html).

>Would you share a simple code snippet?

V4L2 is not simple at all, and a "simple" example would require some thousands lines of code. Instead you can check this links:

https://stackoverflow.com/questions/36426826/v4l2-python-streaming-video-mapping-buffers
https://www.kernel.org/doc/html/latest/media/uapi/v4l/capture.c.html
https://github.com/webcamoid/webcamoid/blob/master/libAvKys/Plugins/VirtualCamera/src/v4l2sys/VCamIPC/src/ipcbridge.cpp#L1997

and use that as a reference for interfacing with OpenCV.

lucabelluccini, 2020-05-18:
Thank you for your prompt feedback!
I was checking another project, which relies on v4l2 loopback (https://github.com/jremmons/pyfakewebcam/blob/master/README.md) which exposes a simple API.

hipersayanX, 2020-05-19:
I'll write an example later, but I've not time for that now.

lucabelluccini, 2020-05-19:
@hipersayanX Do not worry! I do not think this is a priority!
I'll try to help if I have time on next weekends.

fangfufu, 2020-05-23:
Is there any ways to get pyfakewebcam to work with akvcam? That would be convenient.

hipersayanX, 2020-05-24:
@fangfufu yes of course, the driver use an standard v4l2 API, any program can interface with it following the API properly, but I did not found any documented example for the output API. The output API is not much different of the capture one.
You can still use the [IPC code](https://github.com/webcamoid/webcamoid/blob/90a7d3e01b3bd513aeb12d70aea101c395e5b36a/libAvKys/Plugins/VirtualCamera/src/v4l2sys/VCamIPC/src/ipcbridge.cpp#L873) as a reference, but right now I'm quite busy with other things to provide a simpler example.

drjerk1, 2020-06-01:
# Wrote a simple python class that is able to send images to akvcam, camera needs to be setup in YUY2 format

from fcntl import ioctl
import v4l2
import os
import threading
from queue import Queue
import cv2
import numpy as np

class CameraWriter:
    def open_yuyv_camera(self):
        d = os.open(self.webcam, os.O_RDWR)
        cap = v4l2.v4l2_capability()
        ioctl(d, v4l2.VIDIOC_QUERYCAP, cap)
        vid_format = v4l2.v4l2_format()
        vid_format.type = v4l2.V4L2_BUF_TYPE_VIDEO_OUTPUT
        vid_format.fmt.pix.width = self.width
        vid_format.fmt.pix.height = self.height
        vid_format.fmt.pix.pixelformat = v4l2.V4L2_PIX_FMT_YUYV
        vid_format.fmt.pix.field = v4l2.V4L2_FIELD_NONE
        vid_format.fmt.pix.colorspace = v4l2.V4L2_COLORSPACE_SRGB
        ioctl(d, v4l2.VIDIOC_S_FMT, vid_format)
        return d

    def writer_thread(self):
        def rgb2yuyv(image):
            img_yuv = cv2.cvtColor(image, cv2.COLOR_RGB2YUV)
            y0 = np.expand_dims(img_yuv[...,0][::,::2], axis=2)
            u = np.expand_dims(img_yuv[...,1][::,::2], axis=2)
            y1 = np.expand_dims(img_yuv[...,0][::,1::2], axis=2)
            v = np.expand_dims(img_yuv[...,2][::,::2], axis=2)
            img_yuyv = np.concatenate((y0, u, y1, v), axis=2)
            img_yuyv_cvt = img_yuyv.reshape(img_yuyv.shape[0], img_yuyv.shape[1] * 2, int(img_yuyv.shape[2] / 2))
            return img_yuyv_cvt

        while True:
            elem = self.queue.get()
            if elem is None:
                break
            image_data = rgb2yuyv(cv2.resize(elem, (self.width, self.height))).tobytes()
            try:
                os.write(self.d, image_data)
            except Exception:
                break

    def __init__(self, webcam, width, height):
        self.webcam = webcam
        self.width = width
        self.height = height
        self.d = self.open_yuyv_camera()
        self.queue = Queue(maxsize=1)
        self.thread = threading.Thread(target=self.writer_thread)
        self.thread.start()

    def write(self, image):
        self.queue.put(image)

    def close(self):
        self.queue.put(None)
        os.close(self.d)
if __name__ == "__main__":
       camera_w, camera_h = 1280, 720
       image1, image2, image3 = ... #np.array(w, h, 3, dtype=np.uint8)
      writer = CameraWriter("/dev/video1", camera_w, camera_h)
      writer.write(image1)
      time.sleep(0.5)
      writer.write(image2)
      time.sleep(0.5)
      writer.write(image3)
      time.sleep(0.5)
      writer.close()

fangfufu, 2020-06-01:
@drjerk1 , thanks! :)

hipersayanX, 2020-06-01:
@drjerk1 Thanks! :smile:

fangfufu, 2020-06-03:
@drjerk1, after ``modprobe akvcam``, I get 2 devices, let's call them ``/dev/video3`` and ``/dev/video4``, which device should I point your script to? 

Webcam viewing software can only open ``/dev/video4``. If I get your script to write to ``/dev/video4``, it throws the following error:
```
Traceback (most recent call last):
  File "akvcam.py", line 66, in <module>
    writer = AkvCameraWriter("/dev/video4", camera_w, camera_h)
  File "akvcam.py", line 15, in __init__
    self.d = self.open_yuyv_camera()
  File "akvcam.py", line 31, in open_yuyv_camera
    ioctl(d, v4l2.VIDIOC_S_FMT, vid_format)
OSError: [Errno 22] Invalid argument
```

If I write to ``/dev/video3``, the script just gets stuck. If I then open ``/dev/video4``, it just shows akvcam's default screen. 

By the way, I changed your main function to:
```
if __name__ == "__main__":
    camera_w, camera_h = 1280, 720
    writer = AkvCameraWriter("/dev/video4", camera_w, camera_h)
    while True:
        image = np.random.randint(0, 255, [camera_w, camera_h, 3], dtype=np.uint8)
        writer.write(image)
        time.sleep(0.5)
    writer.close()
```
My ``/dev/video2`` is the v4l2loopback device, and surprisingly enough it works fine with your script. With the current script, it just shows random noise which gets refreshed once every 0.5 sec.

fangfufu, 2020-06-03:
Ah never mind, it turns out that you need to configure ``/etc/akvcam/config.ini`` to enable ``rw`` on the output camera. More specifically, I have the following:
```
[Cameras]
cameras\1\description=akvcam (out)
cameras\1\formats=1, 2, 3, 4, 5, 6
cameras\1\mode=mmap, userptr, rw
cameras\1\type=output
cameras\2\description=akvcam
cameras\2\formats=7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28,$
cameras\2\mode=mmap, userptr
cameras\2\type=capture
cameras\size=2
```

The script provided by @drjerk1 actually works equally badly with pyfakewebcam. I get crazy images like this: 
[image]

It is meant to look like this: 
[image]

If anyone knows what's going on, please do let me know.

fangfufu, 2020-06-03:
Okay, I figured out why, the default ``/etc/akvcam/config.ini`` is not compatible with pyfakewebcam.

drjerk1, 2020-06-03:
@fangfufu that crazy image is probably caused with inappropriate camera format, you need to setup both input and output camera in yuy2 format

fangfufu, 2020-06-03:
@drjerk1, I figured out, it is not honouring the following line for some reason. 
```
vid_format.fmt.pix.pixelformat = v4l2.V4L2_PIX_FMT_YUYV
```
Then again, the python v4l2 package doesn't work on my installation for some reason... so I am using the v4l2 shipped with pyfakewebcam. 

So I am holding off firing a bug report with respect to that for now - I am not sure if it is akvcam's problem. 

The final version of my script looks like this: https://github.com/fangfufu/Linux-Fake-Background-Webcam/blob/master/fakecam/akvcam.py, I skipped all the pixel format conversion, and it works fine.

drjerk1, 2020-06-03:
@fangfufu but keep in mind that yuy2 format is needed for more compability, ex skype and zoom do not work with rgb24 cameras, but yuy2 works fine

fangfufu, 2020-06-03:
@drjerk1, the output format is handled automatically by akvcam. I am just changing the input format. I just tested it, it works fine in both Skype and Zoom. I just checked. I am just writing up my documentation now.

hipersayanX, 2020-06-03:
@fangfufu 

> Webcam viewing software can only open `/dev/video4`. If I get your script to write to `/dev/video4`, it throws the following error:

Before using a V4L2 you must check if it is an output or capture device, using VIDIOC_QUERYCAP:

```
v4l2_capability capability;
ioctl(fd, VIDIOC_QUERYCAP, &capability)
```

then you check if **capability.capabilities** has V4L2_CAP_VIDEO_OUTPUT or V4L2_CAP_VIDEO_OUTPUT_MPLANE, if the device has those flags you can write the frames there.

>If I write to /dev/video3, the script just gets stuck. If I then open /dev/video4, it just shows akvcam's default screen.
>My /dev/video2 is the v4l2loopback device, and surprisingly enough it works fine with your script. With the current script, it just shows random noise which gets refreshed once every 0.5 sec.

You already solved it, but I will explain anyways, the driver works in 3 possible modes: Read/Write,  Streaming I/O (Memory Mapping) and Streaming I/O (User Pointers), you can check [the documentation](https://www.kernel.org/doc/html/latest/media/uapi/v4l/io.html), [this script](https://github.com/webcamoid/akvcam/issues/26#issuecomment-636848904) only works with Read/Write, so if you don't configure the output as **rw** it won't work.
The same happens with v4l2loopback and all programs using it like FFmpeg and GStreamer, only works as a Read/Write device.
Webcamoid instead can work with the 3 modes.

>The script provided by @drjerk1 actually works equally badly with pyfakewebcam. I get crazy images like this:

Because you must write the frames as RGB24 (issue #27), and then the frames will be converted to YUY2 or whatever format in the other side.

You must always remember that akvcam unlike v4l2loopback, needs at least 2 devices, one for reading and the other for writing, the formats and operating modes for each one are independent, and the driver will convert the frames from one side to the format required by the other side.
v4l2loopback instead is just a passthrough driver, it just creates one unique device for reading and writing, and it will copy the frames as is to the other side.

fangfufu, 2020-06-03:
Thanks, @hipersayanX, so can akvcam accept frames in YUY2 at all? I meant, does it simply ignore ``vid_format.fmt.pix.pixelformat = v4l2.V4L2_PIX_FMT_YUYV``?

hipersayanX, 2020-06-03:
@fangfufu The output can receive RGB24 only, the capture can offer YUY2.

allo-, 2020-06-03:
For me it works to set both akvcam devices to format `YUY2` when streaming using `pyfakewebcam`.

pyfakewebcam converts from RGB to YUY2 itself anyway, but it seems that for some reason the kernel module assumes the frame is RGB24.

drjerk1, 2020-06-03:
@fangfufu yep but i configured both devices to yuy2 format, your config where conversion is handled by akvcam seems to be more clear way to do that

fangfufu, 2020-06-03:
> @fangfufu The output can receive RGB24 only, the capture can offer YUY2.

@hipersayanX , it seems that what @drjerk1 and @allo- reported contradict what you said. Could it be that the colour space / pixel format conversion code in akvcam isn't working properly?

drjerk1, 2020-06-03:
@fangfufu both input and output cameras configured to yuy2 so probably no conversion performed

allo-, 2020-06-03:
@drjerk1 pyfakewebcam already converts RGB images to YUY2:

https://github.com/jremmons/pyfakewebcam/blob/41abbab42917a1f83eef04a1438284dfe43092f6/pyfakewebcam/pyfakewebcam.py#L80

drjerk1, 2020-06-03:
@allo-, yep it seems so, but it's better to use queue and threading because os.write blocks thread while image would not be shown by camera (~1/fps secs)

fangfufu, 2020-06-03:
Anyway, @drjerk1, @allo- , this is my config.ini: 
https://github.com/fangfufu/Linux-Fake-Background-Webcam/blob/master/fakecam/akvcam/config.ini

As you can probably see, I definitely did specify YUY2 as an accepted input/output format. However akvcam definitely did not like it as an input format.

allo-, 2020-06-03:
I can only tell what worked.
Your config looks quite long and I do not really know much of akvcam, yet.

The example config contained `format = RGB24, YUY2` and did not work, when I removed `RGB24` it did work. So maybe you need to force the input to be YUY2 instead of only offering the format.

I could configure the correct format and using different resolution works as well. I did not test with different frame rates, and I am not sure about how frame rates different from the input device are handled. I guess akvcam does something more sophisticated than v4l2loopback that just does not return anything when there is no new frame.

fangfufu, 2020-06-06:
@hipersayanX, I really think this should be classified as a bug...

hipersayanX, 2020-06-07:
@allo- 

>pyfakewebcam converts from RGB to YUY2 itself anyway, but it seems that for some reason the kernel module assumes the frame is RGB24.

If I remember well, akvcam does not put any restriction on using other than RGB24 or BGR24 format for the output, if you are sending a YUY2 640x480 frame and the client expects the same format without any modification in between (like for example modifying the contrast of the image) the driver will just make a pass-through, but if you are sending a format different than expected or a modified version it will reject the frame and send a black frame.

@fangfufu 

>@hipersayanX , it seems that what @drjerk1 and @allo- reported contradict what you said. 

Same as above.

>@hipersayanX, I really think this should be classified as a bug...

But the original topic was about providing a code example on how to use the virtual camera in other programs, and the topic diverged to discuss how to properly use the driver, and all "bugs" found here are more like an improper use of the driver.
Instead, what we can discuss in a new issue is about adding support for more output formats.

allo-, 2020-06-07:
@hipersayanX I am not sure about the scope of this issue and if there is a bug or feature request.

@fangfufu and me have two variants of a fake webcam with a neural network for segmentation, that we want to use to convert the image from a real webcam and send it to a virtual webcam driver. I am not sure what other features are needed and what are the restrictions of the approach of passing through YUY2 frames.
Are there issues when using another approach proves to be more reliably (e.g. supporting more types of webcams or more output formats needed by different programs?

As far as I currently see, both our programs import BGR24 frames using opencv and export YUY2 frames to a fake webcam driver, either converted in pyfakewebcam or using the code snippet from @drjerk1. I did not analyse the full operation of constructing the frames in a slightly different way, but it looks like it could be mathematically equivalent.

So the thing what I wonders about is, will a program that outputs YUY2 cause any problems or be limited in some way by the driver? When not I do not see much of an issue _here_. The remaining question for @fangfufu and me is, if the code above (and mainining it ourself) may be better than using the (non-maintained) pyfakewebcam module.

> pyfakewebcam converts from RGB to YUY2 itself anyway, but it seems that for some reason the kernel module assumes the frame is RGB24.

My issue here is only, that the default configuration from the wiki (and @fangfufu config generated by webcamoid) seem to create a device, that can receive RGB24 and YUY2 frames, but passing a YUY2 frame is not auto-detected. I do not know, if it should be or if there are some setup calls missing in pyfakewebcam and I suppose it is just okay to setup akvcam to use YUY2 input and YUY2 output. At least it works for me and as long as you don't tell me "You use the driver in an unsupported mode that may stop working in the future", I am fine with it.

hipersayanX, 2020-06-07:
@allo- 

>At least it works for me and as long as you don't tell me "You use the driver in an unsupported mode that may stop working in the future", I am fine with it.

No, don't worry about that. Meanwhile just send RGB24 frames, I'm already planning to add new output formats in future versions.

fangfufu, 2020-06-07:
@allo- , just for the record, I am sending RGB24 frames to akvcam.

hipersayanX, 2020-06-14:
Added an usage example in C in the latest commit.

hipersayanX, 2020-06-27:
C example must be enough to understand how it works, other developers can start hacking their own solution for other languages and programs. Closing.

closed 2020-06-27 by hipersayanX