webcamoid/akvcam #27: How to properly configure video resolution / format

skuro, 2020-05-22:
I'm trying to use an `akvcam` device as a tool to stream over e.g. Google Meet by composing my scene with OBS. While I'm able to actually stream some video through the virtual camera, the output looks weird with the scene being replicated multiple times and in grayscale. 

Here's what I mean: 
[image]

I guess something is off either in my `akvcam` configuration, in my OBS recording configuration, in my `ffmpeg` command or in any valid permutation of the interactions between the three tools :-)
Thing is, I have no clue what might be the source of the issue.

Here's my `akvcam` configuration file:

```ini
[Cameras]
cameras/size = 2

cameras/1/type = output
cameras/1/mode = mmap, userptr, rw
cameras/1/description = Virtual Camera (output device)
cameras/1/formats = 2

cameras/2/type = capture
cameras/2/mode = mmap, rw
cameras/2/description = Virtual Camera
cameras/2/formats = 1, 2

[Formats]
formats/size = 2

# -> into akvcam
formats/1/format = YUY2
formats/1/width = 3840
formats/1/height = 2160
formats/1/fps = 30/1

# <- out of akvcam
formats/2/format = RGB24
formats/2/width = 3840
formats/2/height = 2160
formats/2/fps = 20/1, 15/2

[Connections]
connections/size = 1
connections/1/connection = 1:2
```
As well as my OBS recording configuration:
[image]

And the `ffmpeg` command that I use to connect OBS to `akvcam`:

`ffmpeg -i udp://localhost:12345 -f v4l2 -s 3840x2160 /dev/video4`

hipersayanX, 2020-05-23:
When you run ffmpeg it shows the streaming information like the codecs and pixel formats used,  can you give me that information?

skuro, 2020-05-25:
sure, here's the full output from `ffmpeg`: 

```
$ ffmpeg -i udp://localhost:12345 -f v4l2 -s 3840x2160 /dev/video4
ffmpeg version 3.4.6-0ubuntu0.18.04.1 Copyright (c) 2000-2019 the FFmpeg developers
  built with gcc 7 (Ubuntu 7.3.0-16ubuntu3)
  configuration: --prefix=/usr --extra-version=0ubuntu0.18.04.1 --toolchain=hardened --libdir=/usr/lib/x86_64-linux-gnu --incdir=/usr/include/x86_64-linux-gnu --enable-gpl --disable-stripping --enable-avresample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librubberband --enable-librsvg --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libopencv --enable-libx264 --enable-shared
  libavutil      55. 78.100 / 55. 78.100
  libavcodec     57.107.100 / 57.107.100
  libavformat    57. 83.100 / 57. 83.100
  libavdevice    57. 10.100 / 57. 10.100
  libavfilter     6.107.100 /  6.107.100
  libavresample   3.  7.  0 /  3.  7.  0
  libswscale      4.  8.100 /  4.  8.100
  libswresample   2.  9.100 /  2.  9.100
  libpostproc    54.  7.100 / 54.  7.100
Input #0, mpegts, from 'udp://localhost:12345':
  Duration: N/A, start: 0.000000, bitrate: N/A
  Program 1 
    Metadata:
      service_name    : Service01
      service_provider: FFmpeg
    Stream #0:0[0x100]: Video: mpeg4 (Simple Profile) ([16][0][0][0] / 0x0010), yuv420p, 3840x2160 [SAR 1:1 DAR 16:9], 30 fps, 30 tbr, 90k tbn, 30 tbc
    Stream #0:1[0x101]: Audio: aac (LC) ([15][0][0][0] / 0x000F), 44100 Hz, stereo, fltp, 110 kb/s
Stream mapping:
  Stream #0:0 -> #0:0 (mpeg4 (native) -> rawvideo (native))
Press [q] to stop, [?] for help
Output #0, v4l2, to '/dev/video4':
  Metadata:
    encoder         : Lavf57.83.100
    Stream #0:0: Video: rawvideo (I420 / 0x30323449), yuv420p, 3840x2160 [SAR 1:1 DAR 16:9], q=2-31, 2985984 kb/s, 30 fps, 30 tbn, 30 tbc
    Metadata:
      encoder         : Lavc57.107.100 rawvideo
```

hipersayanX, 2020-05-25:
You must set ```-pix_fmt rgb24```, check the [wiki](https://github.com/webcamoid/akvcam/wiki/Usage-and-examples#streaming-with-ffmpeg).

skuro, 2020-05-26:
Alright, thanks a lot!

closed 2020-05-26 by skuro