webcamoid/akvcam #28: Example akvcam configuration file

fangfufu, 2020-06-03:
The example akvcam configuration file (https://github.com/webcamoid/akvcam/blob/master/share/config_example.ini) is not very capable, the one generated by webcamoid 8.5.0 (the one that's in my repository) supports way more format. So I think example ``config_example.ini`` should be replaced by the one that's generated by webcamoid.

hipersayanX, 2020-06-03:
The file is working fine, the file is not supposed to be fully configured with all possible options because the user will get lost with the options and will not understand what are the most basic lines required to work. Webcamoid instead just throw every option possible at it, and good luck if you can read it :laughing:

fangfufu, 2020-06-03:
I literally have the config file generated by Webcamoid, in case the users want to use my akvcam without installing webcamoid. 😑

But you are right, it is quite hard to read... 😂

fangfufu, 2020-06-03:
Well, I for one, don't understand why these modes are not enabled by default. Is there a downside for enabling those extra modes?

hipersayanX, 2020-06-03:
Yes, if you enable rw, it will disable the camera controls, flipping, scaling etc in output, and brightness, contrast, etc in the capture.

fangfufu, 2020-06-03:
But what if you don't enable the rw? Why not enable all those other output modes by default?

hipersayanX, 2020-06-03:
And they [are](https://github.com/webcamoid/webcamoid/blob/08f786d6860f137cecd654da245720bc7688cc3f/libAvKys/Plugins/VirtualCamera/src/v4l2sys/VCamIPC/src/ipcbridge.cpp#L2655), Webcamoid only enables mmap and usrptr in both sides, because I need the controls in both side, later you can add rw to the output if you need it, or a lot much better, just add support for mmap and usrptr in the program sending the frames.

hipersayanX, 2020-06-07:
What if in the future I add support for more formats? then the client will need to iterate every format * resolution * fps combination. For me it's ok as it is, just declare the formats you need and that is, I prefer flexibility and let the user decide what he wants.

fangfufu, 2020-06-07:
Oh okay, I just want to check my understanding. If you add more formats into the webcam, then when you ask the webcam which format it supports, it will end up iterating everything, is that right? To avoid this situation, you prefer the users to set which format the webcam should support, right?

hipersayanX, 2020-06-07:
Correct, and I've tried in the past and it can really slow down client's GUI since you are basically running a O(n^3) algorithm, maybe not that much but it does.

fangfufu, 2020-06-07:
Ah ok, closing the issue. :)

closed 2020-06-07 by fangfufu