webcamoid/akvcam #29: How to connect to camera?

baradhili, 2020-06-04:
I've got a dvcam connected on /dev/fw0 
I can see it on vlc using `vlc -vvv dv/rawdv:///dev/fw0` - but of course zoom etc can't see it

Now I'm scratching my head as to how to connect a cam to akvcam? Is this even possible?

Thanks

hipersayanX, 2020-06-07:
If I understand you correctly, you want to send your firewire input to the virtual camera, right? if that is the case try with:

```
ffmpeg -i /dev/fw0 -f v4l2 -vcodec rawvideo -pix_fmt rgb24 /dev/videoX
```

Replace the X with the output number. Also check the [wiki](https://github.com/webcamoid/akvcam/wiki).

baradhili, 2020-06-10:
Actually I'm still a bit lost.. the ffmpeg works fine, but how do I tell akvcam to use say /dev/video2 where ffmpeg is sending data to instead of /dev/video0 which the usb cam is on? The wiki never seems to mention this

baradhili, 2020-06-10:
aha! this bit needs to be in the wiki :)

so to be clear, lets say I use the example config.ini 
so I would modprobe in akvcam first, it will then create two new devices under /dev/videoX
then I use ffmpeg to point my dvcam to the input device; and 
tell OBS to look at my output device?

hipersayanX, 2020-06-11:
@baradhili Correct, akvcam unlike v4l2loopback for example, create a splitted device scheme.

baradhili, 2020-06-11:
so I got a bit further.. started up akvcam with almost a stock config.ini (just changed the input camera def to have 720x576 and YUV2 spec to match output from camera I hope
akvcam created video2, video3 in /dev
I pointed ffmpeg at video 3
ffmpeg -i /dev/fw0 -f v4l2 -vcodec rawvideo -pix_fmt rgb24 /dev/video3

however opening up the virtual camera on skype I only got a black screen

hipersayanX, 2020-06-12:
>aha! this bit needs to be in the wiki :)

Added

>however opening up the virtual camera on skype I only got a black screen

This could be a bug not sure. Try latest commit I have added **videonr** option so now you can define which /dev/videoX is your output and which one is your capture device, instead of guessing which device is which, otherwise we are not going anywhere :confused:

baradhili, 2020-06-14:
hmm well a step forward.. I get noise now :)
my config.ini looks like this

```
# 'formats' is a comma separated list of index in the format list bellow.
cameras/1/type = output
cameras/1/mode = mmap, userptr, rw
cameras/1/description = Virtual Camera (output device)
cameras/1/formats = 2
cameras/1/videonr = 7

cameras/2/type = capture
cameras/2/mode = mmap, rw
cameras/2/description = Virtual Camera
cameras/2/formats = 1, 2
cameras/2/videonr = 4
...
formats/2/format = YUY2
formats/2/width = 720 
formats/2/height = 576
formats/2/fps = 25
```
.. all the rest is default

then I did
`sudo ffmpeg -i /dev/fw0 -f v4l2 -vcodec rawvideo -pix_fmt rgb24 /dev/video4`

that gave me "noise"

but if I did `vlc -vvv dv/rawdv:///dev/video4` I got a picture.. so I might have something wrong in the ffmpeg line

hipersayanX, 2020-06-14:
```
sudo ffmpeg -i /dev/fw0 -f v4l2 -vcodec rawvideo -pix_fmt rgb24 /dev/video4
```

Itsnt working because you are sending the frames to the capture device, and you must send the frames to the output device. And please, **DON'T USE SUDO**.

```
ffmpeg -i /dev/fw0 -f v4l2 -vcodec rawvideo -pix_fmt rgb24 /dev/video7
```

eecheng87, 2020-06-26:
I think I stuck in similar problem here.
I tried to feed sample video to output device through `ffmpeg` and capture by capturing device through `vlc`, and I get random noise on `vlc` instead my sample video.

Following is my command:

terminal 1:

    ffmpeg -i small.webm -s 640x480 -r 30 -f v4l2 -vcodec rawvideo -pix_fmt rgb24 /dev/video7

terminal 2:

    vlc v4l2:///dev/video9

Following is part of my config file:
```
[Cameras]
cameras/size = 2

cameras/1/type = output
cameras/1/mode = mmap, userptr, rw
cameras/1/description = Virtual Camera (output device)
cameras/1/formats = 2
cameras/1/videonr = 7

cameras/2/type = capture
cameras/2/mode = mmap, rw
cameras/2/description = Virtual Camera
cameras/2/formats = 1, 2
cameras/2/videonr = 9

```

I wondering is this bug or my misunderstanding? Hope know, thanks!

hipersayanX, 2020-06-26:
@eecheng87 

>I wondering is this bug or my misunderstanding? Hope know, thanks!

```
connections/1/connection = 7:9
```

must be

```
connections/1/connection = 1:2
```

**connection** has the format:

```
connections/1/connection = camera_output_index:camera_capture_index[:camera_capture_index:camera_capture_index...]
```

>BTW, I am interest where you implement generating random signal ( value of pixel ) when user open /dev/videox if user doesn't pass input.

[here](https://github.com/webcamoid/akvcam/blob/ce8c98e9b69407a3563697ab2f571c74b2aab5eb/src/device.c#L492)

eecheng87, 2020-06-27:
Really appreciate your reply, it works!

hipersayanX, 2020-06-27:
It seems that all doubts has been solved, closing.

closed 2020-06-27 by hipersayanX