webcamoid/akvcam #30: provide a docker usage

shockjiang, 2020-07-08:
provide a docker way of using this tool
- docker image
- input video source, say, input.mp4
- virtual device, say, /dev/video9

hipersayanX, 2020-07-11:
Is similar to using a chroot image, you can check the [build.sh](https://github.com/webcamoid/akvcam/blob/ce8c98e9b69407a3563697ab2f571c74b2aab5eb/ports/ci/travis/build.sh#L61) file. Once you get module loaded with it's virtual devices you map the device with the **--device** option, check [this](https://stackoverflow.com/questions/24225647/docker-a-way-to-give-access-to-a-host-usb-or-serial-device).

Niek, 2020-11-20:
You don't have access to a running kernel in Docker, so I guess this will not work unless there's a way to run this in userspace or use the host's kernel.

hipersayanX, 2020-11-20:
@Niek No, you can't run a kernel module in Docket, but... you can use Docker to build the module and then [run it in Qemu](https://github.com/webcamoid/akvcam/blob/a17d5a1767ce2f4b801c00886ff2d82a68a4d2e5/ports/ci/travis/build.sh#L61) :wink:

hipersayanX, 2021-07-30:
I been investigating a bit more of this, and have found [this](https://docs.docker.com/engine/reference/commandline/run/#add-host-device-to-container---device). You just need to load the module in the host and then map the video device (either output or capture) to docker like:

```
docker run \
    --device=/dev/video0:/dev/video0 \
    OTHER_OPTIONS \
    ...
```

Now, loading the module inside Docker and using it in the host, that's impossible without using some sort of black magic sorcery.
I don't see much reason to document it in the wiki, because it's already documented in the Docker documentation.

closed 2021-07-30 by hipersayanX