webcamoid/akvcam #31: Programmatically managing webcams on demand

notdanilo, 2020-07-30:
Is there any way to programmatically create and remove virtual webcams on demand?

hipersayanX, 2020-08-03:
While the driver is loaded, no. Should be possible to add that feature but not sure which strategy it must follow, maybe creating creating a node file and then writing the settings there, but not sure if there is a better way.

hipersayanX, 2021-07-30:
Now I think this feature is really bad idea, it will just bring an unnecessary complexity to the project, the user will try to add or remove devices while the device is still in use, adding a new source of segfaults and instability.
To any developer reading this, feel free to experiment with this feature and show me your results, but I predict the results won't be good.

closed 2021-07-30 by hipersayanX