webcamoid/akvcam #33: unable to compile on fedora 32

damnms, 2020-09-29:
```
oli @ ~/Downloads/akvcam-1.1.0/src - [] $ make
make -C /usr/lib/modules/5.8.11-200.fc32.x86_64/build M=/home/oli/Downloads/akvcam-1.1.0/src  modules
make[1]: Verzeichnis „/usr/src/kernels/5.8.11-200.fc32.x86_64“ wird betreten
  CC [M]  /home/oli/Downloads/akvcam-1.1.0/src/module.o
  CC [M]  /home/oli/Downloads/akvcam-1.1.0/src/attributes.o
  CC [M]  /home/oli/Downloads/akvcam-1.1.0/src/buffer.o
  CC [M]  /home/oli/Downloads/akvcam-1.1.0/src/buffers.o
  CC [M]  /home/oli/Downloads/akvcam-1.1.0/src/controls.o
  CC [M]  /home/oli/Downloads/akvcam-1.1.0/src/device.o
  CC [M]  /home/oli/Downloads/akvcam-1.1.0/src/driver.o
  CC [M]  /home/oli/Downloads/akvcam-1.1.0/src/events.o
  CC [M]  /home/oli/Downloads/akvcam-1.1.0/src/file_read.o
  CC [M]  /home/oli/Downloads/akvcam-1.1.0/src/format.o
  CC [M]  /home/oli/Downloads/akvcam-1.1.0/src/frame.o
  CC [M]  /home/oli/Downloads/akvcam-1.1.0/src/global_deleter.o
  CC [M]  /home/oli/Downloads/akvcam-1.1.0/src/ioctl.o
/home/oli/Downloads/akvcam-1.1.0/src/ioctl.c: In Funktion »akvcam_ioctls_querybuf«:
/home/oli/Downloads/akvcam-1.1.0/src/ioctl.c:993:5: Warnung: Der Rückgabewert von »copy_from_user«, der mit dem Attribut »warn_unused_result« deklariert wurde, wird ignoriert [-Wunused-result]
  993 |     copy_from_user(planes,
      |     ^~~~~~~~~~~~~~~~~~~~~~
  994 |                    (char __user *) buffer->m.planes,
      |                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  995 |                    buffer->length * sizeof(struct v4l2_plane));
      |                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/oli/Downloads/akvcam-1.1.0/src/ioctl.c:1017:5: Warnung: Der Rückgabewert von »copy_to_user«, der mit dem Attribut »warn_unused_result« deklariert wurde, wird ignoriert [-Wunused-result]
 1017 |     copy_to_user((char __user *) buffer->m.planes,
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
 1018 |                  planes,
      |                  ~~~~~~~
 1019 |                  buffer->length * sizeof(struct v4l2_plane));
      |                  ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/oli/Downloads/akvcam-1.1.0/src/ioctl.c: In Funktion »akvcam_ioctl_qbuf«:
/home/oli/Downloads/akvcam-1.1.0/src/ioctl.c:1065:17: Warnung: Der Rückgabewert von »copy_from_user«, der mit dem Attribut »warn_unused_result« deklariert wurde, wird ignoriert [-Wunused-result]
 1065 |                 copy_from_user(planes,
      |                 ^~~~~~~~~~~~~~~~~~~~~~
 1066 |                                (char __user *) buffer->m.planes,
      |                                ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
 1067 |                                buffer->length * sizeof(struct v4l2_plane));
      |                                ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/oli/Downloads/akvcam-1.1.0/src/ioctl.c:1070:21: Warnung: Der Rückgabewert von »copy_from_user«, der mit dem Attribut »warn_unused_result« deklariert wurde, wird ignoriert [-Wunused-result]
 1070 |                     copy_from_user((char *) data + akvcam_format_offset(format, i),
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
 1071 |                                    (char __user *) planes[i].m.userptr,
      |                                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
 1072 |                                    planes[i].length);
      |                                    ~~~~~~~~~~~~~~~~~
/home/oli/Downloads/akvcam-1.1.0/src/ioctl.c:1076:17: Warnung: Der Rückgabewert von »copy_from_user«, der mit dem Attribut »warn_unused_result« deklariert wurde, wird ignoriert [-Wunused-result]
 1076 |                 copy_from_user(data,
      |                 ^~~~~~~~~~~~~~~~~~~~
 1077 |                                (char __user *) buffer->m.userptr,
      |                                ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
 1078 |                                buffer->length);
      |                                ~~~~~~~~~~~~~~~
/home/oli/Downloads/akvcam-1.1.0/src/ioctl.c: In Funktion »akvcam_ioctl_dqbuf«:
/home/oli/Downloads/akvcam-1.1.0/src/ioctl.c:1116:17: Warnung: Der Rückgabewert von »copy_from_user«, der mit dem Attribut »warn_unused_result« deklariert wurde, wird ignoriert [-Wunused-result]
 1116 |                 copy_from_user(planes,
      |                 ^~~~~~~~~~~~~~~~~~~~~~
 1117 |                                (char __user *) buffer->m.planes,
      |                                ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
 1118 |                                buffer->length * sizeof(struct v4l2_plane));
      |                                ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/oli/Downloads/akvcam-1.1.0/src/ioctl.c:1121:21: Warnung: Der Rückgabewert von »copy_to_user«, der mit dem Attribut »warn_unused_result« deklariert wurde, wird ignoriert [-Wunused-result]
 1121 |                     copy_to_user((char __user *) planes[i].m.userptr,
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
 1122 |                                  (char *) data + akvcam_format_offset(format, i),
      |                                  ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
 1123 |                                  planes[i].length);
      |                                  ~~~~~~~~~~~~~~~~~
/home/oli/Downloads/akvcam-1.1.0/src/ioctl.c:1127:17: Warnung: Der Rückgabewert von »copy_to_user«, der mit dem Attribut »warn_unused_result« deklariert wurde, wird ignoriert [-Wunused-result]
[... 37 lines cut ...]
/home/oli/Downloads/akvcam-1.1.0/src/settings.c:233:5: Warnung: Der Rückgabewert von »kstrtou32«, der mit dem Attribut »warn_unused_result« deklariert wurde, wird ignoriert [-Wunused-result]
  233 |     kstrtou32(array_size_str, 10, (u32 *) &array_size);
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  CC [M]  /home/oli/Downloads/akvcam-1.1.0/src/utils.o
  LD [M]  /home/oli/Downloads/akvcam-1.1.0/src/akvcam.o
  MODPOST /home/oli/Downloads/akvcam-1.1.0/src/Module.symvers
  CC [M]  /home/oli/Downloads/akvcam-1.1.0/src/akvcam.mod.o
  LD [M]  /home/oli/Downloads/akvcam-1.1.0/src/akvcam.ko
make[1]: Verzeichnis „/usr/src/kernels/5.8.11-200.fc32.x86_64“ wird verlassen
oli @ ~/Downloads/akvcam-1.1.0/src - [] $ su
Passwort: 
[root@DESKTOP-DM65VNV src]# make install
make -C /usr/lib/modules/5.8.11-200.fc32.x86_64/build M=/home/oli/Downloads/akvcam-1.1.0/src modules_install
make[1]: Verzeichnis „/usr/src/kernels/5.8.11-200.fc32.x86_64“ wird betreten
  INSTALL /home/oli/Downloads/akvcam-1.1.0/src/akvcam.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: crypto/bio/bss_file.c:69
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: crypto/bio/bss_file.c:76
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  5.8.11-200.fc32.x86_64
make[1]: Verzeichnis „/usr/src/kernels/5.8.11-200.fc32.x86_64“ wird verlassen
[root@DESKTOP-DM65VNV src]# depmod -a
[root@DESKTOP-DM65VNV src]# 

```

hipersayanX, 2020-10-03:
It's just throwing some warnings, but the module is building fine, you must [install it with dkms](https://github.com/webcamoid/akvcam/wiki/Build-and-install#managing-the-module-with-dkms).

damnms, 2020-10-03:
for me this looks like an error. make install returns non null and depmod -a returns nothing. maybe the wiki should be updated that it is _required_ to install with dkms. because without, it just wont work. at least on my system it did not work

hipersayanX, 2020-10-04:
>for me this looks like an error. 

```
/home/oli/Downloads/akvcam-1.1.0/src/settings.c:233:5: Warnung: Der Rückgabewert von »kstrtou32«, der mit dem Attribut »warn_unused_result« deklariert wurde, wird ignoriert [-Wunused-result]
  233 |     kstrtou32(array_size_str, 10, (u32 *) &array_size);
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

Warnung == Warning && Warning != Error

>make install returns non null and depmod -a returns nothing. maybe the wiki should be updated that it is required to install with dkms. because without, it just wont work. at least on my system it did not work

The wiki already explain how to install with dkms, ```make install``` is also a valid way to install the module.
The difference between these two is that dkms takes care of recompiling and signing the module every time you update the kernel, while ```make install``` just copy the module to _/lib/modules_ and that is all, if you are using an unsigned kernel and you are not planning to update it, then the last option is fine.

damnms, 2020-10-04:
i do not complain about the build process, i complain about the INSTALL process.
And the install process does not throw warnings, it throws errors!
"SSL error" not "SSL warning"

hipersayanX, 2020-10-04:
The module installs and works fine in Arch and people having the [same problem](https://github.com/webcamoid/akvcam/issues/15#issuecomment-638261405) as you reported that dkms was the solution, beside that I can't give you any other solution.

damnms, 2020-10-05:
so you already know about the problem but still do not update the documentation? thats wierd to me.
why not make a note: "if you encounter ... ssl bio error ... use dkms because otherwise it wont work on your machine" ?

hipersayanX, 2020-10-05:
Wiki updated.

Eutropia, 2020-10-15:
I was working on building an RPM for this for Fedora Silverblue using podman, based off of the approach [atomic-wireguard](https://github.com/jdoss/atomic-wireguard) took when I encountered these same compilation warnings (which are being treated as errors by the default `make && make install` process).

Two things that stick out to me:
First - https://github.com/webcamoid/akvcam/blob/master/src/device.c#L44
There's no alternative declaration for kernels higher than 5.7, e.g: On fedora 32 silverblue I am on kernel 5.8.x
leading to this:
```
/tmp/akvcam-1.0.1/src/device.c:113:28: error: 'VFL_TYPE_GRABBER' undeclared (first use in this function); did you mean 'VFL_TYPE_SUBDEV'?
  113 |     self->vdev->vfl_type = VFL_TYPE_GRABBER;
```

Second - https://github.com/webcamoid/akvcam/blob/master/src/utils.c#L276
The alternative is provided, but the compiler also -Werrors out:
```
/tmp/akvcam-1.0.1/src/buffers.c: In function 'akvcam_buffers_dequeue':
/tmp/akvcam-1.0.1/src/buffers.c:488:30: error: passing argument 1 of 'akvcam_get_timestamp' from incompatible pointer type [-Werror=incompatible-pointer-types]
  488 |         akvcam_get_timestamp(&v4l2_buff->timestamp);
      |                              ^~~~~~~~~~~~~~~~~~~~~
      |                              |
      |                              struct __kernel_v4l2_timeval *
```

I don't know enough about this code to trust silencing the `-Werror` wont break things for this kernel.

Thank you for your time.

hipersayanX, 2020-10-19:
@Eutropia Why are you trying to create a package for an old version (a more than a 1 year old version)? those problems were already solved long time ago. Check [latest release](https://github.com/webcamoid/akvcam/releases).
I will solve the warning messages later.

Eutropia, 2020-10-19:
You must be referring to the `1.0.1` version tag in the build output.  Sorry for the confusion, my build script reads that directly from: https://github.com/webcamoid/akvcam/blob/master/src/module.c#L59

I am on latest master. `4f6548e080419b5b4aa85fd2ba58f132ead0c4db`.  Which problems are you referring to that have been "solved"?

hipersayanX, 2020-10-19:
>You must be referring to the 1.0.1 version tag in the build output. Sorry for the confusion, my build script reads that directly from: https://github.com/webcamoid/akvcam/blob/master/src/module.c#L59

Oh ok, I forgot to update the version. But if you actually reading the version from there as-is, then that is the problem, your script is downloading 1.0.1 instead of 1.1.0.

>I am on latest master. 4f6548e080419b5b4aa85fd2ba58f132ead0c4db. Which problems are you referring to that have been "solved"?

Because you pointed me a line of code of an older version, so I thought you were talking about that, and those compilation error does not appears in [1.1.0 with 5.8 kernel](https://travis-ci.org/github/webcamoid/akvcam/jobs/732580733), I still need to test with 5.9.
But again the code you were compiling was wrong because your download script was downloading the wrong version, try parsing the version from [here](https://github.com/webcamoid/akvcam/tags) instead.

hipersayanX, 2020-11-14:
Please, try latest commit. Now, the driver can be installed as:

```
sudo make install USE_DKMS=1
```

this mill copy the sources to /usr/src/akvcam-version, and with **USE_DKMS=1** will run dkms.

hipersayanX, 2020-11-19:
The issue should have been fixed for 1.1.1, closing

closed 2020-11-19 by hipersayanX