webcamoid/akvcam #39: kernel crash (random but short time)

joe-mojo, 2020-11-20:
I got akvcam working, and I tested video ouput and video capture. ffmpeg/ffplay example from the wiki is working nicely and I can see the virtual capture device with softwares like Cheese. 
**BUT**
While not writing to ouput device nor reading from capture device, I got kernel crash (Ubuntu 20.04.1, Linux 5.4.0-53-generic).
This happens at random times, but very quickly.
Here is a sample trace :
```
Nov 20 01:11:26 MyComputer kernel: [   92.981085] ------------[ cut here ]------------
Nov 20 01:11:26 MyComputer kernel: [   92.981088] kernel BUG at mm/slub.c:3995!
Nov 20 01:11:26 MyComputer kernel: [   92.981098] invalid opcode: 0000 [#1] SMP PTI
Nov 20 01:11:26 MyComputer kernel: [   92.981102] CPU: 1 PID: 5826 Comm: Socket Thread Tainted: P           O      5.4.0-53-generic #59-Ubuntu
Nov 20 01:11:26 MyComputer kernel: [   92.981104] Hardware name: Dell Inc. XPS 15 7590/0VYV0G, BIOS 1.8.1 07/03/2020
Nov 20 01:11:26 MyComputer kernel: [   92.981110] RIP: 0010:kfree+0x1de/0x250
Nov 20 01:11:26 MyComputer kernel: [   92.981114] Code: ff ff 49 8b 04 24 45 31 ed a9 00 00 01 00 74 06 45 0f b6 6c 24 51 49 8b 04 24 a9 00 00 01 00 75 0b 49 8b 44 24 08 a8 01 75 02 <0f> 0b 49 8b 04 24 44 89 e9 ba ff ff ff ff be 06 00 00 00 d3 e2 48
Nov 20 01:11:26 MyComputer kernel: [   92.981116] RSP: 0018:ffffaa270167faa0 EFLAGS: 00010246
Nov 20 01:11:26 MyComputer kernel: [   92.981119] RAX: ffffe0f6912021c8 RBX: ffff93320808818f RCX: 0000000000000000
Nov 20 01:11:26 MyComputer kernel: [   92.981121] RDX: 0000000000000000 RSI: 0000000000000000 RDI: ffff93320808818f
Nov 20 01:11:26 MyComputer kernel: [   92.981123] RBP: ffffaa270167fac0 R08: 0000000000000010 R09: ffff9332407d5d00
Nov 20 01:11:26 MyComputer kernel: [   92.981125] R10: 0000000000000000 R11: ffff9332407d5000 R12: ffffe0f691202200
Nov 20 01:11:26 MyComputer kernel: [   92.981127] R13: 0000000000000000 R14: 0000000000000000 R15: ffff93320808825b
Nov 20 01:11:26 MyComputer kernel: [   92.981129] FS:  00007f6ce86be700(0000) GS:ffff933230040000(0000) knlGS:0000000000000000
Nov 20 01:11:26 MyComputer kernel: [   92.981132] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Nov 20 01:11:26 MyComputer kernel: [   92.981134] CR2: 000055ec36ef6e38 CR3: 0000000417412004 CR4: 00000000003606e0
Nov 20 01:11:26 MyComputer kernel: [   92.981136] Call Trace:
Nov 20 01:11:26 MyComputer kernel: [   92.981143]  do_sys_poll+0x125/0x520
Nov 20 01:11:26 MyComputer kernel: [   92.981149]  ? __enqueue_entity+0x96/0xa0
Nov 20 01:11:26 MyComputer kernel: [   92.981154]  ? poll_select_finish+0x210/0x210
Nov 20 01:11:26 MyComputer kernel: [   92.981158]  ? poll_select_finish+0x210/0x210
Nov 20 01:11:26 MyComputer kernel: [   92.981160]  ? poll_select_finish+0x210/0x210
Nov 20 01:11:26 MyComputer kernel: [   92.981163]  ? poll_select_finish+0x210/0x210
Nov 20 01:11:26 MyComputer kernel: [   92.981166]  ? poll_select_finish+0x210/0x210
Nov 20 01:11:26 MyComputer kernel: [   92.981169]  ? poll_select_finish+0x210/0x210
Nov 20 01:11:26 MyComputer kernel: [   92.981172]  ? poll_select_finish+0x210/0x210
Nov 20 01:11:26 MyComputer kernel: [   92.981175]  ? poll_select_finish+0x210/0x210
Nov 20 01:11:26 MyComputer kernel: [   92.981178]  ? poll_select_finish+0x210/0x210
Nov 20 01:11:26 MyComputer kernel: [   92.981181]  __x64_sys_poll+0x3b/0x140
Nov 20 01:11:26 MyComputer kernel: [   92.981187]  do_syscall_64+0x57/0x190
Nov 20 01:11:26 MyComputer kernel: [   92.981192]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
Nov 20 01:11:26 MyComputer kernel: [   92.981195] RIP: 0033:0x7f6cf8f4aaff
Nov 20 01:11:26 MyComputer kernel: [   92.981198] Code: 54 24 1c 48 89 74 24 10 48 89 7c 24 08 e8 79 1c f8 ff 8b 54 24 1c 48 8b 74 24 10 41 89 c0 48 8b 7c 24 08 b8 07 00 00 00 0f 05 <48> 3d 00 f0 ff ff 77 2b 44 89 c7 89 44 24 08 e8 ad 1c f8 ff 8b 44
Nov 20 01:11:26 MyComputer kernel: [   92.981200] RSP: 002b:00007f6ce86bd6f0 EFLAGS: 00000293 ORIG_RAX: 0000000000000007
Nov 20 01:11:26 MyComputer kernel: [   92.981203] RAX: ffffffffffffffda RBX: 00007f6cb81c3000 RCX: 00007f6cf8f4aaff
Nov 20 01:11:26 MyComputer kernel: [   92.981205] RDX: 00000000ffffffff RSI: 0000000000000036 RDI: 00007f6ce86bd780
Nov 20 01:11:26 MyComputer kernel: [   92.981206] RBP: 0000000000000001 R08: 0000000000000000 R09: 00007f6cf8c65fd0
Nov 20 01:11:26 MyComputer kernel: [   92.981209] R10: 00007f6ce86bd9f8 R11: 0000000000000293 R12: 0000000000000036
Nov 20 01:11:26 MyComputer kernel: [   92.981210] R13: 00007f6ce86bd780 R14: ffffffffffffffca R15: 00000000ffffffff
Nov 20 01:11:26 MyComputer kernel: [   92.981214] Modules linked in: rfcomm xt_conntrack xt_MASQUERADE nf_conntrack_netlink nfnetlink xfrm_user xfrm_algo xt_addrtype iptable_filter iptable_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c bpfilter br_netfilter bridge stp llc vboxnetadp(O) vboxnetflt(O) vboxdrv(O) ccm aufs cmac algif_hash algif_skcipher overlay af_alg bnep dm_crypt snd_hda_codec_hdmi snd_hda_codec_realtek snd_hda_codec_generic nvidia_uvm(O) x86_pkg_temp_thermal intel_powerclamp coretemp mei_hdcp joydev nvidia_drm(PO) kvm_intel snd_sof_pci intel_rapl_msr nvidia_modeset(PO) snd_sof_intel_hda_common kvm snd_soc_hdac_hda snd_sof_intel_hda snd_sof_intel_byt snd_sof_intel_ipc crct10dif_pclmul nls_iso8859_1 snd_sof snd_sof_xtensa_dsp snd_hda_ext_core ghash_clmulni_intel snd_soc_acpi_intel_match snd_soc_acpi aesni_intel snd_soc_core crypto_simd cryptd glue_helper snd_compress ac97_bus snd_pcm_dmaengine dell_laptop nvidia(PO) ledtrig_audio rapl snd_hda_intel snd_intel_dspcfg snd_hda_codec snd_hda_core
Nov 20 01:11:26 MyComputer kernel: [   92.981256]  snd_hwdep snd_pcm snd_seq_midi snd_seq_midi_event intel_cstate snd_rawmidi dell_wmi btusb btrtl input_leds btbcm btintel dell_smbios bluetooth uvcvideo iwlmvm videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 serio_raw dcdbas mac80211 ecdh_generic cdc_acm videobuf2_common ecc snd_seq libarc4 dell_wmi_descriptor i915 wmi_bmof intel_wmi_thunderbolt snd_seq_device snd_timer snd iwlwifi mxm_wmi drm_kms_helper cfg80211 i2c_algo_bit processor_thermal_device rtsx_pci_ms ucsi_acpi fb_sys_fops intel_rapl_common typec_ucsi mei_me soundcore memstick hid_multitouch syscopyarea typec sysfillrect mei sysimgblt intel_soc_dts_iosf intel_pch_thermal mac_hid int3403_thermal int340x_thermal_zone intel_hid dell_smo8800 sparse_keymap acpi_pad int3400_thermal acpi_thermal_rel sch_fq_codel parport_pc ppdev lp parport akvcam(O) videodev mc drm ip_tables x_tables autofs4 hid_generic rtsx_pci_sdmmc crc32_pclmul nvme psmouse i2c_i801 nvme_core rtsx_pci intel_lpss_pci thunderbolt intel_lpss ahci idma64
Nov 20 01:11:26 MyComputer kernel: [   92.981303]  libahci virt_dma i2c_hid hid video pinctrl_cannonlake wmi pinctrl_intel
Nov 20 01:11:26 MyComputer kernel: [   92.981312] ---[ end trace 2837fe6b6634f791 ]---
Nov 20 01:11:26 MyComputer kernel: [   92.981315] RIP: 0010:kfree+0x1de/0x250
Nov 20 01:11:26 MyComputer kernel: [   92.981318] Code: ff ff 49 8b 04 24 45 31 ed a9 00 00 01 00 74 06 45 0f b6 6c 24 51 49 8b 04 24 a9 00 00 01 00 75 0b 49 8b 44 24 08 a8 01 75 02 <0f> 0b 49 8b 04 24 44 89 e9 ba ff ff ff ff be 06 00 00 00 d3 e2 48
Nov 20 01:11:26 MyComputer kernel: [   92.981320] RSP: 0018:ffffaa270167faa0 EFLAGS: 00010246
Nov 20 01:11:26 MyComputer kernel: [   92.981322] RAX: ffffe0f6912021c8 RBX: ffff93320808818f RCX: 0000000000000000
Nov 20 01:11:26 MyComputer kernel: [   92.981324] RDX: 0000000000000000 RSI: 0000000000000000 RDI: ffff93320808818f
Nov 20 01:11:26 MyComputer kernel: [   92.981326] RBP: ffffaa270167fac0 R08: 0000000000000010 R09: ffff9332407d5d00
Nov 20 01:11:26 MyComputer kernel: [   92.981327] R10: 0000000000000000 R11: ffff9332407d5000 R12: ffffe0f691202200
Nov 20 01:11:26 MyComputer kernel: [   92.981329] R13: 0000000000000000 R14: 0000000000000000 R15: ffff93320808825b
Nov 20 01:11:26 MyComputer kernel: [   92.981332] FS:  00007f6ce86be700(0000) GS:ffff933230040000(0000) knlGS:0000000000000000
Nov 20 01:11:26 MyComputer kernel: [   92.981334] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Nov 20 01:11:26 MyComputer kernel: [   92.981335] CR2: 000055ec36ef6e38 CR3: 0000000417412004 CR4: 00000000003606e0
Nov 20 01:11:29 MyComputer kernel: [   96.582260] general protection fault: 0000 [#2] SMP PTI
Nov 20 01:11:29 MyComputer kernel: [   96.582271] CPU: 10 PID: 5462 Comm: cheese Tainted: P      D    O      5.4.0-53-generic #59-Ubuntu
Nov 20 01:11:29 MyComputer kernel: [   96.582275] Hardware name: Dell Inc. XPS 15 7590/0VYV0G, BIOS 1.8.1 07/03/2020
[... cut ...]
Nov 20 01:11:29 MyComputer kernel: [   96.583683] R10: 0000000000000000 R11: ffff9332407d5000 R12: ffffe0f691202200
Nov 20 01:11:29 MyComputer kernel: [   96.583685] R13: 0000000000000000 R14: 0000000000000000 R15: ffff93320808825b
Nov 20 01:11:29 MyComputer kernel: [   96.583690] FS:  00007fc021a83a80(0000) GS:ffff933230280000(0000) knlGS:0000000000000000
Nov 20 01:11:29 MyComputer kernel: [   96.583693] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Nov 20 01:11:29 MyComputer kernel: [   96.583696] CR2: 00007fc00b15b020 CR3: 000000039980c001 CR4: 00000000003606e0
```

hipersayanX, 2020-11-20:
Beside akvcam being loaded, the log does not show nothing related to akvcam and it looks like a crash in your graphics driver, not sure, maybe try loading the module with loglevel=7 and let see if it gives more information.

hipersayanX, 2020-11-22:
I won't discard at all this is an akvcam bug, but I'm missing akvcam log, and that dmesg log is totally useless because is not pointing anywhere in akvcam source code. If no more information available I'll close this issue.

joe-mojo, 2020-11-24:
Sure, I will provide this ASAP.

hipersayanX, 2020-11-25:
I was not able to reproduce this specific bug, but I've found some race conditions when testing in Arch that did not appeared in Ubuntu running in VirtualBox.

joe-mojo, 2020-11-26:
For now, I didn't succeed in capturing dmesg, but I reproduced it. I'll give it another try tomorrow.

hipersayanX, 2021-01-10:
This should have been fixed in 1.2.0.

closed 2021-01-10 by hipersayanX

joe-mojo, 2021-01-11:
Seems stable now ! Is the 1s delay in cheese and vlc usual ?

hipersayanX, 2021-01-11:
I been testing today and had the same delay problem, it probably need some tweak or optimization in the buffers transferring.