webcamoid/akvcam #41: kernel BUG and full system freeze

cosarara, 2020-11-30:
Here's the log:
```
------------[ cut here ]------------
kernel BUG at mm/slub.c:4113!
invalid opcode: 0000 [#1] PREEMPT SMP NOPTI
CPU: 13 PID: 2556 Comm: libobs: graphic Tainted: P           OE     5.9.10-zen1-1-zen #1
Hardware name: To Be Filled By O.E.M. To Be Filled By O.E.M./X570 Phantom Gaming 4, BIOS P1.60 08/21/2019
RIP: 0010:kfree+0x3a2/0x450
Code: 90 fd ff ff 48 8b 45 00 45 31 e4 a9 00 00 01 00 74 05 44 0f b6 65 51 48 8b 45 00 a9 00 00 01 00 75 0a 48 8b 45 08 a8 01 75 02 <0f> 0b 48 8b 45 00 44>
RSP: 0018:ffffba5a8aaffe30 EFLAGS: 00010246
RAX: ffffe4160c36f748 RBX: ffff9b8200000000 RCX: 0000000000000000
RDX: 0000000000000000 RSI: 0000000000000005 RDI: ffff9b8200000000
RBP: ffffe4160c000000 R08: ffff9b828a4f882c R09: ffff9b838a4f880b
R10: ffffffffffffffff R11: 0000000050424752 R12: 0000000000000000
R13: ffff9b81fe46d120 R14: 0000000000000000 R15: 0000000000000040
FS:  00007f5fe1b93640(0000) GS:ffff9b830eb40000(0000) knlGS:0000000000000000
CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
CR2: 00007f60082cf000 CR3: 000000038f208000 CR4: 00000000003506e0
Call Trace:
 ? snprintf+0x51/0x70
 akvcam_list_delete+0x86/0xd0 [akvcam]
 akvcam_ioctl_enum_fmt+0xcb/0x130 [akvcam]
 ? akvcam_ioctl_s_output+0x80/0x80 [akvcam]
 akvcam_ioctl_do+0xa9/0x170 [akvcam]
 akvcam_node_ioctl+0x22/0x30 [akvcam]
 v4l2_ioctl+0x48/0x50 [videodev]
 __x64_sys_ioctl+0x83/0xb0
 do_syscall_64+0x33/0x40
 entry_SYSCALL_64_after_hwframe+0x44/0xa9
RIP: 0033:0x7f60127dbd5d
Code: 00 c3 66 2e 0f 1f 84 00 00 00 00 00 90 f3 0f 1e fa 48 89 f8 48 89 f7 48 89 d6 48 89 ca 4d 89 c2 4d 89 c8 4c 8b 4c 24 08 0f 05 <48> 3d 01 f0 ff ff 73>
RSP: 002b:00007f5fe1b921a8 EFLAGS: 00000246 ORIG_RAX: 0000000000000010
RAX: ffffffffffffffda RBX: 00007f5fe0050280 RCX: 00007f60127dbd5d
RDX: 00007f5fe1b92210 RSI: 00000000c0405602 RDI: 0000000000000030
RBP: 00007f5fe1b921e0 R08: 00007f5fe1b92430 R09: 00007f5fe1b92430
R10: 00000000ffffffff R11: 0000000000000246 R12: 0000000000000001
R13: 0000000000000009 R14: 00007f5fe1b92210 R15: 00007f5fd41acb30
Modules linked in: akvcam(OE) fuse wireguard curve25519_x86_64 libchacha20poly1305 chacha_x86_64 poly1305_x86_64 libblake2s blake2s_x86_64 ip6_udp_tunnel >
 i2c_algo_bit dca wmi pinctrl_amd acpi_cpufreq evdev joydev input_leds mac_hid nfsd vboxnetflt(OE) vboxnetadp(OE) auth_rpcgss nfs_acl lockd vboxdrv(OE) gr>
---[ end trace 047a5b141a572126 ]---
RIP: 0010:kfree+0x3a2/0x450
Code: 90 fd ff ff 48 8b 45 00 45 31 e4 a9 00 00 01 00 74 05 44 0f b6 65 51 48 8b 45 00 a9 00 00 01 00 75 0a 48 8b 45 08 a8 01 75 02 <0f> 0b 48 8b 45 00 44>
RSP: 0018:ffffba5a8aaffe30 EFLAGS: 00010246
RAX: ffffe4160c36f748 RBX: ffff9b8200000000 RCX: 0000000000000000
RDX: 0000000000000000 RSI: 0000000000000005 RDI: ffff9b8200000000
RBP: ffffe4160c000000 R08: ffff9b828a4f882c R09: ffff9b838a4f880b
R10: ffffffffffffffff R11: 0000000050424752 R12: 0000000000000000
R13: ffff9b81fe46d120 R14: 0000000000000000 R15: 0000000000000040
FS:  00007f5fe1b93640(0000) GS:ffff9b830eb40000(0000) knlGS:0000000000000000
CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
CR2: 00007f60082cf000 CR3: 000000038f208000 CR4: 00000000003506e0
```

If it happens again I'll try to note down the exact command used. I'm just trying to get an mjpeg stream into the output device with gst-launch, and it's proving harder than expected.

cosarara, 2020-11-30:
I got it again, but I can't reproduce it on demand. I'm trying to get [gstreamer](https://github.com/webcamoid/akvcam/wiki/Usage-and-examples#streaming-with-gstreamer) to output to the virtual webcam, but it stops after the first frame:

```
> gst-launch-1.0 -v videotestsrc ! v4l2sink device=/dev/video7
Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
/GstPipeline:pipeline0/GstVideoTestSrc:videotestsrc0.GstPad:src: caps = video/x-raw, format=(string)YUY2, width=(int)1920, height=(int)1080, framerate=(fraction)30/1, multiview-mode=(string)mono, interlace-mode=(string)progressive, colorimetry=(string)2:0:0:0, pixel-aspect-ratio=(fraction)1/1
/GstPipeline:pipeline0/GstV4l2Sink:v4l2sink0.GstPad:sink: caps = video/x-raw, format=(string)YUY2, width=(int)1920, height=(int)1080, framerate=(fraction)30/1, multiview-mode=(string)mono, interlace-mode=(string)progressive, colorimetry=(string)2:0:0:0, pixel-aspect-ratio=(fraction)1/1
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
0:00:00.0 / 99:99:99.
```
(stays at 0:00:00.0, displaying from the webcam shows a still image)

hipersayanX, 2020-12-01:
I'm working into this right now, there are a lot of problems with memory management, it will took me some time to solve it.

hipersayanX, 2021-01-10:
This should have been fixed in 1.2.0.

closed 2021-01-10 by hipersayanX