webcamoid/akvcam #42: akvcam module compilation fails with kernel 5.10-rc6

darkonen, 2020-12-06:
compilation error with kernel 5.10-rc6
(same error since kernel 5.10-rc1)
[make.log](https://github.com/webcamoid/akvcam/files/5649038/make.log)

xeaon, 2020-12-13:
i can confirm

```
make -C /usr/lib/modules/5.10.0-051000rc6-generic/build M=/home/user/Downloads/akvcam/src  modules
make[1]: Entering directory '/usr/src/linux-headers-5.10.0-051000rc6-generic'
  CC [M]  /home/user/Downloads/akvcam/src/module.o
  CC [M]  /home/user/Downloads/akvcam/src/attributes.o
  CC [M]  /home/user/Downloads/akvcam/src/buffer.o
  CC [M]  /home/user/Downloads/akvcam/src/buffers.o
  CC [M]  /home/user/Downloads/akvcam/src/controls.o
  CC [M]  /home/user/Downloads/akvcam/src/device.o
  CC [M]  /home/user/Downloads/akvcam/src/driver.o
  CC [M]  /home/user/Downloads/akvcam/src/events.o
  CC [M]  /home/user/Downloads/akvcam/src/file_read.o
/home/user/Downloads/akvcam/src/file_read.c: In function ‘akvcam_file_exists’:
/home/user/Downloads/akvcam/src/file_read.c:149:13: error: implicit declaration of function ‘get_fs’; did you mean ‘sget_fc’? [-Werror=implicit-function-declaration]
  149 |     oldfs = get_fs();
      |             ^~~~~~
      |             sget_fc
/home/user/Downloads/akvcam/src/file_read.c:149:13: error: incompatible types when assigning to type ‘mm_segment_t’ from type ‘int’
/home/user/Downloads/akvcam/src/file_read.c:150:5: error: implicit declaration of function ‘set_fs’; did you mean ‘sget_fc’? [-Werror=implicit-function-declaration]
  150 |     set_fs(KERNEL_DS);
      |     ^~~~~~
      |     sget_fc
/home/user/Downloads/akvcam/src/file_read.c:150:12: error: ‘KERNEL_DS’ undeclared (first use in this function)
  150 |     set_fs(KERNEL_DS);
      |            ^~~~~~~~~
/home/user/Downloads/akvcam/src/file_read.c:150:12: note: each undeclared identifier is reported only once for each function it appears in
/home/user/Downloads/akvcam/src/file_read.c: In function ‘akvcam_file_size’:
/home/user/Downloads/akvcam/src/file_read.c:166:13: error: incompatible types when assigning to type ‘mm_segment_t’ from type ‘int’
  166 |     oldfs = get_fs();
      |             ^~~~~~
/home/user/Downloads/akvcam/src/file_read.c:167:12: error: ‘KERNEL_DS’ undeclared (first use in this function)
  167 |     set_fs(KERNEL_DS);
      |            ^~~~~~~~~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:283: /home/user/Downloads/akvcam/src/file_read.o] Error 1
make[1]: *** [Makefile:1797: /home/user/Downloads/akvcam/src] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.10.0-051000rc6-generic'
make: *** [Makefile:48: all] Error 2

```

hipersayanX, 2021-01-10:
Added support for linux 5.10 in latest release.

closed 2021-01-10 by hipersayanX