webcamoid/akvcam #43: how do i uninstall akvcam 1.1.1 installed using the .run file?

ocaeon, 2020-12-27:
hello, i installed akvcam 1.1.1 using the prepackaged installer akvcam-installer-1.1.1.run and now i seem to be going round in circles trying to activate the module, it's all very unfamilliar to me.  got through creating the files in /etc/akvcam/ but still having problems like the .ko not being recognised, not understanding what dkms is supposed to be doing .. basicaly having it installed like that didn't help me understand what it was doing and what still needed to be done. 
 so i think it's time to remove it all and try again .. but there's no indication of how to uninstall, or where to find each of the files the installer created and what options and configurations it may have modified. 
 please could anyone help with uninstalling and perhaps even explain why each bit goes where it does?

hipersayanX, 2020-12-28:
>and now i seem to be going round in circles trying to activate the module, it's all very unfamilliar to me.

What does "going round in circles trying to activate the module" means? What's the error message?

>got through creating the files in /etc/akvcam/ but still having problems like the .ko not being recognised

If you did not let the installed finish the installation then probably the module was not properly compiled and installed. Your sentences are too vague to figure out what could be the problem.

>not understanding what dkms is supposed to be doing .

[DKMS](https://github.com/dell/dkms) take charge of recompiling all out-of-tree kernel modules every time you update your kernel.

>basicaly having it installed like that didn't help me understand what it was doing and what still needed to be done.

Install the module and then read [the documentation](https://github.com/webcamoid/akvcam/wiki) on how to use it.

>but there's no indication of how to uninstall, or where to find each of the files the installer created and what options and configurations it may have modified.

If you did not modified the install path, the uninstaller (MaintenanceTool) must be located at _/opt/akvcam_, run it to uninstall akvcam.

ocaeon, 2020-12-29:
thankyou, i found the uninstaller where you said at /opt/akvcam so now i can follow the step by step instructions to build from source. 

 sorry i was vague, i was only explaining why i was trying to uninstall.  and thankyou for telling me what dkms is about.

closed 2020-12-29 by ocaeon