webcamoid/akvcam #44: Installation failed - Error during installation process

bdutta, 2021-01-04:
This is on a Ubuntu 18.04.4 LTS setup running on intel i3-3240 w/ 4GB DDR3 RAM, using Liquorix kernel: 5.8.0-10.1-liquorix-amd64 #1 ZEN SMP PREEMPT liquorix 5.8-8ubuntu1~bionic (2020-09-17) x86_64 x86_64 x86_64 GNU/Linux

Error during installation process (com.akvcamprj.akvcam):
Execution failed (Unexpected exit code: 255): "dkms install akvcam/1.1.1"

hipersayanX, 2021-01-05:
Check if there is any log in _/var/lib/dkms/akvcam/1.1.1/build_, I need that information.

```
find /var/lib/dkms/akvcam/1.1.1 -type f -name make.log
```

hipersayanX, 2021-01-10:
Report is too vague, I can't do nothing with this, closing.

closed 2021-01-10 by hipersayanX

GaGag23, 2021-02-18:
You haven't got dkms installed
`sudo apt-get install dkms`