webcamoid/akvcam #46: Kernel panic

sh4dowb, 2021-01-23:
after compiling latest version, using the example config file, loading the module and running the following:
`sudo ffmpeg -i /dev/video0  -f v4l2 /dev/video7`

/dev/video2 output was weird, like this:
[image]

Then a few seconds later I got kernel panic:

```
Jan 23 15:56:40 laptop kernel: [ 4311.764840] ------------[ cut here ]------------
Jan 23 15:56:40 laptop kernel: [ 4311.764843] refcount_t: underflow; use-after-free.
Jan 23 15:56:40 laptop kernel: [ 4311.764879] WARNING: CPU: 6 PID: 23877 at lib/refcount.c:28 refcount_warn_saturate+0xae/0xf0
Jan 23 15:56:40 laptop kernel: [ 4311.764880] Modules linked in: akvcam(OE) thunderbolt rfcomm nft_reject_inet nf_reject_ipv4 nf_reject_ipv6 nft_reject nft_masq xt_conntrack nft_ct xt_MASQUERADE nf_conntr
ack_netlink xfrm_user xfrm_algo xt_addrtype iptable_filter iptable_nat bpfilter br_netfilter nft_chain_nat bridge nf_nat stp nf_conntrack llc nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c nf_tables nfnetlink vb
oxnetadp(OE) vboxnetflt(OE) vboxdrv(OE) ccm aufs cmac algif_hash algif_skcipher af_alg overlay bnep nls_iso8859_1 intel_rapl_msr nvidia_uvm(O) mei_hdcp intel_rapl_common x86_pkg_temp_thermal nvidia_drm(PO
) intel_powerclamp nvidia_modeset(PO) coretemp kvm_intel kvm nvidia(PO) intel_cstate snd_hda_codec_realtek snd_hda_codec_generic ledtrig_audio snd_hda_codec_hdmi uvcvideo snd_hda_intel videobuf2_vmalloc s
nd_intel_dspcfg videobuf2_memops videobuf2_v4l2 snd_hda_codec videobuf2_common snd_hda_core videodev btusb snd_hwdep btrtl btbcm mc btintel snd_pcm bluetooth snd_seq_midi snd_seq_midi_event ecdh_generic e
cc snd_rawmidi
Jan 23 15:56:40 laptop kernel: [ 4311.764965]  iwlmvm snd_seq mac80211 input_leds joydev libarc4 snd_seq_device snd_timer serio_raw intel_wmi_thunderbolt intel_rapl_perf iwlwifi wmi_bmof snd cfg80211 ipmi
_devintf ipmi_msghandler mei_me soundcore mei intel_pch_thermal ideapad_laptop sparse_keymap mac_hid acpi_pad sch_fq_codel parport_pc ppdev lp parport ip_tables x_tables autofs4 dm_crypt hid_logitech_hidp
p hid_logitech_dj usbhid hid_generic i915 crct10dif_pclmul crc32_pclmul ghash_clmulni_intel i2c_algo_bit aesni_intel drm_kms_helper crypto_simd cryptd glue_helper syscopyarea sysfillrect sysimgblt fb_sys_
fops psmouse cec rc_core nvme intel_lpss_pci drm i2c_i801 intel_lpss nvme_core r8169 ahci idma64 realtek libahci virt_dma i2c_hid hid wmi video pinctrl_sunrisepoint pinctrl_intel
Jan 23 15:56:40 laptop kernel: [ 4311.765047] CPU: 6 PID: 23877 Comm: akvcam-thread-3 Tainted: P           OE     5.6.0-1026-oem #26-Ubuntu
Jan 23 15:56:40 laptop kernel: [ 4311.765050] Hardware name: LENOVO INVA/LNVNB161216, BIOS 4GCN38WW 07/18/2018
Jan 23 15:56:40 laptop kernel: [ 4311.765058] RIP: 0010:refcount_warn_saturate+0xae/0xf0
Jan 23 15:56:40 laptop kernel: [ 4311.765064] Code: 7b 31 2e 01 01 e8 77 22 b5 ff 0f 0b 5d c3 80 3d 68 31 2e 01 00 75 91 48 c7 c7 a0 8e 9e bc c6 05 58 31 2e 01 01 e8 57 22 b5 ff <0f> 0b 5d c3 80 3d 46 31 
2e 01 00 0f 85 6d ff ff ff 48 c7 c7 f8 8e
Jan 23 15:56:40 laptop kernel: [ 4311.765067] RSP: 0018:ffffb40b46fcfe48 EFLAGS: 00010286
Jan 23 15:56:40 laptop kernel: [ 4311.765071] RAX: 0000000000000000 RBX: 0000000000000000 RCX: 0000000000000007
Jan 23 15:56:40 laptop kernel: [ 4311.765074] RDX: 0000000000000007 RSI: 0000000000000082 RDI: ffff8e7f9ed99800
Jan 23 15:56:40 laptop kernel: [ 4311.765077] RBP: ffffb40b46fcfe48 R08: 00000000000006fc R09: 0000000000000004
Jan 23 15:56:40 laptop kernel: [ 4311.765080] R10: 0000000000000000 R11: 0000000000000001 R12: ffff8e7e337a2bc0
Jan 23 15:56:40 laptop kernel: [ 4311.765082] R13: 0000000000000000 R14: ffff8e7daf811600 R15: ffffb40b46e7fc40
Jan 23 15:56:40 laptop kernel: [ 4311.765086] FS:  0000000000000000(0000) GS:ffff8e7f9ed80000(0000) knlGS:0000000000000000
Jan 23 15:56:40 laptop kernel: [ 4311.765089] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Jan 23 15:56:40 laptop kernel: [ 4311.765092] CR2: 00007f0408100188 CR3: 00000001f660a002 CR4: 00000000003606e0
Jan 23 15:56:40 laptop kernel: [ 4311.765095] Call Trace:
Jan 23 15:56:40 laptop kernel: [ 4311.765130]  akvcam_frame_delete.part.0+0x52/0x60 [akvcam]
Jan 23 15:56:40 laptop kernel: [ 4311.765148]  akvcam_frame_delete+0x13/0x20 [akvcam]
Jan 23 15:56:40 laptop kernel: [ 4311.765162]  akvcam_default_frame.part.0+0x3e/0xd0 [akvcam]
Jan 23 15:56:40 laptop kernel: [ 4311.765177]  akvcam_device_clock_run_once+0x1e3/0x260 [akvcam]
Jan 23 15:56:40 laptop kernel: [ 4311.765191]  akvcam_device_clock_timeout+0x3d/0x60 [akvcam]
Jan 23 15:56:40 laptop kernel: [ 4311.765199]  kthread+0x104/0x140
Jan 23 15:56:40 laptop kernel: [ 4311.765212]  ? akvcam_device_clock_run_once+0x260/0x260 [akvcam]
Jan 23 15:56:40 laptop kernel: [ 4311.765218]  ? kthread_park+0x90/0x90
Jan 23 15:56:40 laptop kernel: [ 4311.765227]  ret_from_fork+0x35/0x40
Jan 23 15:56:40 laptop kernel: [ 4311.765234] ---[ end trace 24815edb4fa9ce14 ]---
```

hipersayanX, 2021-01-25:
Try, with:

```
ffmpeg -i /dev/video0 -f v4l2 -vcodec rawvideo -pix_fmt rgb24 /dev/videoX
```

and don't use sudo, sudo is not needed here.

sh4dowb, 2021-01-25:
> Try, with:
> 
> ```
> ffmpeg -i /dev/video0 -f v4l2 -vcodec rawvideo -pix_fmt rgb24 /dev/videoX
> ```
> 
> and don't use sudo, sudo is not needed here.

still the same issue. after about 20 seconds whole system just freezes. any ideas on further debugging?

hipersayanX, 2021-01-25:
>any ideas on further debugging?

[The wiki explains it](https://github.com/webcamoid/akvcam/wiki/Debugging-the-driver).

francescortiz, 2021-02-03:
I also get crashes after a few seconds. I tried both RT and non RT kernel. Same issue.

Here are the logs:

https://gist.github.com/francescortiz/1fc016416f0db2738b0f4986d08519e9

francescortiz, 2021-02-03:
Additional information: for me the image looks good, though, I have no rendering issues.

hipersayanX, 2021-02-06:
@francescortiz It's crashing [here](https://gist.github.com/francescortiz/1fc016416f0db2738b0f4986d08519e9#file-akvcam_log-rt-txt-L12944), very similar to @sh4dowb bug, probably due to a race condition.

francescortiz, 2021-02-08:
Hi @hipersayanX , thanks for feedback. Do you know if there is there a workaround? Thanks!

hipersayanX, 2021-02-08:
@francescortiz try replacing the following line [here](https://github.com/webcamoid/akvcam/blob/3d7690e710ab5a74887aaf0550404f5189f071c6/src/device.c#L472) with:

```c++
akvcam_frame_t default_frame = NULL;
```

Make the change in _/usr/src/akvcam-1.2.0_, and rebuild it with **dkms autoinstall**.

francescortiz, 2021-02-09:
Good news! Given your input, I tried now but with `default_frame` option properly set and it is stable. I created a .bmp with the frame size matching the virtual device, 680x480. So I think that the direction you were pointing is the source of the problem.

hipersayanX, 2021-07-30:
This bug should be fixed withing the latest commits, I will make a new release in a few days.

commit 20cccc92 (2021-07-29): Removed the global_deleter, and all global objects, objects now have a well defined lifetime, this should fix #46.
closed 2021-07-31 by hipersayanX