webcamoid/akvcam #47: System freeze on Debian with Kernel 4.19

cytopia, 2021-01-23:
# Sytem freeze on Debian

## Note

Happy to provide other information if required.

## Steps to reproduce

* **Version:** Using `1.2.0` or `master` branch (2020-01-23)

Running any of the execution commands specified below will lead to a complete system freeze. This does not happen at the first run, but around the 4th (give or take) run.

During the system freeze my screen is still visible (but frozen). If I happen to have e.g. youtube running in the background with sound, I can hear the sound to continue for around 10sec, until it also stops. Only thing I can do from this moment on, is to power off my computer via the power button.

## Execution

### webm
```bash
ffmpeg -i test.webm -s 1280x720 -r 30 -f v4l2 -vcodec rawvideo -pix_fmt rgb24 /dev/video7
```
```
  Metadata:
    encoder         : GStreamer matroskamux version 1.14.4
    creation_time   : 2021-01-20T18:00:18.000000Z
  Duration: 00:00:01.92, start: 0.000000, bitrate: 1908 kb/s
    Stream #0:0(eng): Video: vp8, yuv420p(progressive), 960x540, SAR 1:1 DAR 16:9, 15 fps, 15 tbr, 1k tbn, 1k tbc (default)
    Metadata:
      title           : Video
    Stream #0:1(eng): Audio: vorbis, 44100 Hz, mono, fltp (default)
    Metadata:
      title           : Audio
```

### mp4
```bash
ffmpeg -i test.mp4 -s 1280x720 -r 30 -f v4l2 -vcodec rawvideo -pix_fmt rgb24 /dev/video7
```
```
  Metadata:
    major_brand     : isom
    minor_version   : 512
    compatible_brands: isomiso2avc1mp41
    encoder         : Lavf58.20.100
  Duration: 00:00:01.93, start: 0.000000, bitrate: 1943 kb/s
    Stream #0:0(eng): Video: h264 (High) (avc1 / 0x31637661), yuv420p, 960x540 [SAR 1:1 DAR 16:9], 1846 kb/s, 15 fps, 15 tbr, 15360 tbn, 30 tbc (default)
    Metadata:
      handler_name    : VideoHandler
    Stream #0:1(eng): Audio: aac (LC) (mp4a / 0x6134706D), 44100 Hz, mono, fltp, 92 kb/s (default)
    Metadata:
      handler_name    : SoundHandler
```

## System information
```bash
$ uname
4.19.0-13-amd64 #1 SMP Debian 4.19.160-2 (2020-11-28) x86_64 GNU/Linux
```

`config.ini`
```ini
[Cameras]
cameras/size = 2

cameras/1/type = output
cameras/1/mode = mmap, userptr, rw
cameras/1/description = Virtual Camera (output device)
cameras/1/formats = 2
cameras/1/videonr = 7

cameras/2/type = capture
cameras/2/mode = mmap, userptr, rw
cameras/2/description = Virtual Camera (capture device)
cameras/2/formats = 2, 3
cameras/2/videonr = 9

[Formats]
formats/size = 3

formats/1/format = RGB24, YUY2
formats/1/width = 1280
formats/1/height = 720
formats/1/fps = 30/1, 15/1, 15/2

formats/2/format = RGB24
formats/2/width = 640
formats/2/height = 480
formats/2/fps = 30/1, 15/1, 15/2

formats/3/format = RGB24
formats/3/width = 1280
formats/3/height = 720
formats/3/fps = 30/1, 15/1, 15/2

[Connections]
connections/size = 1
connections/1/connection = 1:2
```

GribanovIvan, 2021-04-09:
Same problem. ArchLinux, 5.11.11-zen kernel

wonderful27x, 2021-07-08:
same as 5.8.0-59-generic x86_64  ubuntu20.04

hipersayanX, 2021-07-30:
This bug should be fixed withing the latest commits, I will make a new release in a few days.

closed 2021-07-31 by hipersayanX