webcamoid/akvcam #49: Is Raspberry Pi supported?

utacc, 2021-02-25:
During make, I get:
`make -C /lib/modules/5.10.11-v7l+/build M=/home/pi/akvcam/src  modules
make[1]: Entering directory '/usr/src/linux-headers-5.10.11-v7l+'
  CC [M]  /home/pi/akvcam/src/module.o
  CC [M]  /home/pi/akvcam/src/attributes.o
  CC [M]  /home/pi/akvcam/src/buffers.o
  CC [M]  /home/pi/akvcam/src/controls.o
  CC [M]  /home/pi/akvcam/src/device.o
  CC [M]  /home/pi/akvcam/src/driver.o
  CC [M]  /home/pi/akvcam/src/file_read.o
  CC [M]  /home/pi/akvcam/src/format.o
  CC [M]  /home/pi/akvcam/src/frame.o
  CC [M]  /home/pi/akvcam/src/global_deleter.o
  CC [M]  /home/pi/akvcam/src/ioctl.o
  CC [M]  /home/pi/akvcam/src/list.o
  CC [M]  /home/pi/akvcam/src/log.o
  CC [M]  /home/pi/akvcam/src/map.o
  CC [M]  /home/pi/akvcam/src/rbuffer.o
  CC [M]  /home/pi/akvcam/src/settings.o
  CC [M]  /home/pi/akvcam/src/utils.o
  LD [M]  /home/pi/akvcam/src/akvcam.o
  MODPOST /home/pi/akvcam/src/Module.symvers
ERROR: modpost: "__aeabi_ldivmod" [/home/pi/akvcam/src/akvcam.ko] undefined!
make[2]: *** [scripts/Makefile.modpost:111: /home/pi/akvcam/src/Module.symvers] Error 1
make[2]: *** Deleting file '/home/pi/akvcam/src/Module.symvers'
make[1]: *** [Makefile:1712: modules] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.10.11-v7l+'
make: *** [Makefile:44: all] Error 2`

Not really unlike #40. I did the

`sudo depmod -a
sudo modprobe v4l2-common  <-- this module wasn't found
sudo modprobe videobuf2-core  <-- neither was this one
sudo modprobe videobuf2-v4l2
sudo modprobe videobuf2-vmalloc
sudo modprobe videodev`

bit, and the modules noted weren't found. I'm not sure what to try next...?
Thank you for your time and consideration.

hipersayanX, 2021-03-01:
I hadn't time to test it but, if the driver compiles fine, then it should be.

```
sudo depmod -a
sudo modprobe v4l2-common <-- this module wasn't found
sudo modprobe videobuf2-core <-- neither was this one
sudo modprobe videobuf2-v4l2
sudo modprobe videobuf2-vmalloc
sudo modprobe videodev
```

>bit, and the modules noted weren't found. I'm not sure what to try next...?

The driver won't work without those modules, either you install those modules, recompile the kernel, or use another distro.

utacc, 2021-03-02:
Thank you for the reply!
Did a bit more research; apparently it's very iffy to try to get more than
the very basic v4l installed and running on a Raspberry Pi. Besides, my
desired target for my video stream is Zoom, which isn't in any way
compatible with a Raspberry Pi. I ultimately wound up switching to Mint
Linux on a different machine. I had to remove the default packaged v4l2 and
replace it with the most current compiled version, but that was uneventful
and entirely routine. After that, the build/install/launch for akvcam was
also uneventful. Configuration took a day or so to figure out, but I was
over-thinking it.
At this point, akvcam is installed and running with Zoom on Mint Linux; I
left it up while I was at work yesterday; I can report no logged errors in
over 8 hours of continuous operation.
It's important on Mint to do a "sudo apt update & sudo apt dist-upgrade"
though. Although compilation and install of akvcam can succeed, it seemed
like cinnamon wasn't happy when I used ffplay /dev/videoX; when I
overlapped windows, I'd see "ghost" images of my video playback under the
xterms. Did an upgrade and CPU load dropped sharply.
I noticed that re-nicing the akvcam process helped with frame rate
(slightly). Sometimes I wish we could just specify a thread as
"non-preempt-able"...  XD

Excellent work on this! Totally loving akvcam!

-Rich

hipersayanX, 2021-03-02:
>Did a bit more research; apparently it's very iffy to try to get more than the very basic v4l installed and running on a Raspberry Pi. 

But that is more a like limitation of the kernel used in the distro you have installed in the Raspberry, right? I doubt v4l2 does not works in ARM. I would suggest trying with [Arch](https://archlinuxarm.org/), maybe or another distro.
I've my RP collecting dust, but I had not time to test it :neutral_face:

utacc, 2021-03-04:
It's more like the kernel modules; it feels as if someone's gone in and
just done the minimal to make the loopback adapter work. At some point, I'm
going to go build up one pi as a throw-away machine and figure out the
build recipe for the full v4l2 toolkit, but right now only three of the
five modules are present.

I've hooked up a websocket video feed to akvcam, and I'm suddenly wishing
akvcam could schedule frames any more quickly, but that is definitely a
kernel/scheduling issue.

closed 2021-07-30 by hipersayanX