webcamoid/akvcam #52: Makefile doesn't seem to be working right?

SlideChaser, 2021-06-24:
I've had a couple problems with  make files lately, so maybe I'm the problem lol

getting this result for sudo make dkms_install

dkms remove akvcam/1.2.0 --all
make: dkms: Command not found
make: [Makefile:63: dkms_uninstall] Error 127 (ignored)
rm -rvf /usr/src/akvcam-1.2.0
rm -rvf /usr/src/akvcam-1.2.0
mkdir -p /usr/src/akvcam-1.2.0
cp -f Makefile /usr/src/akvcam-1.2.0
cp -f dkms.conf /usr/src/akvcam-1.2.0
cp -f *.h /usr/src/akvcam-1.2.0
cp -f *.c /usr/src/akvcam-1.2.0
dkms install akvcam/1.2.0
make: dkms: Command not found
make: *** [Makefile:57: dkms_install] Error 127

hipersayanX, 2021-06-25:
As the error says, you must install dkms.

SlideChaser, 2021-07-07:
ok.

closed 2021-07-07 by SlideChaser