webcamoid/akvcam #53: libv4l2: error getting pixformat: Invalid argument

taylorchu, 2021-07-10:
I am using latest master commit (7e58309c38248a0ff857679f0a632b4b213277e2) with the example config.

```
+ modinfo akvcam
filename:       /lib/modules/5.4.0-1042-gke/updates/dkms/akvcam.ko
version:        1.2.0
description:    AkVCam Virtual Camera
author:         Gonzalo Exequiel Pedone
license:        GPL
srcversion:     1454E557E4345DE61A4E204
depends:        videobuf2-v4l2,videodev,videobuf2-vmalloc,videobuf2-common
retpoline:      Y
name:           akvcam
vermagic:       5.4.0-1042-gke SMP mod_unload modversions
parm:           loglevel:Debug verbosity (-2 to 7) (int)
parm:           config_file:Full path to virtual cameras config file (string)
```

```
[Cameras]
cameras/size = 64
cameras/1/type = output
cameras/1/mode = mmap, userptr, rw
cameras/1/description = Virtual Camera 1 (output device)
cameras/1/formats = 1
cameras/1/videonr = 1
cameras/2/type = output
cameras/2/mode = mmap, userptr, rw
cameras/2/description = Virtual Camera 2 (output device)
cameras/2/formats = 1
cameras/2/videonr = 2
cameras/3/type = output
cameras/3/mode = mmap, userptr, rw
cameras/3/description = Virtual Camera 3 (output device)
cameras/3/formats = 1
cameras/3/videonr = 3
cameras/4/type = output
cameras/4/mode = mmap, userptr, rw
cameras/4/description = Virtual Camera 4 (output device)
cameras/4/formats = 1
cameras/4/videonr = 4
cameras/5/type = output
cameras/5/mode = mmap, userptr, rw
cameras/5/description = Virtual Camera 5 (output device)
cameras/5/formats = 1
cameras/5/videonr = 5
cameras/6/type = output
cameras/6/mode = mmap, userptr, rw
cameras/6/description = Virtual Camera 6 (output device)
cameras/6/formats = 1
cameras/6/videonr = 6
cameras/7/type = output
cameras/7/mode = mmap, userptr, rw
cameras/7/description = Virtual Camera 7 (output device)
cameras/7/formats = 1
cameras/7/videonr = 7
cameras/8/type = output
cameras/8/mode = mmap, userptr, rw
cameras/8/description = Virtual Camera 8 (output device)
cameras/8/formats = 1
cameras/8/videonr = 8
cameras/9/type = output
cameras/9/mode = mmap, userptr, rw
cameras/9/description = Virtual Camera 9 (output device)
cameras/9/formats = 1
cameras/9/videonr = 9
cameras/10/type = output
cameras/10/mode = mmap, userptr, rw
cameras/10/description = Virtual Camera 10 (output device)
cameras/10/formats = 1
[... 293 lines cut ...]
connections/11/connection = 11:43
connections/12/connection = 12:44
connections/13/connection = 13:45
connections/14/connection = 14:46
connections/15/connection = 15:47
connections/16/connection = 16:48
connections/17/connection = 17:49
connections/18/connection = 18:50
connections/19/connection = 19:51
connections/20/connection = 20:52
connections/21/connection = 21:53
connections/22/connection = 22:54
connections/23/connection = 23:55
connections/24/connection = 24:56
connections/25/connection = 25:57
connections/26/connection = 26:58
connections/27/connection = 27:59
connections/28/connection = 28:60
connections/29/connection = 29:61
connections/30/connection = 30:62
connections/31/connection = 31:63
connections/32/connection = 32:64
```

After more debugging, I found that all v4l2-compliance tests passed EXCEPT for create/querybuf.

taylorchu, 2021-07-10:
```
v4l2-compliance SHA   : not available

Driver Info:
	Driver name   : akvcam
	Card type     : Virtual Camera 1 (output device
	Bus info      : platform:akvcam-1
	Driver version: 5.4.101
	Capabilities  : 0x85200002
		Video Output
		Read/Write
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps   : 0x05200002
		Video Output
		Read/Write
		Streaming
		Extended Pix Format

Compliance test for device /dev/video1 (not using libv4l2):

Required ioctls:
	test VIDIOC_QUERYCAP: OK

Allow for multiple opens:
	test second video open: OK
	test VIDIOC_QUERYCAP: OK
	test VIDIOC_G/S_PRIORITY: OK
	test for unlimited opens: OK

Debug ioctls:
	test VIDIOC_DBG_G/S_REGISTER: OK (Not Supported)
	test VIDIOC_LOG_STATUS: OK

Input ioctls:
	test VIDIOC_G/S_TUNER/ENUM_FREQ_BANDS: OK (Not Supported)
	test VIDIOC_G/S_FREQUENCY: OK (Not Supported)
	test VIDIOC_S_HW_FREQ_SEEK: OK (Not Supported)
	test VIDIOC_ENUMAUDIO: OK (Not Supported)
	test VIDIOC_G/S/ENUMINPUT: OK (Not Supported)
	test VIDIOC_G/S_AUDIO: OK (Not Supported)
	Inputs: 0 Audio Inputs: 0 Tuners: 0

Output ioctls:
	test VIDIOC_G/S_MODULATOR: OK (Not Supported)
	test VIDIOC_G/S_FREQUENCY: OK (Not Supported)
	test VIDIOC_ENUMAUDOUT: OK (Not Supported)
	test VIDIOC_G/S/ENUMOUTPUT: OK
	test VIDIOC_G/S_AUDOUT: OK (Not Supported)
	Outputs: 1 Audio Outputs: 0 Modulators: 0

Input/Output configuration ioctls:
	test VIDIOC_ENUM/G/S/QUERY_STD: OK (Not Supported)
	test VIDIOC_ENUM/G/S/QUERY_DV_TIMINGS: OK (Not Supported)
	test VIDIOC_DV_TIMINGS_CAP: OK (Not Supported)
	test VIDIOC_G/S_EDID: OK (Not Supported)

Test output 0:

	Control ioctls:
		test VIDIOC_QUERY_EXT_CTRL/QUERYMENU: OK
		test VIDIOC_QUERYCTRL: OK
		test VIDIOC_G/S_CTRL: OK
		test VIDIOC_G/S/TRY_EXT_CTRLS: OK
		test VIDIOC_(UN)SUBSCRIBE_EVENT/DQEVENT: OK
		test VIDIOC_G/S_JPEGCOMP: OK (Not Supported)
		Standard Controls: 3 Private Controls: 3

	Format ioctls:
		test VIDIOC_ENUM_FMT/FRAMESIZES/FRAMEINTERVALS: OK
		test VIDIOC_G/S_PARM: OK
		test VIDIOC_G_FBUF: OK (Not Supported)
		test VIDIOC_G_FMT: OK
		test VIDIOC_TRY_FMT: OK
		test VIDIOC_S_FMT: OK
		test VIDIOC_G_SLICED_VBI_CAP: OK (Not Supported)
		test Cropping: OK (Not Supported)
		test Composing: OK (Not Supported)
		test Scaling: OK (Not Supported)

	Codec ioctls:
		test VIDIOC_(TRY_)ENCODER_CMD: OK (Not Supported)
		test VIDIOC_G_ENC_INDEX: OK (Not Supported)
		test VIDIOC_(TRY_)DECODER_CMD: OK (Not Supported)

	Buffer ioctls:
		fail: v4l2-test-buffers.cpp(533): check_0(reqbufs.reserved, sizeof(reqbufs.reserved))
		test VIDIOC_REQBUFS/CREATE_BUFS/QUERYBUF: FAIL
		test VIDIOC_EXPBUF: OK

Test output 0:

Total: 43, Succeeded: 42, Failed: 1, Warnings: 0
```

hipersayanX, 2021-07-10:
I will check, the VIDIOC_REQBUFS/CREATE_BUFS/QUERYBUF was [managed internally](https://github.com/webcamoid/akvcam/blob/7e58309c38248a0ff857679f0a632b4b213277e2/src/buffers.c#L23) by the v4l2 kernel API, not sure if it's a bug in my side, or you hitted some limit in the API.

taylorchu, 2021-07-10:
Thanks v4l2loopback works on the same machine. So I think there is something wrong in the code

taylorchu, 2021-07-10:
v4loopback compliance test on the same node:
```
v4l2-compliance -d /dev/video0
v4l2-compliance SHA   : not available

Driver Info:
	Driver name   : v4l2 loopback
	Card type     : Dummy video device (0x0000)
	Bus info      : platform:v4l2loopback-000
	Driver version: 5.4.101
	Capabilities  : 0x85208003
		Video Capture
		Video Output
		Video Memory-to-Memory
		Read/Write
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps   : 0x05208003
		Video Capture
		Video Output
		Video Memory-to-Memory
		Read/Write
		Streaming
		Extended Pix Format

Compliance test for device /dev/video0 (not using libv4l2):

Required ioctls:
		fail: v4l2-compliance.cpp(584): (dcaps & input_caps) && (dcaps & output_caps)
	test VIDIOC_QUERYCAP: FAIL

Allow for multiple opens:
	test second video open: OK
		fail: v4l2-compliance.cpp(584): (dcaps & input_caps) && (dcaps & output_caps)
	test VIDIOC_QUERYCAP: FAIL
	test VIDIOC_G/S_PRIORITY: OK
		fail: v4l2-compliance.cpp(632): !ok
	test for unlimited opens: FAIL

Debug ioctls:
	test VIDIOC_DBG_G/S_REGISTER: OK (Not Supported)
	test VIDIOC_LOG_STATUS: OK (Not Supported)

Input ioctls:
	test VIDIOC_G/S_TUNER/ENUM_FREQ_BANDS: OK (Not Supported)
	test VIDIOC_G/S_FREQUENCY: OK (Not Supported)
	test VIDIOC_S_HW_FREQ_SEEK: OK (Not Supported)
	test VIDIOC_ENUMAUDIO: OK (Not Supported)
	test VIDIOC_G/S/ENUMINPUT: OK
	test VIDIOC_G/S_AUDIO: OK (Not Supported)
	Inputs: 1 Audio Inputs: 0 Tuners: 0

Output ioctls:
	test VIDIOC_G/S_MODULATOR: OK (Not Supported)
	test VIDIOC_G/S_FREQUENCY: OK (Not Supported)
	test VIDIOC_ENUMAUDOUT: OK (Not Supported)
	test VIDIOC_G/S/ENUMOUTPUT: OK
	test VIDIOC_G/S_AUDOUT: OK (Not Supported)
	Outputs: 1 Audio Outputs: 0 Modulators: 0

Input/Output configuration ioctls:
	test VIDIOC_ENUM/G/S/QUERY_STD: OK (Not Supported)
	test VIDIOC_ENUM/G/S/QUERY_DV_TIMINGS: OK (Not Supported)
	test VIDIOC_DV_TIMINGS_CAP: OK (Not Supported)
	test VIDIOC_G/S_EDID: OK (Not Supported)

Test input 0:

	Control ioctls:
		test VIDIOC_QUERY_EXT_CTRL/QUERYMENU: OK
		test VIDIOC_QUERYCTRL: OK
		test VIDIOC_G/S_CTRL: OK
		test VIDIOC_G/S/TRY_EXT_CTRLS: OK
		test VIDIOC_(UN)SUBSCRIBE_EVENT/DQEVENT: OK
		test VIDIOC_G/S_JPEGCOMP: OK (Not Supported)
		Standard Controls: 1 Private Controls: 4

	Format ioctls:
		fail: v4l2-test-formats.cpp(303): Video Capture cap set, but no Video Capture formats defined
		test VIDIOC_ENUM_FMT/FRAMESIZES/FRAMEINTERVALS: FAIL
		fail: v4l2-test-formats.cpp(1213): got error 22 when setting parms for buftype 1
		test VIDIOC_G/S_PARM: FAIL
		test VIDIOC_G_FBUF: OK (Not Supported)
		fail: v4l2-test-formats.cpp(590): Video Capture cap set, but no Video Capture formats defined
		test VIDIOC_G_FMT: FAIL
		test VIDIOC_TRY_FMT: OK (Not Supported)
		test VIDIOC_S_FMT: OK (Not Supported)
		test VIDIOC_G_SLICED_VBI_CAP: OK (Not Supported)
		test Cropping: OK (Not Supported)
		test Composing: OK (Not Supported)
		fail: v4l2-test-formats.cpp(1645): doioctl(node, VIDIOC_G_FMT, &fmt)
		test Scaling: FAIL

	Codec ioctls:
		test VIDIOC_(TRY_)ENCODER_CMD: OK (Not Supported)
		test VIDIOC_G_ENC_INDEX: OK (Not Supported)
		test VIDIOC_(TRY_)DECODER_CMD: OK (Not Supported)

	Buffer ioctls:
		test VIDIOC_REQBUFS/CREATE_BUFS/QUERYBUF: OK
		test VIDIOC_EXPBUF: OK (Not Supported)

Test input 0:

Total: 43, Succeeded: 36, Failed: 7, Warnings: 0
```

hipersayanX, 2021-07-10:
@taylorchu which version of v4l-utils are you using?

taylorchu, 2021-07-11:
1.14.2-1

```
Package: v4l-utils
Version: 1.14.2-1
Priority: optional
Section: universe/utils
Origin: Ubuntu
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Original-Maintainer: Gregor Jasny <gjasny@googlemail.com>
Bugs: https://bugs.launchpad.net/ubuntu/+filebug
Installed-Size: 1461 kB
Depends: libv4l-0 (= 1.14.2-1), libv4l2rds0 (= 1.14.2-1), libc6 (>= 2.15), libgcc1 (>= 1:3.0), libstdc++6 (>= 5.2), libudev1 (>= 183)
Breaks: ivtv-utils (<< 1.4.1-2), media-ctl
Replaces: ivtv-utils (<< 1.4.1-2), media-ctl
Homepage: https://linuxtv.org/downloads/v4l-utils/
Download-Size: 407 kB
APT-Manual-Installed: yes
APT-Sources: http://archive.ubuntu.com/ubuntu bionic/universe amd64 Packages
Description: Collection of command line video4linux utilities
```

hipersayanX, 2021-07-11:
Yes, the fail is pointing [here](https://github.com/gjasny/v4l-utils/blob/d7d42f2c7bc7d228788149741201cac518e317b7/utils/v4l2-compliance/v4l2-test-buffers.cpp#L675), and VIDIOC_REQBUFS is managed by vb2_ioctl_reqbufs [here](https://github.com/webcamoid/akvcam/blob/7e58309c38248a0ff857679f0a632b4b213277e2/src/ioctl.c#L90), I haven't any kind of access to the v4l2_requestbuffers structure, the bug is in the kernel.

hipersayanX, 2021-07-30:
I been testing your config.ini as-is and it was working fine for me, all tests passed. One thing I forgot to ask you was the dmesg log, because now I think the ```test VIDIOC_REQBUFS/CREATE_BUFS/QUERYBUF: FAIL``` could have been caused by a segfault in the module.
The module should be more stable now with the latest commits, I'll make a new release in a few days.

closed 2021-07-31 by hipersayanX