webcamoid/akvcam #54: Error when installing

sandyandi, 2021-07-25:
Hi guys,

I got the following error when running `make`:
```bash
make[1]: *** No rule to make target 'Source/akvcam/src'.  Stop.
```

I already double checked I have dependencies (`gcc`, `make` and the appropriate `linux-headers`).

For reference, I'm running Pop!_OS 20.04 LTS with `linux-headers-5.11.0-7614-generic`.

I successfully installed it before my kernel got upgraded to 5.11 but it seems to fail now. Is kernel 5.11 not supported yet, or?

hipersayanX, 2021-07-25:
>Is kernel 5.11 not supported yet, or?

It is.
The error looks more like you are running make in a different directory, you must:

```
cd Source/akvcam/src
make
```

Look at the [wiki](https://github.com/webcamoid/akvcam/wiki/Build-and-install#build-and-install)

sandyandi, 2021-07-25:
I ran `make` where the Makefile is located (`akvcam/src`). Below is the complete output:
```
make -C /usr/lib/modules/5.11.0-7620-generic/build M=<truncated for brevity>/akvcam/src  modules
make[1]: Entering directory '/usr/src/linux-headers-5.11.0-7620-generic'
make[1]: *** No rule to make target 'Source/akvcam/src'.  Stop.
make[1]: Leaving directory '/usr/src/linux-headers-5.11.0-7620-generic'
make: *** [Makefile:44: all] Error 2
```

sandyandi, 2021-07-26:
I found the issue. It's a problem with directory containing space. I cloned akvcam in `Projects/Open Source/akvcam` and it's only reading the `Source/akvcam`. Notice it's missing "Open ".

I fixed it by removing the space between "Open" and "Source".

closed 2021-07-26 by sandyandi