webcamoid/akvcam #57: Staticy output, no idea what I did in config

Dimitri-D-Hoffman, 2021-12-05:
As previously mentioned in title, I am trying to get this thing to display my webcamoid video but I'm starting to wonder if I have it all wrong. I set most of the setting settings as standard, and when I went to check the display I had created, this was the output: 
[image]

would this mean the [connections] settings are all wrong? or the cameras themselves?

hipersayanX, 2021-12-06:
Did you wrote the setting manually or you let Webcamoid set it for you?.
If you wrote it manually, just paste the code here.

Dimitri-D-Hoffman, 2021-12-06:
sure thing! one moment;

Dimitri-D-Hoffman, 2021-12-06:
# Virtual camera configuration file example.
#
# Please, read the instructions to the end.

[Cameras]
# First at all you must define how many virtual cameras will be created.
cameras/size = 3

# Then, define it's properties.
#
# A virtual camera can be of 2 types: 'capture' and 'output'.
# A 'capture' device will be seen as a normal webcam by any webcam capture
# program.
# A 'output' device will receive frames from a producer program and send it to
# one or many 'capture' devices.
#
# A camera can have also 3 capture/output modes: 'mmap', 'userptr' and 'rw'.
# 'mmap' is the most widely supported mode by far, enabling this is more than
# enough in most cases. 'rw' allow you to "echo" or "cat" frames as raw data
# directly to the device using the default frame format. Enabling 'rw' mode will
# disable emulated camera controls in the 'capture' device (brightness,
# contrast, saturation, etc.).
# A device can support all 3 modes at same time.
#
# 'formats' is a comma separated list of index in the format list bellow.
#
# It's also possible to set the device number by setting the 'videonr' property,
# if for example videonr=7 the the device will be created as "/dev/video7".
# If 'videonr' is already taken, negative or not set, the driver will assign the
# first free device number.
cameras/1/type = output
cameras/1/mode = mmap, userptr, rw
cameras/1/description = Virtual Camera (output device)
cameras/1/formats = 2
cameras/1/videonr = 1

cameras/2/type = capture
cameras/2/mode = mmap, rw
cameras/2/description = VrtCamDatura
cameras/2/formats = 1, 2

cameras/3/type = capture
cameras/3/mode = mmap, rw
cameras/3/description = VirtualCamera2
cameras/3/formats = 1, 2

[Formats]
# Define how many formats will be supported by the camera.
formats/size = 2

# Now define the frame pixel formats, resolutions and frame rates supported by
# the camera.
#
# Supported capture formats:
#
#     RGB32
#     RGB24
#     RGB16
#     RGB15
#     BGR32
#     BGR24
#     UYVY
#     YUY2
#
# Supported output formats:
#
#     RGB24
#     BGR24
#
# YUY2 640x480 is one of the most widely supported formats in webcam capture
# programs. First format defined is the default frame format for
# 'capture'/'output'.
# 'width', 'height' and 'fps' are unsigned integers.
formats/1/format = YUY2
formats/1/width = 640
formats/1/height = 480
formats/1/fps = 30

# The parameters can also be specified as a comma separated list, so it's
# possible to combine the parameters to define several formats in one group.
# 'fps' can also be defined as a fraction.
# The following lines will define 4 formats:
#
#     RGB24 640x480 20  FPS
#     RGB24 640x480 7.5 FPS
#     YUY2  640x480 20  FPS
#     YUY2  640x480 7.5 FPS
formats/2/format = RGB24, YUY2
formats/2/width = 640
formats/2/height = 480
formats/2/fps = 20/1, 15/2

# Finally, to create a fully working virtual camera, you must connect one
# 'output' to one or many 'capture' devices.
# Connections are made by index, separated by a colon. The first index is the
# 'output' device, the following index are 'capture' devices.
[Connections]
connections/size = 2
connections/1/connection = 1:2
connections/2/connection = 1:3
# You can also define a default frame when a 'capture' device is not receiving
# any input. Only 24 bpp and 32 bpp BMP files are supported.
[General]
default_frame = /etc/akvcam/file.bmp
# This config will take effect on modprobe/insmod.

Dimitri-D-Hoffman, 2021-12-06:
sorry, not much of a git user

 `# Virtual camera configuration file example.
#
# Please, read the instructions to the end.

[Cameras]
# First at all you must define how many virtual cameras will be created.
cameras/size = 3

# Then, define it's properties.
#
# A virtual camera can be of 2 types: 'capture' and 'output'.
# A 'capture' device will be seen as a normal webcam by any webcam capture
# program.
# A 'output' device will receive frames from a producer program and send it to
# one or many 'capture' devices.
#
# A camera can have also 3 capture/output modes: 'mmap', 'userptr' and 'rw'.
# 'mmap' is the most widely supported mode by far, enabling this is more than
# enough in most cases. 'rw' allow you to "echo" or "cat" frames as raw data
# directly to the device using the default frame format. Enabling 'rw' mode will
# disable emulated camera controls in the 'capture' device (brightness,
# contrast, saturation, etc.).
# A device can support all 3 modes at same time.
#
# 'formats' is a comma separated list of index in the format list bellow.
#
# It's also possible to set the device number by setting the 'videonr' property,
# if for example videonr=7 the the device will be created as "/dev/video7".
# If 'videonr' is already taken, negative or not set, the driver will assign the
# first free device number.
cameras/1/type = output
cameras/1/mode = mmap, userptr, rw
cameras/1/description = Virtual Camera (output device)
cameras/1/formats = 2
cameras/1/videonr = 1

cameras/2/type = capture
cameras/2/mode = mmap, rw
cameras/2/description = VrtCamDatura
cameras/2/formats = 1, 2

cameras/3/type = capture
cameras/3/mode = mmap, rw
cameras/3/description = VirtualCamera2
cameras/3/formats = 1, 2

[Formats]
# Define how many formats will be supported by the camera.
formats/size = 2

# Now define the frame pixel formats, resolutions and frame rates supported by
# the camera.
#
# Supported capture formats:
#
#     RGB32
#     RGB24
#     RGB16
#     RGB15
#     BGR32
#     BGR24
#     UYVY
#     YUY2
#
# Supported output formats:
#
#     RGB24
#     BGR24
#
# YUY2 640x480 is one of the most widely supported formats in webcam capture
# programs. First format defined is the default frame format for
# 'capture'/'output'.
# 'width', 'height' and 'fps' are unsigned integers.
formats/1/format = YUY2
formats/1/width = 640
formats/1/height = 480
formats/1/fps = 30

# The parameters can also be specified as a comma separated list, so it's
# possible to combine the parameters to define several formats in one group.
# 'fps' can also be defined as a fraction.
# The following lines will define 4 formats:
#
#     RGB24 640x480 20  FPS
#     RGB24 640x480 7.5 FPS
#     YUY2  640x480 20  FPS
#     YUY2  640x480 7.5 FPS
formats/2/format = RGB24, YUY2
formats/2/width = 640
formats/2/height = 480
formats/2/fps = 20/1, 15/2

# Finally, to create a fully working virtual camera, you must connect one
# 'output' to one or many 'capture' devices.
# Connections are made by index, separated by a colon. The first index is the
# 'output' device, the following index are 'capture' devices.
[Connections]
connections/size = 2
connections/1/connection = 1:2
connections/2/connection = 1:3
# You can also define a default frame when a 'capture' device is not receiving
# any input. Only 24 bpp and 32 bpp BMP files are supported.
[General]
default_frame = /etc/akvcam/file.bmp

# This config will take effect on modprobe/insmod.`

Dimitri-D-Hoffman, 2021-12-06:
> Did you wrote the setting manually or you let Webcamoid set it for you?. If you wrote it manually, just paste the code here.

wait how do I have it do that

Dimitri-D-Hoffman, 2021-12-06:
you know, I think I know what happened, there was no detected output. Static was just the 'random dot pattern' as stated in the config,

how would I fix this?

Dimitri-D-Hoffman, 2021-12-06:
so good news, I'm no longer being given false-positives that my camera should be working, bad news:
```computer@user:/etc/akvcam/src$ v4l2-compliance -d /dev/video3 -f -s
v4l2-compliance SHA: not available, 64 bits

Compliance test for akvcam device /dev/video3:

Driver Info:
	Driver name      : akvcam
	Card type        : VDaturaCam
	Bus info         : platform:akvcam-3
	Driver version   : 5.11.22
	Capabilities     : 0x85200001
		Video Capture
		Read/Write
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps      : 0x05200001
		Video Capture
		Read/Write
		Streaming
		Extended Pix Format

Required ioctls:
	test VIDIOC_QUERYCAP: OK

Allow for multiple opens:
	test second /dev/video3 open: OK
	test VIDIOC_QUERYCAP: OK
	test VIDIOC_G/S_PRIORITY: OK
	test for unlimited opens: OK

Debug ioctls:
	test VIDIOC_DBG_G/S_REGISTER: OK (Not Supported)
	test VIDIOC_LOG_STATUS: OK

Input ioctls:
	test VIDIOC_G/S_TUNER/ENUM_FREQ_BANDS: OK (Not Supported)
	test VIDIOC_G/S_FREQUENCY: OK (Not Supported)
	test VIDIOC_S_HW_FREQ_SEEK: OK (Not Supported)
	test VIDIOC_ENUMAUDIO: OK (Not Supported)
	test VIDIOC_G/S/ENUMINPUT: OK
	test VIDIOC_G/S_AUDIO: OK (Not Supported)
	Inputs: 1 Audio Inputs: 0 Tuners: 0

Output ioctls:
	test VIDIOC_G/S_MODULATOR: OK (Not Supported)
	test VIDIOC_G/S_FREQUENCY: OK (Not Supported)
	test VIDIOC_ENUMAUDOUT: OK (Not Supported)
	test VIDIOC_G/S/ENUMOUTPUT: OK (Not Supported)
	test VIDIOC_G/S_AUDOUT: OK (Not Supported)
	Outputs: 0 Audio Outputs: 0 Modulators: 0

Input/Output configuration ioctls:
	test VIDIOC_ENUM/G/S/QUERY_STD: OK (Not Supported)
	test VIDIOC_ENUM/G/S/QUERY_DV_TIMINGS: OK (Not Supported)
	test VIDIOC_DV_TIMINGS_CAP: OK (Not Supported)
	test VIDIOC_G/S_EDID: OK (Not Supported)

Control ioctls (Input 0):
	test VIDIOC_QUERY_EXT_CTRL/QUERYMENU: OK
	test VIDIOC_QUERYCTRL: OK
	test VIDIOC_G/S_CTRL: OK
	test VIDIOC_G/S/TRY_EXT_CTRLS: OK
	test VIDIOC_(UN)SUBSCRIBE_EVENT/DQEVENT: OK
	test VIDIOC_G/S_JPEGCOMP: OK (Not Supported)
	Standard Controls: 9 Private Controls: 0

Format ioctls (Input 0):
	test VIDIOC_ENUM_FMT/FRAMESIZES/FRAMEINTERVALS: OK
[... 254 lines cut ...]
		fail: v4l2-test-buffers.cpp(2556): fmt.g_sizeimage() <= size
		fail: v4l2-test-buffers.cpp(2336): q.reqbufs(node, 3)
		Stride 1344, Field None: FAIL   
		fail: v4l2-test-buffers.cpp(2336): q.reqbufs(node, 3)
		Stride 0, Field Sequential Bottom-Top: FAIL   
		fail: v4l2-test-buffers.cpp(2556): fmt.g_sizeimage() <= size
		fail: v4l2-test-buffers.cpp(2336): q.reqbufs(node, 3)
		Stride 1344, Field None: FAIL   
		fail: v4l2-test-buffers.cpp(2336): q.reqbufs(node, 3)
		Stride 0, Field Alternating: FAIL   
		fail: v4l2-test-buffers.cpp(2556): fmt.g_sizeimage() <= size
		fail: v4l2-test-buffers.cpp(2336): q.reqbufs(node, 3)
		Stride 1344, Field None: FAIL   
		fail: v4l2-test-buffers.cpp(2336): q.reqbufs(node, 3)
		Stride 0, Field Interlaced Top-Bottom: FAIL   
		fail: v4l2-test-buffers.cpp(2556): fmt.g_sizeimage() <= size
		fail: v4l2-test-buffers.cpp(2336): q.reqbufs(node, 3)
		Stride 1344, Field None: FAIL   
		fail: v4l2-test-buffers.cpp(2336): q.reqbufs(node, 3)
		Stride 0, Field Interlaced Bottom-Top: FAIL   
		fail: v4l2-test-buffers.cpp(2556): fmt.g_sizeimage() <= size
		fail: v4l2-test-buffers.cpp(2336): q.reqbufs(node, 3)
		Stride 1344, Field None: FAIL   
Total for akvcam device /dev/video3: 143, Succeeded: 44, Failed: 99, Warnings: 0```
I don't know how to read this error message, just wanted to show it to you if you knew what's causing issues.

Dimitri-D-Hoffman, 2021-12-06:
after going through issue 296 on webcamoid's issue page I think I fixed it, at least to my liking. No idea wtf I did.

hipersayanX, 2022-10-02:
No answer, not reproducible, and the issue is too old, closing.

closed 2022-10-02 by hipersayanX