webcamoid/akvcam #59: akvcam-1.2.2: build warnings

joakim-tjernlund, 2022-03-17:
```
In file included from /var/tmp/portage/media-video/akvcam-1.2.2/work/akvcam-1.2.2/src/list.h:25,
                 from /var/tmp/portage/media-video/akvcam-1.2.2/work/akvcam-1.2.2/src/settings.c:26:
/var/tmp/portage/media-video/akvcam-1.2.2/work/akvcam-1.2.2/src/settings.c: In function ‘akvcam_settings_begin_array’:
/var/tmp/portage/media-video/akvcam-1.2.2/work/akvcam-1.2.2/src/utils.h:157:15: warning: ‘strnlen’ specified bound 1024 exceeds source size 6 [-Wstringop-overread]
  157 |         len = strnlen(str, AKVCAM_MAX_STRING_SIZE); \
      |               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/var/tmp/portage/media-video/akvcam-1.2.2/work/akvcam-1.2.2/src/settings.c:229:28: note: in expansion of macro ‘akvcam_strlen’
  229 |     array_key_size = len + akvcam_strlen("/size") + 1;
      |                            ^~~~~~~~~~~~~
```

ohquait, 2022-06-06:
and a other warning about cast between incompatible function types from

DKMS make.log for akvcam-1.2.2-1pclos2022 for kernel 5.17.11-pclos1 (x86_64)
Mon Jun  6 10:11:30 AM UTC 2022
make -C /lib/modules/5.17.11-pclos1/build M=/var/lib/dkms/akvcam/1.2.2-1pclos2022/build  modules
make[1]: Entering directory '/usr/src/kernel-devel-5.17.11-pclos1'
  CC [M]  /var/lib/dkms/akvcam/1.2.2-1pclos2022/build/module.o
  CC [M]  /var/lib/dkms/akvcam/1.2.2-1pclos2022/build/attributes.o
  CC [M]  /var/lib/dkms/akvcam/1.2.2-1pclos2022/build/buffers.o
  CC [M]  /var/lib/dkms/akvcam/1.2.2-1pclos2022/build/controls.o
  CC [M]  /var/lib/dkms/akvcam/1.2.2-1pclos2022/build/device.o
In file included from /var/lib/dkms/akvcam/1.2.2-1pclos2022/build/buffers.h:28,
                 from /var/lib/dkms/akvcam/1.2.2-1pclos2022/build/device.c:30:
/var/lib/dkms/akvcam/1.2.2-1pclos2022/build/device.c: In function ‘akvcam_device_new’:
/var/lib/dkms/akvcam/1.2.2-1pclos2022/build/utils.h:91:36: warning: cast between incompatible function types from ‘void (*)(struct akvcam_device *, __u32,  __s32)’ {aka ‘void (*)(struct akvcam_device *, unsigned int,  int)’} to ‘int (*)(void *, __u32,  __s32)’ {aka ‘int (*)(void *, unsigned int,  int)’} [-Wcast-function-type]
   91 |         signal_callback.callback = (akvcam_##class##_##signal##_proc) method; \
      |                                    ^
/var/lib/dkms/akvcam/1.2.2-1pclos2022/build/device.c:138:5: note: in expansion of macro ‘akvcam_connect’
  138 |     akvcam_connect(controls, self->controls, updated, self, akvcam_device_controls_updated);
      |     ^~~~~~~~~~~~~~
/var/lib/dkms/akvcam/1.2.2-1pclos2022/build/utils.h:91:36: warning: cast between incompatible function types from ‘void (*)(struct akvcam_device *)’ to ‘int (*)(void *)’ [-Wcast-function-type]
   91 |         signal_callback.callback = (akvcam_##class##_##signal##_proc) method; \
      |                                    ^
/var/lib/dkms/akvcam/1.2.2-1pclos2022/build/device.c:140:5: note: in expansion of macro ‘akvcam_connect’
  140 |     akvcam_connect(buffers, self->buffers, streaming_stopped, self, akvcam_device_stop_streaming);
      |     ^~~~~~~~~~~~~~
  CC [M]  /var/lib/dkms/akvcam/1.2.2-1pclos2022/build/driver.o
  CC [M]  /var/lib/dkms/akvcam/1.2.2-1pclos2022/build/file_read.o
  CC [M]  /var/lib/dkms/akvcam/1.2.2-1pclos2022/build/format.o
  CC [M]  /var/lib/dkms/akvcam/1.2.2-1pclos2022/build/frame.o
  CC [M]  /var/lib/dkms/akvcam/1.2.2-1pclos2022/build/frame_filter.o
  CC [M]  /var/lib/dkms/akvcam/1.2.2-1pclos2022/build/ioctl.o
  CC [M]  /var/lib/dkms/akvcam/1.2.2-1pclos2022/build/list.o
  CC [M]  /var/lib/dkms/akvcam/1.2.2-1pclos2022/build/log.o
  CC [M]  /var/lib/dkms/akvcam/1.2.2-1pclos2022/build/map.o
  CC [M]  /var/lib/dkms/akvcam/1.2.2-1pclos2022/build/rbuffer.o
  CC [M]  /var/lib/dkms/akvcam/1.2.2-1pclos2022/build/settings.o
  CC [M]  /var/lib/dkms/akvcam/1.2.2-1pclos2022/build/utils.o
  LD [M]  /var/lib/dkms/akvcam/1.2.2-1pclos2022/build/akvcam.o
  MODPOST /var/lib/dkms/akvcam/1.2.2-1pclos2022/build/Module.symvers
  CC [M]  /var/lib/dkms/akvcam/1.2.2-1pclos2022/build/akvcam.mod.o
  LD [M]  /var/lib/dkms/akvcam/1.2.2-1pclos2022/build/akvcam.ko
make[1]: Leaving directory '/usr/src/kernel-devel-5.17.11-pclos1'

hipersayanX, 2022-10-02:
This warnings were fixed in 1.2.3, closing

closed 2022-10-02 by hipersayanX