webcamoid/akvcam #61: akvcam module compilation

darkonen, 2022-04-22:
akvcam module compilation failed in kernel 5.18-rc3
[make.log](https://github.com/webcamoid/akvcam/files/8538735/make.log)

ohquait, 2022-06-07:
on kernel-5.18.2-pclos1 compilation works, but a other issue come up:

# modprobe akvcam
modprobe: ERROR: could not insert 'akvcam': Exec format error

ohquait, 2022-06-07:
SOLVED for PCLinuxOS 
a patch to /src/MAKEFILE was needed to enable rightly working dkms functions on this special distribution

hipersayanX, 2022-06-11:
@ohquait which patches did you applied?

ohquait, 2022-06-14:
@hipersayanX 
i use this:
link:
https://www.qua-it.org/XFCE/shots/share-with-pclos/pclos/akvcam-Makefile-pclos.patch

ohquait, 2022-06-14:
have make reply on your git, how can post code in raw there? even i will
post a patch the format of strings will breaks
...
so i post you the links to patch and resulting patched Makefile
Greetings

darkonen, 2022-06-14:
I have patched Makefile with this patch only on kernel 5.18.3.
make => compilation failed (cf: make_5.18.3.log)
then compilation with DKMS on not-patched kernel 5.19.0-rc2.
same errors  (cf: make_5.19.0-rc2.log)

[make_5.18.3.log](https://github.com/webcamoid/akvcam/files/8900364/make_5.18.3.log)
[make_5.19.0-rc2.log](https://github.com/webcamoid/akvcam/files/8900366/make_5.19.0-rc2.log)

ohquait, 2022-06-14:
here is the dkms.conf from pclos...
with it all should works:
link:
https://www.qua-it.org/XFCE/shots/share-with-pclos/pclos/akvcam-pclos-dkms.conf
in raw:

PACKAGE_NAME="akvcam"
PACKAGE_VERSION="1.2.2-4pclos2022"
BUILT_MODULE_NAME[0]="akvcam"
DEST_MODULE_LOCATION[0]="/kernel/3rdparty/akvcam"
MAKE[0]="export IGNORE_CC_MISMATCH=1; 'make' clean SKIP_STACK_VALIDATION=1
all"
CLEAN="make clean"
AUTOINSTALL="yes"

ohquait, 2022-06-14:
here is the dkms.conf from pclos...with it all should works:link:
https://www.qua-it.org/XFCE/shots/share-with-pclos/pclos/akvcam-pclos-dkms.conf

in raw:
PACKAGE_NAME="akvcam"
PACKAGE_VERSION="1.2.2-4pclos2022"
BUILT_MODULE_NAME[0]="akvcam"
DEST_MODULE_LOCATION[0]="/kernel/3rdparty/akvcam"
MAKE[0]="export IGNORE_CC_MISMATCH=1; 'make' clean SKIP_STACK_VALIDATION=1 all"
CLEAN="make clean"
AUTOINSTALL="yes"

ohquait, 2022-06-15:
ups, i must do it again,
the patch not solving the problem, i have a pre installed version on differ place,
now i clean my kernel-tree and all dkms from tree and begin a newer build,
if i can check success by install on kernel-5.17.... and then boot in kernel 5.18.3... and the command
dkms status and a modprobe akvcam  give my not errors, then i post here ....

ohquait, 2022-06-18:
for me can close

hipersayanX, 2022-06-18:
@ohquait

>for me can close

Don't, I must still check what I can take from the patch.

darkonen, 2022-10-02:
> akvcam module compilation failed in kernel 5.18-rc3 [make.log](https://github.com/webcamoid/akvcam/files/8538735/make.log)

closed 2022-10-02 by darkonen

darkonen, 2022-10-02:
Hello,

it's OK now for me

akvcam/1.2.3, 5.19.12, x86_64: installed
akvcam/1.2.3, 6.0.0-rc7, x86_64: installed
v4l2loopback/0.12.7, 5.19.12, x86_64: installed
v4l2loopback/0.12.7, 6.0.0-rc7, x86_64: installed
virtualbox/6.1.38, 5.19.12, x86_64: installed
virtualbox/6.1.38, 6.0.0-rc7, x86_64: installed

darkonen, 2022-10-11:
> now it's working
> build as rpm dkms-akvcam on PCLinuxOS kernel-5.17.15, upload to my repo,
> then install with: apt-get update ; apt-get install dkms-akvcam
> no errors during modules build, reboot and : lsmod | grep akvcam
> akvcam 192512 0
> videobuf2_vmalloc 16384 1 akvcam
> videobuf2_v4l2 36864 1 akvcam
> videobuf2_common 69632 4
> videobuf2_vmalloc,videobuf2_v4l2,akvcam,videobuf2_memops
> videodev 270336 3 videobuf2_v4l2,videobuf2_common,akvcam
>
> now reboot to kernel-5.18.4
> and wait a few minutes, dkms automatical build for this kernel, if
> ready reboot,
> and : lsmod | grep akvcam
> akvcam 192512 0
> videobuf2_vmalloc 16384 1 akvcam
> videobuf2_v4l2 36864 1 akvcam
> videobuf2_common 69632 4
> videobuf2_vmalloc,videobuf2_v4l2,akvcam,videobuf2_memops
> videodev 270336 3 videobuf2_v4l2,videobuf2_common,akvcam
>
> no Exec format errors by do # modprobe akvcam
> READY
>
> [... 44 lines cut ...]
>
> PCLinuxOS still keep older version but good configured dkms-stuff
> without systemd's
>
> Hope this will help to compile the modules on other distri's

it's not working on my Debian/Sid distribution.

What's your gcc version (may be the problem)?

I made compilation with gcc 11.3.0 version

ohquait, 2022-10-11:
just to inform, all seem to be good with 1.2.3 release:
for check:
[https://www.qua-it.org/XFCE/repo/SRPMS.quait/SRPMS.x86_64/dkms-akvcam-1.2.3-1pclos2022.src.rpm](url)