webcamoid/akvcam #64: [incompatibility] akvcam virtual cam generating errors on web platforms like omegle,zoom,jitsi etc

Arpit-exe, 2022-07-12:
[image]
[image]
  the webcamoid app uses the akvcam driver to emulate a virtual usb device which is now not detected on omegle {error} the later versions of webcamoid dont install the akvcam driver implicitly 

the new webcamoid app should easily integrate Akvcam driver

Arpit-exe, 2022-07-12:
[image]

hipersayanX, 2022-07-18:
You must try with the latest version of Webcamoid and the latesr version of the driver. Also, did you did something special? why the video device appears as ```/akvcam/video1``` instead of ```/dev/video1```?