webcamoid/akvcam #65: akvcam - gone after reboiot.

AndKe, 2022-09-09:
webcamoid-portable-linux-9.0.0-x86_64.AppImage offers to install akvcam - I do, restart the app - it works.
After a reboot, webcamoid says the "virtual camera is not installed"   - attempt to install it will fail as the /opt/akvcam already exists.

How do I properly troubleshoot it?

hipersayanX, 2022-09-09:
You must create the akvcam.conf file in /etc/modules-load.d, check [here](https://github.com/webcamoid/akvcam/wiki/Usage-and-examples#load-on-boot).

AndKe, 2022-09-09:
hmm, do you have any other suggestions?
```
$ sudo echo akvcam > /etc/modules-load.d/akvcam.conf
bash: /etc/modules-load.d/akvcam.conf: Permission denied
$ sudo ls -la /etc/modules-load.d/
total 20
drwxr-xr-x   2 root root  4096 juli  29 20:50 .
drwxr-xr-x 152 root root 12288 sep.   9 08:53 ..
-rw-r--r--   1 root root   119 april 11 22:19 cups-filters.conf
lrwxrwxrwx   1 root root    10 juni  27 20:28 modules.conf -> ../modules
```

To complete the friendly output module install experience, the load-after--boot should have been implicit.

hipersayanX, 2022-09-10:
I've updated the wiki, you must use

```
echo akvcam | sudo tee /etc/modules-load.d/akvcam.conf
```

>To complete the friendly output module install experience, the load-after--boot should have been implicit.

Isn't possible because the module has not any device or format configured by default, even if the module is loaded on boot it won't show any device.

AndKe, 2022-09-11:
the installer could tell user to do the "echo akvcam | sudo tee /etc/modules-load.d/akvcam.conf" ?

AndKe, 2022-09-16:
@hipersayanX  - my conf file is ok, 
```
$ cat /etc/modules-load.d/akvcam.conf
akvcam
$
```

...but the device is still gone after reboot - any other suggestions?

hipersayanX, 2022-09-16:
Give me the output of the following commands:

```
uname -a
cat /etc/lsb-release
cat /etc/os-release
```

Also try loading the driver as:

```
sudo modprobe akvcam loglevel=7
```

Then give me the output of:

```
dmesg | tail -200
```

AndKe, 2022-09-16:
Please see https://gist.github.com/AndKe/f0e3719838756eb5525ff00a5b3b6bf6

AndKe, 2022-09-16:
That's kind of the issue here... I did install the driver (I let the webcamoid-portable-linux-9.0.0-x86_64.AppImage  do the job)

AndKe, 2022-09-16:
Please see how build/install from git succeeds, then fail the "tests"

```
andre@Loke:~/Downloads/akvcam-master/src$ sudo make dkms_install
[sudo] password for andre: 
dkms remove akvcam/1.2.2 --all
Module akvcam-1.2.2 for kernel 5.15.0-46-generic (x86_64).
Before uninstall, this module version was ACTIVE on this kernel.

akvcam.ko:
 - Uninstallation
   - Deleting from: /lib/modules/5.15.0-46-generic/updates/dkms/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

depmod...
Deleting module akvcam-1.2.2 completely from the DKMS tree.
rm -rvf /usr/src/akvcam-1.2.2
removed '/usr/src/akvcam-1.2.2'
rm -rvf /usr/src/akvcam-1.2.2
mkdir -p /usr/src/akvcam-1.2.2
cp -f Makefile /usr/src/akvcam-1.2.2
cp -f dkms.conf /usr/src/akvcam-1.2.2
cp -f *.h /usr/src/akvcam-1.2.2
cp -f *.c /usr/src/akvcam-1.2.2
dkms install akvcam/1.2.2
Creating symlink /var/lib/dkms/akvcam/1.2.2/source -> /usr/src/akvcam-1.2.2

Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area...
make -j20 KERNELRELEASE=5.15.0-47-generic KERNEL_DIR=/lib/modules/5.15.0-47-generic/build all...
Signing module:
 - /var/lib/dkms/akvcam/1.2.2/5.15.0-47-generic/x86_64/module/akvcam.ko
Secure Boot not enabled on this system.
cleaning build area...

akvcam.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.15.0-47-generic/updates/dkms/

depmod...
andre@Loke:~/Downloads/akvcam-master/src$ ls /lib/modules/$(uname -r)/extra/akvcam.ko*
ls: cannot access '/lib/modules/5.15.0-47-generic/extra/akvcam.ko*': No such file or directory
andre@Loke:~/Downloads/akvcam-master/src$ sudo !!
sudo ls /lib/modules/$(uname -r)/extra/akvcam.ko*
ls: cannot access '/lib/modules/5.15.0-47-generic/extra/akvcam.ko*': No such file or directory
andre@Loke:~/Downloads/akvcam-master/src$ sudo modprobe akvcam loglevel=7
andre@Loke:~/Downloads/akvcam-master/src$ sudo ls /lib/modules/$(uname -r)/extra/akvcam.ko*
ls: cannot access '/lib/modules/5.15.0-47-generic/extra/akvcam.ko*': No such file or directory
andre@Loke:~/Downloads/akvcam-master/src$ ls /etc/akvcam
ls: cannot access '/etc/akvcam': No such file or directory
andre@Loke:~/Downloads/akvcam-master/src$ 
```

hipersayanX, 2022-09-16:
>andre@Loke:~/Downloads/akvcam-master/src$ sudo modprobe akvcam loglevel=7

The driver is already installed, here

>   - Installing to /lib/modules/5.15.0-47-generic/updates/dkms/

That's why is showing 

```
andre@Loke:~/Downloads/akvcam-master/src$ ls /lib/modules/$(uname -r)/extra/akvcam.ko*
ls: cannot access '/lib/modules/5.15.0-47-generic/extra/akvcam.ko*': No such file or directory
```

Anyway, you must now create the the file ```/etc/akvcam/config.ini``` with the following content

```
[Cameras]
cameras/size = 2
cameras/1/type = output
cameras/1/mode = mmap, userptr, rw
cameras/1/description = Virtual Camera (output device)
cameras/1/formats = 2
cameras/1/videonr = 7

cameras/2/type = capture
cameras/2/mode = mmap, rw
cameras/2/description = Virtual Camera
cameras/2/formats = 1, 2
cameras/1/videonr = 5

[Formats]
formats/size = 2
formats/1/format = YUY2
formats/1/width = 640
formats/1/height = 480
formats/1/fps = 30

formats/2/format = RGB24, YUY2
formats/2/width = 640
formats/2/height = 480
formats/2/fps = 20/1, 15/2

[Connections]
connections/size = 1
connections/1/connection = 1:2
```

Then 

```
sudo rmmod akvcam
sudo modprobe akvcam loglevel=7
```

and now you must be able to see ```/dev/video5``` and ```/dev/video7```.

AndKe, 2022-09-16:
Thank you - I guess this is almost a success, except for video7 which is ...no there.
```
$ cat  /etc/akvcam/config.ini
[Cameras]
cameras/size = 2
cameras/1/type = output
cameras/1/mode = mmap, userptr, rw
cameras/1/description = Virtual Camera (output device)
cameras/1/formats = 2
cameras/1/videonr = 7

cameras/2/type = capture
cameras/2/mode = mmap, rw
cameras/2/description = Virtual Camera
cameras/2/formats = 1, 2
cameras/1/videonr = 5

[Formats]
formats/size = 2
formats/1/format = YUY2
formats/1/width = 640
formats/1/height = 480
formats/1/fps = 30

formats/2/format = RGB24, YUY2
formats/2/width = 640
formats/2/height = 480
formats/2/fps = 20/1, 15/2

[Connections]
connections/size = 1
connections/1/connection = 1:2
andre@Loke:~/Downloads/akvcam-master/src$ sudo rmmod akvcam
sudo modprobe akvcam loglevel=7
andre@Loke:~/Downloads/akvcam-master/src$ ls /dev/video
video0  video1  video2  video5  
```

hipersayanX, 2022-09-16:
```dmesg | tail -200``` gives you any output?

AndKe, 2022-09-17:
I also tried the last 2000 lines, but the log is filled by the same messages: (most likely while I used the webcamoid)
https://gist.github.com/AndKe/716c63b8711f77c012f386abef7cad00

grep shows no mention of video7 in dmesg output

hipersayanX, 2022-09-23:
Sorry to be late, I was working and did not checked the notifications before. Don't run the virtual camera, you must just reload the driver and soon after run ```dmesg | tail -200```, it may show all the virtual cameras and their features at very beginning.

AndKe, 2022-09-24:
@hipersayanX  - no problem! , I can't remember to have paid for a service level agreement for your support.  :)
please see: https://gist.github.com/AndKe/1f6032478ef5e3b10bce482a44c21efd

hipersayanX, 2022-09-25:
>@hipersayanX - no problem! , I can't remember to have paid for a service level agreement for your support. :)

I would have to charge x100 for the service :laughing: :stuck_out_tongue:

>please see: https://gist.github.com/AndKe/1f6032478ef5e3b10bce482a44c21efd

Oh, it's my fault, I did a quick copy pasting and did not checked it before. In the config.ini, where it say ```cameras/1/videonr = 5```, change it for ```cameras/2/videonr = 5```, reload the driver, and now you must see /dev/video5 and /dev/video7.
If you go to Webcamoid, you must see /dev/video5 as an output.

If everything is working ok, try reverting back everything: uninstall the driver, delete /etc/akvcam and /etc/modules-load.d/akvcam.conf, and try installing the driver again from Webcamoid, if the installation still fails with a clean environment, I will mark it as a bug.
Also don't upgrade your system just before you install the driver. otherwise dkms will fail installing it (it's a normal behavior when upgrading the kernel), the other option is upgrade your system, reboot and then install the driver.

AndKe, 2022-09-26:
I think you can safely mark it a bug...
I did this to clean up:

sudo rmmod akvcam
sudo rm -R /etc/akvcam
sudo rm /etc/modules-load.d/akvcam.conf
sudo rm -R (opt/akvcam

then I rebooted, and then executed as user:
akvcam-linux-1.2.2.run  (during the installation - I got asked for sudo pw)  - just as if the install is invoked by webcamoid

Then:
sudo modprobe akvcam loglevel=7

dmesg shows the same failure...
[  204.006338] [akvcam] driver.c(78): akvcam_driver_init()
[  204.006339] [akvcam] driver.c(83): Initializing driver
[  204.007970] [akvcam] driver.c(90): Reading settings
[  204.007974] [akvcam] file_read.c(107): No such file or directory (ENOENT)
[  204.007975] [akvcam] settings.c(114): Can't open settings file: /etc/akvcam/config.ini
[  204.007976] [akvcam] driver.c(102): Error reading settings
[  204.007977] [akvcam] driver.c(623): No devices found

This is a rather "normal" Ubuntu 22.04

hipersayanX, 2022-09-27:
>I think you can safely mark it a bug...

Are you sure?

[image]

>then I rebooted, and then executed as user:
>akvcam-linux-1.2.2.run (during the installation - I got asked for sudo pw) - just as if the install is invoked by webcamoid

This is what you did:

>Then:
>sudo modprobe akvcam loglevel=7

This is what Webcamoid does:

https://github.com/webcamoid/webcamoid/blob/6093d8cc68592f3e35c68c7a1812d89ba96a9f95/libAvKys/Plugins/VirtualCamera/src/akvcam/src/vcamak.cpp#L444

You are missing like 250 lines of code there :laughing: 

>dmesg shows the same failure...
>[ 204.006338] [akvcam] driver.c(78): akvcam_driver_init()
>[ 204.006339] [akvcam] driver.c(83): Initializing driver
>[ 204.007970] [akvcam] driver.c(90): Reading settings
>[ 204.007974] [akvcam] file_read.c(107): No such file or directory (ENOENT)
>[ 204.007975] [akvcam] settings.c(114): Can't open settings file: /etc/akvcam/config.ini
>[ 204.007976] [akvcam] driver.c(102): Error reading settings
>[ 204.007977] [akvcam] driver.c(623): No devices found
>
>This is a rather "normal" Ubuntu 22.04
>there is no such thing as "/etc/akvcam" - please check if you are trying to create it...

Yes, and that is the normal behavior because just installing the drivers as-is does not provides any device or format by default, you have not created the /etc/akvcam/config.ini and /etc/modules-load.d/akvcam.conf files.
Webcamoid does a lot more than just downloading and loading the driver.

Again, **revert back everything**, and **install the driver using Webcamoid**, if the installation still fails with a clean environment, I will mark it as a bug.

AndKe, 2022-09-27:
so ..I thought it was clever to debug one thing at a time - and failed.
webcamoid-portable-linux-9.0.0-x86_64.AppImage does no longer offer to install the driver - 
```
sudo rmmod akvcam
sudo rm -R /etc/akvcam
sudo rm /etc/modules-load.d/akvcam.conf
sudo rm -R /opt/akvcam
```
- what else needs to be done in order to get it to suggest the installation?

hipersayanX, 2022-09-28:
The problem is that you are bypassing every single step of installation and uninstallation :sweat: 
Webcamoid is still detecting the driver because, even if you have removed /opt/akvcam, [the driver is still installed in the system](https://github.com/webcamoid/webcamoid/blob/6093d8cc68592f3e35c68c7a1812d89ba96a9f95/libAvKys/Plugins/VirtualCamera/src/akvcam/src/vcamak.cpp#L212). For properly uninstalling the driver, you must do:

```
sudo rmmod akvcam
sudo rm -R /etc/akvcam
sudo rm /etc/modules-load.d/akvcam.conf
/opt/akvcam/akvcamUninstall
```

AndKe, 2022-09-28:
after proper removal, I let the webcamoid install it again, and rebooted.
After reboot, this is what I get when I try to add camera:

```
[2022-09-28 22:44:31.140, Webcamoid, 0x7f01af678640,  (0)] debug: Camera input frame format: AkCaps(mimeType="video/unknown",fourcc=QVariant(QString, "YUYV"),fps=QVariant(QString, "30/1"),height=QVariant(uint, 480),width=QVariant(uint, 640))
[2022-09-28 22:44:37.156, Webcamoid, 0x7f01fbb857c0,  (0)] debug: pkexec --version |
       --help |
       --disable-internal-agent |
       [--user username] PROGRAM [ARGUMENTS...]

See the pkexec manual page for more details.

[2022-09-28 22:45:00.522, Webcamoid, 0x7f01fbb857c0, VideoOutputAddEdit.qml (208)] warning: qrc:/Webcamoid/share/qml/VideoOutputAddEdit.qml:208: TypeError: Cannot read property 'right' of null
[2022-09-28 22:45:00.522, Webcamoid, 0x7f01fbb857c0, VideoOutputAddEdit.qml (209)] warning: qrc:/Webcamoid/share/qml/VideoOutputAddEdit.qml:209: TypeError: Cannot read property 'left' of null
[2022-09-28 22:45:00.522, Webcamoid, 0x7f01fbb857c0, VideoOutputAddEdit.qml (208)] warning: qrc:/Webcamoid/share/qml/VideoOutputAddEdit.qml:208: TypeError: Cannot read property 'right' of null
[2022-09-28 22:45:00.522, Webcamoid, 0x7f01fbb857c0, VideoOutputAddEdit.qml (209)] warning: qrc:/Webcamoid/share/qml/VideoOutputAddEdit.qml:209: TypeError: Cannot read property 'left' of null
[2022-09-28 22:45:00.522, Webcamoid, 0x7f01fbb857c0, VideoOutputAddEdit.qml (208)] warning: qrc:/Webcamoid/share/qml/VideoOutputAddEdit.qml:208: TypeError: Cannot read property 'right' of null
[2022-09-28 22:45:00.522, Webcamoid, 0x7f01fbb857c0, VideoOutputAddEdit.qml (209)] warning: qrc:/Webcamoid/share/qml/VideoOutputAddEdit.qml:209: TypeError: Cannot read property 'left' of null
[2022-09-28 22:45:00.522, Webcamoid, 0x7f01fbb857c0, VideoOutputAddEdit.qml (208)] warning: qrc:/Webcamoid/share/qml/VideoOutputAddEdit.qml:208: TypeError: Cannot read property 'right' of null
[2022-09-28 22:45:00.522, Webcamoid, 0x7f01fbb857c0, VideoOutputAddEdit.qml (209)] warning: qrc:/Webcamoid/share/qml/VideoOutputAddEdit.qml:209: TypeError: Cannot read property 'left' of null
[2022-09-28 22:45:00.522, Webcamoid, 0x7f01fbb857c0, VideoOutputAddEdit.qml (208)] warning: qrc:/Webcamoid/share/qml/VideoOutputAddEdit.qml:208: TypeError: Cannot read property 'right' of null
[2022-09-28 22:45:00.522, Webcamoid, 0x7f01fbb857c0, VideoOutputAddEdit.qml (209)] warning: qrc:/Webcamoid/share/qml/VideoOutputAddEdit.qml:209: TypeError: Cannot read property 'left' of null
[2022-09-28 22:45:00.522, Webcamoid, 0x7f01fbb857c0, VideoOutputAddEdit.qml (208)] warning: qrc:/Webcamoid/share/qml/VideoOutputAddEdit.qml:208: TypeError: Cannot read property 'right' of null
[2022-09-28 22:45:00.522, Webcamoid, 0x7f01fbb857c0, VideoOutputAddEdit.qml (209)] warning: qrc:/Webcamoid/share/qml/VideoOutputAddEdit.qml:209: TypeError: Cannot read property 'left' of null
[2022-09-28 22:45:00.522, Webcamoid, 0x7f01fbb857c0, VideoOutputAddEdit.qml (208)] warning: qrc:/Webcamoid/share/qml/VideoOutputAddEdit.qml:208: TypeError: Cannot read property 'right' of null
[2022-09-28 22:45:00.522, Webcamoid, 0x7f01fbb857c0, VideoOutputAddEdit.qml (209)] warning: qrc:/Webcamoid/share/qml/VideoOutputAddEdit.qml:209: TypeError: Cannot read property 'left' of null
[2022-09-28 22:45:00.522, Webcamoid, 0x7f01fbb857c0, VideoOutputAddEdit.qml (208)] warning: qrc:/Webcamoid/share/qml/VideoOutputAddEdit.qml:208: TypeError: Cannot read property 'right' of null
[2022-09-28 22:45:00.522, Webcamoid, 0x7f01fbb857c0, VideoOutputAddEdit.qml (209)] warning: qrc:/Webcamoid/share/qml/VideoOutputAddEdit.qml:209: TypeError: Cannot read property 'left' of null
[2022-09-28 22:45:00.522, Webcamoid, 0x7f01fbb857c0, VideoOutputAddEdit.qml (208)] warning: qrc:/Webcamoid/share/qml/VideoOutputAddEdit.qml:208: TypeError: Cannot read property 'right' of null
[2022-09-28 22:45:00.522, Webcamoid, 0x7f01fbb857c0, VideoOutputAddEdit.qml (209)] warning: qrc:/Webcamoid/share/qml/VideoOutputAddEdit.qml:209: TypeError: Cannot read property 'left' of null
[2022-09-28 22:45:00.522, Webcamoid, 0x7f01fbb857c0, VideoOutputAddEdit.qml (208)] warning: qrc:/Webcamoid/share/qml/VideoOutputAddEdit.qml:208: TypeError: Cannot read property 'right' of null
[2022-09-28 22:45:00.522, Webcamoid, 0x7f01fbb857c0, VideoOutputAddEdit.qml (209)] warning: qrc:/Webcamoid/share/qml/VideoOutputAddEdit.qml:209: TypeError: Cannot read property 'left' of null
[2022-09-28 22:45:00.522, Webcamoid, 0x7f01fbb857c0, VideoOutputAddEdit.qml (208)] warning: qrc:/Webcamoid/share/qml/VideoOutputAddEdit.qml:208: TypeError: Cannot read property 'right' of null
[2022-09-28 22:45:00.522, Webcamoid, 0x7f01fbb857c0, VideoOutputAddEdit.qml (209)] warning: qrc:/Webcamoid/share/qml/VideoOutputAddEdit.qml:209: TypeError: Cannot read property 'left' of null
[2022-09-28 22:45:00.522, Webcamoid, 0x7f01fbb857c0, VideoOutputAddEdit.qml (208)] warning: qrc:/Webcamoid/share/qml/VideoOutputAddEdit.qml:208: TypeError: Cannot read property 'right' of null
[2022-09-28 22:45:00.522, Webcamoid, 0x7f01fbb857c0, VideoOutputAddEdit.qml (209)] warning: qrc:/Webcamoid/share/qml/VideoOutputAddEdit.qml:209: TypeError: Cannot read property 'left' of null
[2022-09-28 22:45:00.522, Webcamoid, 0x7f01fbb857c0, VideoOutputAddEdit.qml (208)] warning: qrc:/Webcamoid/share/qml/VideoOutputAddEdit.qml:208: TypeError: Cannot read property 'right' of null
[... cut ...]
[  774.956212] [akvcam] driver.c(83): Initializing driver
[  774.957706] [akvcam] driver.c(90): Reading settings
[  774.957710] [akvcam] file_read.c(107): No such file or directory (ENOENT)
[  774.957711] [akvcam] settings.c(114): Can't open settings file: /etc/akvcam/config.ini
[  774.957712] [akvcam] driver.c(102): Error reading settings
[  774.957713] [akvcam] driver.c(623): No devices found
andre@Loke:~/Downloads$ 

```

So ... this is all installed by the appimage  - and the result is the same. the /etc/akv* does not exist.

hipersayanX, 2022-09-28:
And that was the reason because the camera was not installed in your case, It's a known bug (https://github.com/webcamoid/webcamoid/issues/510) and should have been [fixed in the Daily build](https://github.com/webcamoid/webcamoid/pull/540). The virtual camera itself is working fine as probed before, but it was a bug in Webcamoid, not in akvcam.

AndKe, 2022-09-29:
I see, by any chance - is there a daily appimage somewhere?

hipersayanX, 2022-09-29:
[Yes](https://github.com/webcamoid/webcamoid/releases/tag/daily-build)

AndKe, 2022-09-30:
the .appimage builds are very old
I also tried to build from source (which builds fine)  - but fails to run due to dependencies, some of which are Qt related - as I already have Qt on the computer and other projects are happy with it - I would prefer an appimage/snap.

hipersayanX, 2022-09-30:
>I also tried to build from source (which builds fine) - but fails to run due to dependencies, some of which are Qt related - as I already have Qt on the computer and other projects are happy with it

These are the dependencies for building and running Webcamoid:

```
sudo apt-get install \
    clang \
    cmake \
    g++ \
    git \
    gstreamer1.0-plugins-base \
    gstreamer1.0-plugins-good \
    libasound2-dev \
    libavcodec-dev \
    libavdevice-dev \
    libavformat-dev \
    libavutil-dev \
    libgl1-mesa-dev \
    libgstreamer-plugins-base1.0-dev \
    libgstreamer1.0-0 \
    libjack-dev \
    libkmod-dev \
    libpipewire-0.3-dev \
    libpulse-dev \
    libswresample-dev \
    libswscale-dev \
    libusb-dev \
    libuvc-dev \
    libv4l-dev \
    libvlc-dev \
    libvlccore-dev \
    linux-libc-dev \
    make \
    pkg-config \
    vlc-plugin-base \
    libqt5opengl5-dev \
    libqt5svg5-dev \
    qml-module-qt-labs-folderlistmodel \
    qml-module-qt-labs-platform \
    qml-module-qt-labs-settings \
    qml-module-qtqml-models2 \
    qml-module-qtquick-controls2 \
    qml-module-qtquick-dialogs \
    qml-module-qtquick-extras \
    qml-module-qtquick-privatewidgets \
    qml-module-qtquick-templates2 \
    qt5-qmake \
    qtdeclarative5-dev \
    qtquickcontrols2-5-dev \
    qttools5-dev-tools \
    qtwayland5
```

AndKe, 2022-09-30:
Thanks - I tried to find them here: https://github.com/webcamoid/webcamoid/wiki/Raw-build-and-install

Any chance of an updated .appimage?

closed 2023-06-01 by hipersayanX