webcamoid/akvcam #67: V4L2_BUF_FLAG_LAST is set and epipe is returned

taylorchu, 2022-12-09:
https://patchwork.kernel.org/project/linux-media/patch/20171208140128.19740-1-p.zabel@pengutronix.de/

I wonder whether we could skip this flag to avoid epipe errors.

taylorchu, 2022-12-12:
what might be the patch here if the kernel is too old, and cannot be easily updated? thanks!

hipersayanX, 2022-12-14:
You can see [here](https://kernel.org/) a list of all kernel versions maintained by the Linux developers, those are the only versions supported by akvcam, all those versions will receive the patch soon or later, and you are most probably using one of those kernels right now, if not just upgrade your system, also v4l2-compliance isn't reporting any bug in the driver, so this is most probably an edge case, therefore, there is not any reason to care about that patch.
If you still insist on applying the patch you can trace the code starting from [here](https://github.com/torvalds/linux/blob/e08466a7c00733a501d3c5328d29ec974478d717/drivers/media/common/videobuf2/videobuf2-v4l2.c#L601) and look where that function is being used.

Also, didn't figured until now but

>Dec. 8, 2017, 2:01 p.m. UTC

That bug was patched 5 years ago. Again there is not any reason to care about.

taylorchu, 2022-12-14:
I am not sure why I still see epipe error then.

hipersayanX, 2022-12-15:
Give me the code snippet that is failing, also ```uname -r```.

taylorchu, 2022-12-15:
https://github.com/wine-mirror/wine/blob/ccdef80543ff5c1f08a8aaff39c3f623522152df/dlls/qcap/v4l.c#L355

This is wine read v4l2 frame.

```
# uname -r
5.4.0-1065-gke
```

hipersayanX, 2022-12-15:
Which program did you tested? Also, I need the [debugging](https://github.com/webcamoid/akvcam/wiki/Debugging-the-driver) log.

taylorchu, 2022-12-15:
It is an internal project, which I cannot share just yet.

> streaming_stopped

I can test a patched akvcam version without explicit streaming_stop. ie. it just starts on module init, and we never call stop in any case. I am not sure how to do that yet.

hipersayanX, 2022-12-17:
>It is an internal project, which I cannot share just yet.

Ok, but did you tried running a working program like ffmpeg/ffplay, or a working [code example](https://github.com/microsoft/Windows-classic-samples/blob/main/Samples/Win7Samples/multimedia/directshow/capture/playcap/playcap.cpp) to discard a bug in your code?

>I can test a patched akvcam version without explicit streaming_stop. ie. it just starts on module init, and we never call stop in any case. I am not sure how to do that yet.

What you expect to happen by removing a key function? if the program have a bug, you must fix the but, not removing code at random expecting the code to be fixed magically.