webcamoid/akvcam #68: Wrong color on little endian machine

mantorchan, 2023-02-22:
After compiling and loaded in android , some color streaming to capture device is wrong, eg. put in rgb24(190,0,0),  would get (155,0,11) from the decoded yuy2 reading directly from /dev/video*. I have tried swap the r and b in akvcam_RGB24 struct,  and it's blue now. Am I missing something?

hipersayanX, 2023-02-22:
>After compiling and loaded in android

How you did that? Android camera drivers are not supposed to work with V4L2, if you want to create a virtual camera for Android you must use [HAL 3](https://source.android.com/docs/core/camera/camera3).

>some color streaming to capture device is wrong, eg. put in rgb24(190,0,0), would get (155,0,11) from the decoded yuy2 reading directly from /dev/video*.

[Here](https://github.com/webcamoid/akvcam/blob/918587d7987caf5b55488471cd7cb3bad2f89177/src/frame.c#L841) you can see the formulas used to convert between RGB <-> YUV, and [here](https://gist.github.com/hipersayanX/f9f5e46fa7f23075df31b8f6b7e2f016) is a code snipped you can use to test it. The conversion should give you:

```
RGB(190, 0, 0) -> YUV(65, 100, 211)
YUV(65, 100, 211) -> RGB(190, 1, 1)
```

Idk how did you get that (155,0,11) result. Also take in mind that YUY2 is half of the U plane, and half of the V plane, so the colors will look much darker than the colors of the original image because there is an information loss during the conversion.

mantorchan, 2023-02-22:
Kernel 5.19 is used in Android 12 , this module can be compiled just like others either out of tree or in source as a driver module. After loaded, akvcam  would be recognized as an external camera.

>   and half of the V plane, so the colors will look much darker than the colors of the original image because there is an information loss during the conversion.

I have compared the same pic in x86 computer, which is normal,at least  could not tell much difference by eye.
The layout of rgb24 color struct used in akvcam wouldn't affect by endianness,I might point the question to a wrong way . The buffer feed to output device, is a normal r,g,b sequence,but the log added in akvcam source shows  unexpected values ,r and b is swapped, the weird thing comes here, the decoded pic read from capture looks like darker than it should be, but is not  r-b swapped. Something might swapped them before go es in the adjust routine.

hipersayanX, 2023-02-24:
>Kernel 5.10 is used in Android 12 , this module can be compiled just like others either out of tree or in source as a driver module. After loaded, akvcam would be recognized as an external camera.

Give me a full simple and clear baby step by step tutorial on how you did it. I don't want to compile the whole Android thing, just want to test it in the emulator. I want to try it later :smile: 

>the weird thing comes here, the decoded pic read from capture looks like darker than it should be, but is not r-b swapped. Something might swapped them back after the adjust routine?

As said before, if the RGB order is ok in the capturing side, let say you draw a red rectangle and you can see it as red,  you draw a green rectangle and you can see it as green,  and you draw a blue rectangle and you can see it as blue, but you see the colors darker than the originals, it's a normal behavior of the RGB <-> YUY2 conversion. Also, you can try maybe swapping the [YUY2 structure](https://github.com/webcamoid/akvcam/blob/918587d7987caf5b55488471cd7cb3bad2f89177/src/frame.c#L92).
Anyway, the [conversion functions](https://github.com/webcamoid/akvcam/blob/918587d7987caf5b55488471cd7cb3bad2f89177/src/frame.c#L1285) are just a quick hack and don't handles endianness. My idea is to port the [video converter from Webcamoid](https://github.com/webcamoid/webcamoid/blob/master/libAvKys/Lib/src/akvideoconverter.cpp) (which actually handles endianness) to C, but don't expect it in the near time.