webcamoid/akvcam #69: Multiple consumers for single video capture device

alopatindev, 2023-04-01:
I want to allow any number of consumers for single video capture device, just like v4l2loopback does. This will allow easy intuitive interaction of any programs that access webcams: no need to check whether specific device is captured right now, no need to fight program that fails to release capture device for some reason, etc.

Is it currently possible with akvcam? If not, is it planned to support, or it's never gonna be implemented due to some limitation/design choice?

Thanks.

hipersayanX, 2023-04-03:
>Is it currently possible with akvcam?

No, isn't possible

>If not, is it planned to support

No and never

>or it's never gonna be implemented due to some limitation/design choice?

The device opening is handled by [v4l2_fh_open](https://github.com/webcamoid/akvcam/blob/9b8c82d9bf48b9d280b88f92602ed435badd27fd/src/device.c#L671) as it's supposed to be implemented according to V4L2 developers, many parts of v4l2loopback code are implemented in a very hacky way.

If you want to stream the same frames to many clients, create as many capture device as the clients you want to send the frames, and [connect one output to all of them](https://github.com/webcamoid/akvcam/wiki/Configure-the-cameras#connecting-the-devices).

closed 2023-04-03 by alopatindev

hipersayanX, 2023-04-03:
>Thanks for explanation, got it, that must be a fundamental [V4L2 limitation](https://www.kernel.org/doc/html/v4.9/media/uapi/v4l/open.html#multiple-opens):

Exactly, did not remembered that part of the documentation, but that was the explanation. v4l2loop can do that, but it do it in a hacky way, while I prefer to play with the standards.

>At least for sound devices we have dsnoop/dmix, but nothing convenient for webcams.

There are, that was the reason why Pipewire was created.

https://blogs.gnome.org/uraeus/2018/01/26/an-update-on-pipewire-the-multimedia-revolution-an-update/