webcamoid/akvcam #70: setup.sh failed on ubuntu

matlad, 2023-07-11:
when I install release using akvcam-installer-cli-linux-1.2.4.run I get error about bad substitution.
```
....
Public License instead of this License.
Please type y to accept, n otherwise: y
Creating directory /opt/akvcam
Verifying archive integrity...  100%   All good.
Uncompressing Install akvcam  100%  
akvcam installation started

Writting install data
Writting undo script
Writting uninstall script
./setup.sh: 417: Bad substitution

```
Script has shebang  #!/bin/sh but use Shell Parameter Expansion which is not part of POSIX and is specific for bash. 
on Ubuntu default shell is dash so this fails

work around:  force running by bash
```
bash setup.sh

```

hipersayanX, 2023-07-19:
It's marking [this line](https://github.com/webcamoid/akvcam/blob/cad51f4636788b5a9c30f70e44b42b440d93b230/ports/deploy/setup.sh#L417). 

```sh
echo "Installing missing dependencies: ${missing_dependencies// /, }"
```

Is there a generic way of making string substitution or I must force bash in the shebang?

matlad, 2023-07-26:
best thing i can thing of is using  `sed` 
example
```
 echo "Installing missing dependencies: $(echo ${missing_dependencies}| sed 's/ /, /g')"
```