webcamoid/akvcam #82: Unknown symbol vb2_vmalloc_memops

0x5e, 2024-01-26:
Hi, I'm trying to build & run akvcam in alpine, when `modprobe akvcam`, dmesg says: "Unknown symbol vb2_vmalloc_memops".
I'm not familiar with linux kernel, ChatGPT tells me I should load `videobuf2_vmalloc` first, either manually run `modprobe videobuf2_vmalloc` or add `request_module("videobuf2_vmalloc")` in akvcam init function. Will it be better adding `request_module("videobuf2_vmalloc")` into this project?

hipersayanX, 2024-01-27:
Try adding ```request_module("videobuf2_vmalloc");``` [here](https://github.com/webcamoid/akvcam/blob/8ddafe1b08ffd3053e4529f0e80db448cd177028/src/module.c#L42) and check if it works.

hipersayanX, 2025-07-03:
No answer, no reproducible, closing.

closed 2025-07-03 by hipersayanX