webcamoid/akvcam #85: Clarification of output device

l4es, 2024-03-25:
Dear akvcam owners,

I would like to know if the output device is the device appearing when for example a webcam plugged into the system or is it part of the virtual pair (output --> capture) of akvcam? 

In case it is part of the virtual pair (output --> capture) of akvcam, if I need to clone a single /dev/video0 of the system (appearing when I plug the webcam in the above example) to multiple captures devices, would I need to stream the system's /dev/video0 to the output devices of the virtual pairs of akvcam? And would the be the limitation of the formats supported by the output device of the virtual pair ?

Best Regards,
Khang Le-Truong

hipersayanX, 2024-03-25:
>I would like to know if the output device is the device appearing when for example a webcam plugged into the system or is it part of the virtual pair (output --> capture) of akvcam?

The output device is the virtual camera, I called it "output device" because I want to expand the output types, to for example streaming in a future.

>In case it is part of the virtual pair (output --> capture) of akvcam, if I need to clone a single /dev/video0 of the system (appearing when I plug the webcam in the above example) to multiple captures devices, would I need to stream the system's /dev/video0 to the output devices of the virtual pairs of akvcam? And would there be the limitation of the formats supported by the output device of the virtual pair ?

If you are using v4l2loopbak as the virtual camera driver, you can create one output device and it can be used by many programs at same time.

hipersayanX, 2024-03-25:
>I would like to know if the output device is the device appearing when for example a webcam plugged into the system or is it part of the virtual pair (output --> capture) of akvcam?

The output device is where you sent the video frames, the capture device is from where the client programs will get the video.

> In case it is part of the virtual pair (output --> capture) of akvcam, if I need to clone a single /dev/video0 of the system (appearing when I plug the webcam in the above example) to multiple [...]

You have to create one output, as many capture devices as you want, then connect the output to each capture device. And [here](https://github.com/webcamoid/akvcam/wiki/Configure-the-cameras#defining-the-formats) is a list of the supported formats for each one.

l4es, 2024-03-25:
Dear  @hipersayanX,

Thanks for your reply. So if I understand correctly for the output device and its supported formats, I will have to convert the pixel format of the  systems' device (/dev/video0) to the format(s) supported by the output device before streaming to it, right? 

Best Regards,
Khang

hipersayanX, 2024-03-26:
>Thanks for your reply. So if I understand correctly for the output device and its supported formats, I will have to convert the pixel format of the systems' device (/dev/video0) to the format(s) supported by the output device before streaming to it, right?

Yes, that's right. I will add more output formats in future versions.

closed 2025-07-03 by hipersayanX