webcamoid/akvcam #86: Can't install on Ubuntu

philippulus, 2024-05-28:
Don't find how to install. 

- $ sudo apt-get install dkms
- Downloaded [akvcam installer gui for Linux](https://github.com/webcamoid/akvcam/releases), changed it to ‘executable as program’, then run it. After that there are several folders with 'akvcam' in the name, but no folder 'akvcam/src' as suggested in the [wiki](cd akvcam/src):
-- /opt/akvcam
-- /usr/src/akvcam-1.2.6
-- /var/lib/dkms/akvcam
- $ cd akvcam/src
  bash: cd: akvcam/src: No such file or directory

It's really unclear to me about how to install the driver, if it is now actually installed, and how to use it.

philippulus, 2024-05-28:
Now trying to follow [this procedure](https://gist.github.com/okapies/550e541535d299717536f103802520a6) but get stuck as well:

- $ git clone https://github.com/webcamoid/akvcam.git
- $ cd akvcam
- $ git checkout refs/tags/1.2.6
- $ cd akvcam/src
- $ make
- $ make clean
- $ make -p /usr/src/akvcam-1.2.6
- $ cp -ar src/* /usr/src/akvcam-1.2.6
  **cp: cannot stat 'src/*': No such file or directory**

hipersayanX, 2024-05-29:
The wiki instructions are for the raw source code, not  the installer, the installer does already all the install steps.

>- $ cd akvcam/src
>bash: cd: akvcam/src: No such file or directory

Because your changing to the wrong directory, in the previous line is clearly marked were the source files where installed.

>- Downloaded [akvcam installer gui for Linux](https://github.com/webcamoid/akvcam/releases), changed it to ‘executable as program’, then run it. After that there are several folders with 'akvcam' in the name, but no folder 'akvcam/src' as suggested in the [wiki](https://github.com/webcamoid/akvcam/wiki/Build-and-install):
>**/opt/akvcam** :point_left: 
>**/usr/src/akvcam-1.2.6** :point_left:
>/var/lib/dkms/akvcam

>It's really unclear to me about how to install the driver, if it is now actually installed, and how to use it.

The module was already installed with the installer. On how to use it:

https://github.com/webcamoid/akvcam/wiki/Configure-the-cameras
https://github.com/webcamoid/akvcam/wiki/Usage-and-examples
https://github.com/webcamoid/akvcam/wiki/Video-output-example-in-C

>Now trying to follow [this procedure](https://gist.github.com/okapies/550e541535d299717536f103802520a6) but get stuck as well

You have already changed to the **src** directory

>$ cd **akvcam/src** :point_left: 
>$ make
>$ make clean
>$ mkdir -p /usr/src/akvcam-1.2.6 **Also, in this line is 'mkdir', not 'make'**
>$ cp -ar **src**/* /usr/src/akvcam-1.2.6  :point_left:
>cp: cannot stat 'src/*': No such file or directory

Just install the files with *
But again all these steps are not necessary because the installer already did all that.
Also an advice in good terms, learn to use the command line before jumping into this level of complexity.

https://web.njit.edu/~alexg/courses/cs332/OLD/F2020/hand3f20/Linux-Tutorial.pdf
https://bjpcjp.github.io/pdfs/devops/linux-commands-handbook.pdf
https://cbw.sh/static/class/linuxbasics/learning_linux.pdf

philippulus, 2024-06-02:
Thank you @hipersayanX 
However, I am not coming any farther. 
I have now started from scratch after uninstalling the module, purging the system of any akvcam files or directories, rebooted the PC and taken it up against by installing via the installer. 
Setting up the camera seems to be OK with the series of command lines:
```
sudo mkdir -p /etc/akvcam
sudo touch /etc/akvcam/config.ini
sudo chmod -vf 644 /etc/akvcam/config.ini
```
Then copying the config.ini files proposed by [Linux Fake Background Camera.
  ](https://github.com/fangfufu/Linux-Fake-Background-Webcam).
  
[image]

The issue is encountered with the next stage of using the camera: 
```
/akvcam/src$ sudo modprobe videodev
[sudo] password for minisforum-um790pro-nicolas: 
sudo insmod akvcam.ko
insmod: ERROR: could not load module akvcam.ko: No such file or directory
```
[image]

What please am I missing here?

philippulus, 2024-06-07:
Please kindly help on how to complete the installation with the installer as proposed.
I have completed all the steps as indicated but cannot get to a usable camera.

hipersayanX, 2024-06-07:
>/akvcam/src$ sudo modprobe videodev
>[sudo] password for 'user': 
>**/akvcam/src**$ sudo **insmod** akvcam.ko :point_left: 
>insmod: ERROR: could not load module akvcam.ko: No such file or directory

**insmod** is for loading a file module, and the module **must be** in the same directory where insmod is executed, dkms doesn't install the compiled modules there.
After installing the modules with the installer, you can check that the module was installed with:

```cat /lib/modules/$(uname -r)/modules.dep | grep akvcam```

and then load the module with:

```sudo modprobe akvcam```

then check that the module was loaded with:

```lsmod | grep akvcam```

closed 2025-07-03 by hipersayanX