webcamoid/akvcam #88: Android 13

Xf0ur, 2024-07-27:
Hello,
I'm unable to cross-compile akvcam for Android 13.
ERROR: modpost: module akvcam uses symbol kernel_read from namespace VFS_internal_I_am_really_a_filesystem_and_am_NOT_a_driver, but does not import it.
ERROR: modpost: module akvcam uses symbol filp_open from namespace VFS_internal_I_am_really_a_filesystem_and_am_NOT_a_driver, but does not import it.
Can you help?

hipersayanX, 2024-07-30:
akvcam is for GNU/Linux, not Android. I doubt that, even if you can compile and somehow install it in Android, you will make it work in there. For creating a virtual camera in Android, as I understand, you must implement it using the [HAL API](https://source.android.com/docs/core/camera/camera3), but I did not tried it yet.
But, if you succeed hacking into it and making it work in Android, please tell me how you did it.

Xf0ur, 2024-07-30:
Actually akvcam works with android emulator.
That's why i tried to cross-compile it and give it a try though v4l2loopback is for linux but it gives no errors if compiled for android.
Thanks anyway.

hipersayanX, 2024-08-06:
>Actually akvcam works with android emulator.

Yes, the virtual camera works in the emulator, because the emulator emulate a real camera, but the module does not (or may not) work in the Android operating system it self. Your confusing the emulated hardware with the operating system which are not the same.
If you want to compile akvcam for Android, first at all you must compile the module again the same exact version and architecture or the kernel behind used by the target device.

[image]

Then you must install it in the device, don't ask me how because I never did that and don't have any idea how to do it.
After installing it there is not any guarantee that Android apps will detect it, for the simple reason that neither Android camera API nor the APKs make use of the kernels's V4L2 API.
Over that, there are not any software capable of using akvcam in Android, not even Webcamoid.
So whatever you want to do trying to compile akvcam for Android is just a mere experiment.

Xf0ur, 2024-08-09:
Hello,
That's what i was doing.
Firstly i cross-compiled the kernel then tried to compile the module but i get the following error just before akvcam.ko is compiled.
```
  CC [M]  /usr/src/akvcam/src/module.o
  CC [M]  /usr/src/akvcam/src/attributes.o
  CC [M]  /usr/src/akvcam/src/buffers.o
  CC [M]  /usr/src/akvcam/src/controls.o
  CC [M]  /usr/src/akvcam/src/device.o
  CC [M]  /usr/src/akvcam/src/driver.o
  CC [M]  /usr/src/akvcam/src/file_read.o
  CC [M]  /usr/src/akvcam/src/format.o
  CC [M]  /usr/src/akvcam/src/frame.o
  CC [M]  /usr/src/akvcam/src/frame_filter.o
  CC [M]  /usr/src/akvcam/src/ioctl.o
  CC [M]  /usr/src/akvcam/src/list.o
  CC [M]  /usr/src/akvcam/src/log.o
  CC [M]  /usr/src/akvcam/src/map.o
  CC [M]  /usr/src/akvcam/src/rbuffer.o
  CC [M]  /usr/src/akvcam/src/settings.o
  CC [M]  /usr/src/akvcam/src/utils.o
  LD [M]  /usr/src/akvcam/src/akvcam.o
  MODPOST /usr/src/akvcam/src/Module.symvers
ERROR: modpost: module akvcam uses symbol kernel_read from namespace VFS_internal_I_am_really_a_filesystem_and_am_NOT_a_driver, but does not import it.
ERROR: modpost: module akvcam uses symbol filp_open from namespace VFS_internal_I_am_really_a_filesystem_and_am_NOT_a_driver, but does not import it.
make[2]: *** [scripts/Makefile.modpost:157: /usr/src/akvcam/src/Module.symvers] Error 1
make[2]: *** Deleting file '/usr/src/akvcam/src/Module.symvers'
make[1]: *** [Makefile:1868: modules] Error 2
```

You're right about the emulator.
akvcam can be used as follows
emulator -camera-back webcam0
So perhaps to make akvcam work in android system one has to implement/modify the camera HAL.

hipersayanX, 2024-08-10:
>So perhaps to make akvcam work in android system one has to implement/modify the camera HAL.

Exactly, take into account that Android isn't a like a normal GNU/Linux distro, Android just use Linux for the resources management and that's all, beside that, Google implements everything on their own.

>Btw,you tried akvcam with waydroid?

Nope, didn't know about that project, will check when I have some time.

Xf0ur, 2024-09-11:
Hello,
That issue can easily be solved by adding the following line to the end of module.c file :)
MODULE_IMPORT_NS(VFS_internal_I_am_really_a_filesystem_and_am_NOT_a_driver);

Regarding Waydroid, it doesn't seem to work though i made some progress as follows.

```
09-11 09:53:39.311  2910  2910 I cameraserver: ServiceManager: 0xf7c59410
09-11 09:53:39.311  2910  2910 I CameraService: CameraService started (pid=2910)
09-11 09:53:39.311  2910  2910 I CameraService: CameraService process starting
09-11 09:53:39.313  2910  2910 I CameraProviderManager: Connecting to new camera provider: legacy/0, isRemote? 1
09-11 09:53:39.313  2910  2910 I CameraProviderManager: Enumerating new camera device: device@3.3/legacy/0
09-11 09:53:39.313  2910  2910 E CameraProviderManager: DeviceInfo3: Converted ICameraDevice instance to nullptr
09-11 09:53:39.313  2910  2910 I CameraProviderManager: Camera provider legacy/0 ready with 1 camera devices
09-11 09:53:39.314  2910  2910 I CameraService: onDeviceStatusChanged: Status changed for cameraId=0, newStatus=1
09-11 09:53:39.314  2910  2910 I CameraService: onDeviceStatusChanged: Unknown camera ID 0, a new camera is added
09-11 09:53:39.314  2910  2912 W CameraProviderManager: addProviderLocked: Camera provider HAL with name 'legacy/0' already registered
09-11 09:53:39.315    88   329 D AF::TrackHandle: OpPlayAudio: track:56 usage:5 not muted
09-11 09:53:39.315    88   196 D AF::TrackHandle: OpPlayAudio: track:55 usage:13 not muted
09-11 09:53:39.315    88   197 D AF::TrackHandle: OpPlayAudio: track:57 usage:13 not muted
09-11 09:53:39.315  2910  2910 I HidlServiceManagement: Registered android.frameworks.cameraservice.service@2.1::ICameraService/default
09-11 09:53:39.315  2910  2910 I CameraService: CameraService pinged cameraservice proxy
09-11 09:53:39.315  2910  2910 I cameraserver: ServiceManager: 0xf7c59410 done instantiate
```
When i list cameras i got
[server] INFO: List of cameras:
    --camera-id=0    (external, 3280x2464, fps=[30])

I dunno why legacy is used as a camera provider and why it's identified as an external camera, any idea?

hipersayanX, 2024-09-13:
>Hello,
>That issue can easily be solved by adding the following line to the end of module.c file :)
>MODULE_IMPORT_NS(VFS_internal_I_am_really_a_filesystem_and_am_NOT_a_driver);

Would that also affect normal builds or its a safe option?

>When i list cameras i got
>[server] INFO: List of cameras:
>--camera-id=0 (external, 3280x2464, fps=[30])
>
>Android emulator output
>--camera-id=0 (back, 640x480, fps=[15, 30])
>
>Although i'm using same akvcam config,

Let's clarify this, you are running those commands from the host system (your PC), not inside Android itself, right?
Because, what understand is that you want to compile the module for using it inside Android.
Also, post the config that you are using.

>I started emulator with -camera-back webcam0 however there is no such option in Waydroid, any idea?

Again, never used Waydroid so I can't help much there.

Xf0ur, 2024-09-14:
> Would` that also affect normal builds or its a safe option?

It's just to bypass the android cross-compilation error, it's not needed for normal builds.

> Let's clarify this, you are running those commands from the host system (your PC), not inside Android itself, right?
> Because, what understand is that you want to compile the module for using it inside Android.
> Also, post the config that you are using.
> 

I surely wanted to compile akvcam module to use inside Android.
I did a quick test already but it didn't work.

```
cameraserver: Connecting to new camera provider: external/0, isRemote? 1
CameraProviderManager: Camera provider external/0 ready with 0 camera devices
cameraserver: Connecting to new camera provider: legacy/0, isRemote? 1
CamPrvdr@2.5-legacy: notifyDeviceStateChange: New device state: 0x0
CameraProviderManager: Camera provider legacy/0 ready with 0 camera devices
CameraProviderManager: addHidlProviderLocked: Camera provider HAL with name 'external/0' already registered
CameraProviderManager: addHidlProviderLocked: Camera provider HAL with name 'legacy/0' already registered
HidlServiceManagement: Registered android.frameworks.cameraservice.service@2.2::ICameraService/default
CameraService: CameraService pinged cameraservice proxy
CameraService_proxy: Could not notify cameraserver, camera service not available.
CameraService_proxy: Could not notify camera service of user switch, retrying...
```

`scrcpy --list-cameras`

```
[server] INFO: List of cameras:
    (none)
```

Now, what i'm trying to make it work with Waydroid.
I load akvcam module on my host pc, launch Waydriod container then run 

> scrcpy --list cameras

Output is :
```
[server] INFO: List of cameras:
--camera-id=0 (external, 3280x2464, fps=[30])
```

It means that it's working however i dunno why it's identified as an external camera with 3280x2464 resolution. 
As a result I only see a black screen in Waydroid default camera app.

I'm using the exmaple config fle from wiki which contain 640x480 resolution. only.

closed 2025-07-03 by hipersayanX