webcamoid/akvcam #89: how to change the output pictures

996Godie, 2024-08-18:
how to change the output pictures , where is the config of picture path, i used the GUI-installer

hipersayanX, 2024-08-18:
It's explained in [the wiki](https://github.com/webcamoid/akvcam/wiki/Configure-the-cameras#no-signal-picture).

996Godie, 2024-08-20:
thanks for reply，the requestion has been resolved

Another question is how to make the output of the camera have a dynamic effect?

hipersayanX, 2024-08-20:
Do you mean an animated no signal picture? that is not possible for the moment, maybe in a future I can add support for reading GIF files, but don't expect it in the near time.
For adding visual effects to the virtual camera, that depends on the program using it, you pre-process the video frames from your side and then send it to the virtual camera.

closed 2025-07-03 by hipersayanX