webcamoid/akvcam #91: akvcam prevents linux to start

j2l, 2025-06-04:
When update doesn't finish before reboot, the kernel is broken and akvcam is cited as blocking the fix.
Please tell me how to remove akvcam competely.

```
sudo dpkg --configure -a
Setting up linux-image-6.12.10-76061203-generic (6.12.10-76061203.202412060638~1748542656~22.04~663e4dc) ...
Setting up linux-headers-6.12.10-76061203-generic (6.12.10-76061203.202412060638~1748542656~22.04~663e4dc) ...
/etc/kernel/header_postinst.d/dkms:
Sign command: /usr/bin/kmodsign
Binary update-secureboot-policy not found, modules won't be signed

Autoinstall of module akvcam/1.2.3 for kernel 6.12.10-76061203-generic (x86_64)
Building module(s)...(bad exit status: 2)
Failed command:
make -j12 KERNELRELEASE=6.12.10-76061203-generic KERNEL_DIR=/lib/modules/6.12.10-76061203-generic/build all
ERROR (dkms apport): binary package for akvcam: 1.2.3 not found

Error! Bad return status for module build on kernel: 6.12.10-76061203-generic (x86_64)
Consult /var/lib/dkms/akvcam/1.2.3/build/make.log for more information.

Autoinstall on 6.12.10-76061203-generic succeeded for module(s) nvidia system76 system76_acpi system76-io v4l2loopback virtualbox.
Autoinstall on 6.12.10-76061203-generic failed for module(s) akvcam(10).

Error! One or more modules failed to install during autoinstall.
Refer to previous errors for more information.
run-parts: /etc/kernel/header_postinst.d/dkms exited with return code 1
dpkg: error processing package linux-headers-6.12.10-76061203-generic (--configure):
 installed linux-headers-6.12.10-76061203-generic package post-installation script subprocess returned error exit status 1
dpkg: dependency problems prevent configuration of linux-headers-generic:
 linux-headers-generic depends on linux-headers-6.12.10-76061203-generic (= 6.12.10-76061203.202412060638~1748542656~22.04~663e4dc); however:
  Package linux-headers-6.12.10-76061203-generic is not configured yet.

dpkg: error processing package linux-headers-generic (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of linux-generic:
 linux-generic depends on linux-headers-generic (= 6.12.10-76061203.202412060638~1748542656~22.04~663e4dc); however:
  Package linux-headers-generic is not configured yet.

dpkg: error processing package linux-generic (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of linux-system76:
 linux-system76 depends on linux-generic (= 6.12.10-76061203.202412060638~1748542656~22.04~663e4dc); however:
  Package linux-generic is not configured yet.

dpkg: error processing package linux-system76 (--configure):
 dependency problems - leaving unconfigured
Processing triggers for linux-image-6.12.10-76061203-generic (6.12.10-76061203.202412060638~1748542656~22.04~663e4dc) ...
/etc/kernel/postinst.d/dkms:
Sign command: /usr/bin/kmodsign
Binary update-secureboot-policy not found, modules won't be signed

Autoinstall of module akvcam/1.2.3 for kernel 6.12.10-76061203-generic (x86_64)
Building module(s)...(bad exit status: 2)
Failed command:
make -j12 KERNELRELEASE=6.12.10-76061203-generic KERNEL_DIR=/lib/modules/6.12.10-76061203-generic/build all
ERROR (dkms apport): binary package for akvcam: 1.2.3 not found

Error! Bad return status for module build on kernel: 6.12.10-76061203-generic (x86_64)
Consult /var/lib/dkms/akvcam/1.2.3/build/make.log for more information.

Autoinstall on 6.12.10-76061203-generic succeeded for module(s) nvidia system76 system76_acpi system76-io v4l2loopback virtualbox.
Autoinstall on 6.12.10-76061203-generic failed for module(s) akvcam(10).

Error! One or more modules failed to install during autoinstall.
Refer to previous errors for more information.
run-parts: /etc/kernel/postinst.d/dkms exited with return code 1
dpkg: error processing package linux-image-6.12.10-76061203-generic (--configure):
 installed linux-image-6.12.10-76061203-generic package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 linux-headers-6.12.10-76061203-generic
 linux-headers-generic
 linux-generic
 linux-system76
 linux-image-6.12.10-76061203-generic
```

```
cat /var/lib/dkms/akvcam/1.2.3/build/make.log
DKMS (dkms-3.2.1) make.log for akvcam/1.2.3 for kernel 6.12.10-76061203-generic (x86_64)
Wed Jun  4 08:57:07 AM CEST 2025

Building module(s)
# command: make -j12 KERNELRELEASE=6.12.10-76061203-generic KERNEL_DIR=/lib/modules/6.12.10-76061203-generic/build all
make -C /lib/modules/6.12.10-76061203-generic/build M=/var/lib/dkms/akvcam/1.2.3/build  modules
make[1]: Entering directory '/usr/src/linux-headers-6.12.10-76061203-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-12 (Ubuntu 12.3.0-1ubuntu1~22.04) 12.3.0
  You are using:           gcc-12 (Ubuntu 12.3.0-1ubuntu1~22.04) 12.3.0
  CC [M]  /var/lib/dkms/akvcam/1.2.3/build/module.o
  CC [M]  /var/lib/dkms/akvcam/1.2.3/build/attributes.o
  CC [M]  /var/lib/dkms/akvcam/1.2.3/build/buffers.o
  CC [M]  /var/lib/dkms/akvcam/1.2.3/build/controls.o
  CC [M]  /var/lib/dkms/akvcam/1.2.3/build/device.o
  CC [M]  /var/lib/dkms/akvcam/1.2.3/build/driver.o
  CC [M]  /var/lib/dkms/akvcam/1.2.3/build/file_read.o
  CC [M]  /var/lib/dkms/akvcam/1.2.3/build/format.o
  CC [M]  /var/lib/dkms/akvcam/1.2.3/build/frame.o
  CC [M]  /var/lib/dkms/akvcam/1.2.3/build/frame_filter.o
  CC [M]  /var/lib/dkms/akvcam/1.2.3/build/ioctl.o
  CC [M]  /var/lib/dkms/akvcam/1.2.3/build/list.o
  CC [M]  /var/lib/dkms/akvcam/1.2.3/build/log.o
  CC [M]  /var/lib/dkms/akvcam/1.2.3/build/map.o
  CC [M]  /var/lib/dkms/akvcam/1.2.3/build/rbuffer.o
  CC [M]  /var/lib/dkms/akvcam/1.2.3/build/settings.o
  CC [M]  /var/lib/dkms/akvcam/1.2.3/build/utils.o
/var/lib/dkms/akvcam/1.2.3/build/buffers.c: In function ‘akvcam_buffers_new’:
/var/lib/dkms/akvcam/1.2.3/build/buffers.c:107:16: error: ‘struct vb2_queue’ has no member named ‘min_buffers_needed’
  107 |     self->queue.min_buffers_needed = AKVCAM_BUFFERS_MIN;
      |                ^
/var/lib/dkms/akvcam/1.2.3/build/buffers.c: In function ‘akvcam_buffers_count’:
/var/lib/dkms/akvcam/1.2.3/build/buffers.c:145:23: error: ‘const struct vb2_queue’ has no member named ‘min_buffers_needed’
  145 |     return self->queue.min_buffers_needed;
      |                       ^
/var/lib/dkms/akvcam/1.2.3/build/buffers.c: In function ‘akvcam_buffers_set_count’:
/var/lib/dkms/akvcam/1.2.3/build/buffers.c:150:16: error: ‘struct vb2_queue’ has no member named ‘min_buffers_needed’
  150 |     self->queue.min_buffers_needed = nbuffers;
      |                ^
/var/lib/dkms/akvcam/1.2.3/build/buffers.c: In function ‘akvcam_buffers_count’:
/var/lib/dkms/akvcam/1.2.3/build/buffers.c:146:1: error: control reaches end of non-void function [-Werror=return-type]
  146 | }
      | ^
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:229: /var/lib/dkms/akvcam/1.2.3/build/buffers.o] Error 1
make[3]: *** Waiting for unfinished jobs....
/var/lib/dkms/akvcam/1.2.3/build/device.c: In function ‘akvcam_device_new’:
/var/lib/dkms/akvcam/1.2.3/build/device.c:133:14: error: ‘V4L2_DEVICE_NAME_SIZE’ undeclared (first use in this function)
  133 |              V4L2_DEVICE_NAME_SIZE,
      |              ^~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/akvcam/1.2.3/build/device.c:133:14: note: each undeclared identifier is reported only once for each function it appears in
make[3]: *** [scripts/Makefile.build:229: /var/lib/dkms/akvcam/1.2.3/build/device.o] Error 1
make[2]: *** [/usr/src/linux-headers-6.12.10-76061203-generic/Makefile:1943: /var/lib/dkms/akvcam/1.2.3/build] Error 2
make[1]: *** [Makefile:224: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.12.10-76061203-generic'
make: *** [Makefile:44: all] Error 2

# exit code: 2
# elapsed time: 00:00:02
----------------------------------------------------------------
```

j2l, 2025-06-04:
I went wild on it:
```
sudo modprobe -r akvcam
sudo systemctl stop akvcam
sudo rm -rf /usr/src/akvcam-*
sudo rm -f /etc/modules-load.d/akvcam.conf
sudo rm -f /etc/modprobe.d/akvcam.conf
sudo rm -f /etc/systemd/system/akvcam.service
sudo rm -f /dev/akvcam*
```
and finally went atomic:
```
sudo rm -R /var/lib/dkms/akvcam
sudo dpkg --configure -a
sudo dkms status
```
-> OK

Lost hours.
Now I'll stick with OBS Studio virtual cam and forget about webcamoid.

hipersayanX, 2025-07-03:
At moment of opening this issue, the latest version was 1.2.6, and you were trying to install a version that was released 3 years ago, of course you will get errors when trying to install/uninstall a kernel module that was made for older kernels.

```
Autoinstall of module akvcam/1.2.3 for kernel 6.12.10-76061203-generic (x86_64)
```

closed 2025-07-03 by hipersayanX